Koenkk/zigbee2mqtt #180: Aqara sensors only report a few messages

dmurray14, 2018-07-05:
New user, spun up a docker instance with a flashed CC2531. Everything runs fine, and I can pair up a couple WSDCGQ11LM sensors, but they don't last long. They will report messages for the first minute or two after pairing, then go completely dark. I understand the WSDCGQ11LM sensors generally only report when the temp changes noticeably, but I'm talking about no reports for days at a time when the temperature has definitely changed - and the sensor is only about 5 feet from the CC2531.

Here's a scrubbed log. This starts with re-pairing the sensor. It reported for about 2 minutes, then nothing. Any ideas?

```
> zigbee2mqtt@0.0.0 start /app
> node index.js

2018-7-4 18:05:24 INFO Starting zigbee2mqtt version 0.0.0 (commit #7cec5f2)
2018-7-4 18:05:24 INFO Starting zigbee-shepherd
2018-7-4 18:05:25 INFO zigbee-shepherd started
2018-7-4 18:05:25 INFO Currently 2 devices are joined:
2018-7-4 18:05:25 INFO basement-thp (0x00158d000xxxxxx3): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-7-4 18:05:25 INFO 0x00158d000xxxxxx1 (0x00158d000xxxxxx1): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
2018-7-4 18:05:25 WARN `permit_join` set to  `true` in configuration.yaml.
2018-7-4 18:05:25 WARN Allowing new devices to join.
2018-7-4 18:05:25 WARN Set `permit_join` to `false` once you joined all devices.
2018-7-4 18:05:25 INFO Zigbee: allowing new devices to join.
2018-7-4 18:05:25 INFO Connecting to MQTT server at mqtt://<mymqttserver>
2018-7-4 18:05:25 INFO zigbee-shepherd ready
2018-7-4 18:05:25 INFO Connected to MQTT server
2018-7-4 18:05:25 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-7-4 18:05:43 WARN Message without device!
spinlock: false []
2018-7-4 18:06:07 INFO Device incoming...
2018-7-4 18:06:07 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-7-4 18:06:13 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.14}'
2018-7-4 18:06:13 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.14,"humidity":80.37}'
2018-7-4 18:06:13 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.14,"humidity":80.37,"pressure":1016}'
2018-7-4 18:06:16 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.22,"humidity":80.37,"pressure":1016}'
2018-7-4 18:06:16 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.22,"humidity":73.42,"pressure":1016}'
2018-7-4 18:06:16 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.22,"humidity":73.42,"pressure":1016}'
2018-7-4 18:06:19 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.21,"humidity":73.42,"pressure":1016}'
2018-7-4 18:06:19 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.21,"humidity":66.86,"pressure":1016}'
2018-7-4 18:06:19 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.21,"humidity":66.86,"pressure":1016}'
2018-7-4 18:06:29 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.56,"humidity":66.86,"pressure":1016}'
2018-7-4 18:06:29 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.56,"humidity":79.19,"pressure":1016}'
2018-7-4 18:06:29 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":21.56,"humidity":79.19,"pressure":1016}'
2018-7-4 18:06:38 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.1,"humidity":79.19,"pressure":1016}'
2018-7-4 18:06:38 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.1,"humidity":85.25,"pressure":1016}'
2018-7-4 18:06:38 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.1,"humidity":85.25,"pressure":1016}'
2018-7-4 18:06:52 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.48,"humidity":85.25,"pressure":1016}'
2018-7-4 18:06:52 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.48,"humidity":72.11,"pressure":1016}'
2018-7-4 18:06:52 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.48,"humidity":72.11,"pressure":1016}'
2018-7-4 18:06:55 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.65,"humidity":72.11,"pressure":1016}'
2018-7-4 18:06:55 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.65,"humidity":65.32,"pressure":1016}'
2018-7-4 18:06:55 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.65,"humidity":65.32,"pressure":1016}'
2018-7-4 18:07:02 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.6,"humidity":65.32,"pressure":1016}'
2018-7-4 18:07:02 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.6,"humidity":58.29,"pressure":1016}'
2018-7-4 18:07:02 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.6,"humidity":58.29,"pressure":1016}'
2018-7-4 18:07:29 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.55,"humidity":58.29,"pressure":1016}'
2018-7-4 18:07:29 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.55,"humidity":52.2,"pressure":1016}'
2018-7-4 18:07:29 INFO MQTT publish, topic: 'zigbee2mqtt/basement-thp', payload: '{"temperature":22.55,"humidity":52.2,"pressure":1016}'
```

Koenkk, 2018-07-05:
@dmurray14 can you try replacing the battery of the sensors?

ciotlosm, 2018-07-05:
Might be a range issue as well.

dmurray14, 2018-07-06:
Has brand new batteries in, but I suppose they could be bad. Will try another set. Don't think it's range - the sensor is only a couple of feet from the CC2531.

dmurray14, 2018-07-09:
This must have been a docker issue of some sort. Installed and set up on a Raspi with the same config and everything works great.

Koenkk, 2018-07-10:
Do other sensors keep working in docker?

closed 2018-09-15 by stale[bot]

krisboit, 2018-10-18:
I had the same issue with door/window sensors when i switched to zigbee2mqtt, it seems the range was the issue, near my raspberrypi the pairing went smooth. Thanks @ciotlosm :)