Koenkk/zigbee2mqtt #181: cc2530+cc2591 firmware with hardware flow control disabled

lolorc, 2018-07-05:
I've been trying to use my cc2530+cc2591 dongle ( http://www.dx.com/p/429601 ) for some hours. 
I've been trying:
 - https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod.hex (not referenced in wiki) 
  - https://github.com/kirovilya/files/blob/master/CC2530ZNP-Test_for_CC2591.hex

I've been able to flash them but I've not been able to use the dongle, no response from serial...
I've been connecting P2_0 to GND.

And then I tried the coordinator firmware provided here : http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
I've been able to flash it, it's been working with zigbee2mqtt but obviously it's missing support for my frontend, so no communication with my devices.

I would love if you could upload a firmware with same set of features + support for my frontend.

Thanks in advance.

kirovilya, 2018-07-06:
@lolorc I using this dongle (but with baseboard) and it working.
Try additional connect p06 and p04 to GND

kirovilya, 2018-07-07:
Did you try it?

lolorc, 2018-07-07:
yes that did the trick thanks.
I look how I can compile this firmware myself I you can't provide it.

kirovilya, 2018-07-07:
@lolorc  Start here https://github.com/Koenkk/Z-Stack-firmware
and then here https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591

lolorc, 2018-07-07:
thanks, compared to my cc2531 I'm getting issues with cc2530/CC2591, some devices don't associate right away. I can't associate a cc2531 as a router to the cc2530/CC2591.
Is that a firmware issue or a device issue ?
would hardware cflow ontrol help ?

kirovilya, 2018-07-07:
@lolorc I do not known about flow control... You can adapt and build yourself firmware and try.
Than tell to us how it works.
About cc2531 router - I can not check. If all other is work, may be it is router firmware issue...

dzungpv, 2018-08-24:
@lolorc you can try my firmware, but the signal still not good, i am trying to make it better, just connect TX/RX and power and you are done.
[CC2530_CC2591ZNP.zip](https://github.com/Koenkk/zigbee2mqtt/files/2317314/CC2530_CC2591ZNP.zip)

lolorc, 2018-08-27:
Hi, I've just tried it (https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod.hex) it works without bridging P2_0, p06 and p04 to GND ! Thanks a bunch.
I agree the signal is not good, it looks like it the same than with a simple cc2531.

It looks like it also fix my issue connecting a cc2531 as a router to cc2530_cc2591 coordinator.

dzungpv, 2018-08-27:
@lolorc You can also try the newest firmware with increase the buffer size https://github.com/dzungpv/Z-Stack-firmware/tree/master/coordinator/CC2530_CC2591

lolorc, 2018-08-27:
@dzungpv as far as I know @Koenkk already merged your changes with the increased buffer. Or are there still differences with your fork/PR ?

lolorc, 2018-08-27:
router now pairs with my coordinator, but it just doesn't route anything, half of my devices stopped responding. I tried repairing it did nothing, everything went back to normal when I unplugged the router.

dzungpv, 2018-08-28:
@lolorc Yes, it is merged. Do you have CC2531 USB? It yes, you can try to use it and test if the problem exist. I am testing CC2530+CC2591 but it exist some problems too, but not like you see.

lolorc, 2018-08-28:
@dzungpv 
I guess I'll eventually move back to CC2531 USB. I've also got pairing and de-pairing issues with CC2530. As far as I remember it was straight forward with CC2531 and I'm absolutely sure It was ok with nxp JN5168.

lolorc, 2018-08-30:
I've got 2 cc2531 usb, just migrated from cc2530+cc2591 to cc2591 coordinor, I'm not able to pair a cc2531 router to the cc2531 coord.
There're really something funny going on here.
let's hope the upgrade to Z-Stack 3.0 will resolve all those weird issues. #211

lolorc, 2018-10-03:
those problems are fixed:
- new cc2530_cc2591 firmware doesn't need hardware flow control anymore
- new router firmware does scan every channel and does permit the use of an arbitrary panid

closed 2018-10-03 by lolorc

djelloul78, 2018-12-08:
Hi there!

@lolorc 
It seems you confirmed that a new firmware for CC2530+CC2591 based hardware have fixed some issues on 3rd October but could you tell me where can we found the file?

As far as I know, the only one available on GitHub is the file "CC2530ZNP-Test_for_CC2591.hex " that hasn't been updated since 03/2018...
Is there something I'm doing wrong to find the file, because I can't find anything newer than this firmware.

Thanks in advance for the help you'll provide to us.

Koenkk, 2018-12-09:
@djelloul78 that latest firmware, which are still in testing phase, can be found here: https://github.com/Koenkk/Z-Stack-firmware/tree/optimize/coordinator

Koenkk, 2018-12-09:
@djelloul78 than probably you didn't wire it right. There are multiple issues explaining how to do this (however I don't have any experience myself).

djelloul78, 2018-12-09:
@Koenkk The only way to wire it bad is to mix TX and RX lines wrong.
I did switch them and nothing is working.
I always have a Zigbee Shepherd Error in the logs.
I even tried with a serial terminal to check if there are some data output but I can't see anything.
I have a second set but it's more complex to use because it's an SMD designed one.
I'll check it and will see if it makes a difference.
Thanks for the help anyway