Koenkk/zigbee2mqtt #182: Provide New Log-File for every Boot

tb-killa, 2018-07-06:
It would be better if zigbee2mqtt would provide the logfile seperated on every Start with Timestamp in Filename.
This would be the advantage for getting easier readable logfiles (and shortend too).

On every Start we generate the logfile with Name: logfile_%timenow%.log

tb-killa, 2018-07-24:
@Koenkk We should use this: https://github.com/winstonjs/winston-daily-rotate-file
From this official module we could use the included functionality
``
filename: Filename to be used to log to. This filename can include the %DATE% placeholder which will include the formatted datePattern at that point in the filename. (default: 'winston.log.%DATE%)
``
What do you think ?

Koenkk, 2018-07-24:
Yes that would be great!

tb-killa, 2018-08-05:
Edit: Okay i think this isn´t really what we need because if i use the ``datePattern`` with Minutes we get rotation every minute. What we need is some Sort of Datepattern in FileName.
I have to look deeper for this.
But for the other point i need your help ;)

Koenkk, 2018-08-05:
I think the easiest solution would be only changing `filename: data.joinPath(`log-${Date.now()}.txt`),`.

Then we should do a automatic cleanup (e.g. only keep the 10 newest files).

 For the log directory I would put something in `configuration.yaml` like:

```yaml
advanced:
  log_directory: /var/log
```

In `logger.js` we could then do: 

```
const path = settings.get().advanced && settings.get().advanced.log_directory ? settings.get().advanced.log_directory : data

            filename: path.joinPath(`log-${Date.now()}.txt`),
```

tb-killa, 2018-08-05:
I´m use ``moment`` for formation and build some testcode like this:
``const formatefilename = 'log-' + moment(new Date()).format('YYYY-MM-DD-HH-mm-ss') + '.txt'; ``
and got this as logfilename: ``log-2018-08-05-18-01-16.txt``

With ``moment`` we could use the same format: http://momentjs.com/docs/#/displaying/format/

tb-killa, 2018-08-05:
Hmm .. there is something wrong with your ``path`` sample :(
```
TypeError: path.joinPath is not a function
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:22:28)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/mqtt.js:2:16)
    at Module._compile (module.js:652:30)
```

Koenkk, 2018-08-05:
I've implemented #207. 

The date pattern looks good! :smile:

We should keep in mind that we also need to do a cleanup (as winston wont do that anymore becuase the log file names change).

tb-killa, 2018-08-05:
Cool thanks.
Hope that the ``Pattern`` could also be implemented ?

What about implemented rudimental fs.unlink function themselve or maybe use some functionality from winston-files like this ? https://github.com/winstonjs/winston/blob/master/lib/winston/transports/file.js#L563

tb-killa, 2018-08-05:
I´v do some more tests.
inside files.js from winston transport i´v search for the maxfiles and found that we need ``dirname`` to get them working. sadly this filecount work on direct names like log.txt and not if we got some dynamic into it like we do with timestamp. To get this working we need some patching of the functions to search for files with wildcards like filename*. This could work because we give the acutally filename via variable over.

tb-killa, 2018-08-07:
Do you mean manually remove of files?

Koenkk, 2018-08-07:
@tb-killa yes

tb-killa, 2018-08-07:
Okay sounds great. Some sort of periodic triggered function (maybe with our ping function??)  who checks for the files inside our definded path and remove them via fs.unlink. This also seems that we doesnt could use any more the Winston internal functions for filesize and maxfiles.
But i think our own implementation would also be good :)

Koenkk, 2018-08-07:
Yes, I think it should be on: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/logger.js#L34 (only on startup). As we won't have more than 12 anyway (max 3 logs of current run, and max 9 old logs)

tb-killa, 2018-08-10:
@Koenkk Should we delete if files older than xx Days or xx Hours ?
I´m actually working on build some sort of remove function.

Koenkk, 2018-08-10:
I think keeping the last 10 files (and removing other ones) is good enough for now.

tb-killa, 2018-08-12:
@Koenkk Please Check my Changes on: https://github.com/tb-killa/zigbee2mqtt/tree/logfile-extended
I´v implemented dynamic logfilename and cleanup-function.
For LogfileName you could use 
``configuration.yaml``
with options like this 
```
advanced:
   log_filename: '[log-]%filenamedateformat%' 
   log_filename_date_format: 'YYYY-MM-DD.HH.mm.ss'
```
for ``log_filename_date_format`` you could use anything like http://momentjs.com/docs/#/displaying/format/ is available.
for ``log_filename`` you could build own name, it has to contain ``%filenamedateformat%`` and other letters has to be escape by ``[ ] ``.
Note: i have given the file extension (txt)!

cleanup-function is implemented by using our ``logDirectory`` and everything else .. read comments in function ;)

**I am very busy in the next few weeks, so I can only look at the project from time to time, if at all.
If you like the changes, you are welcome to provide a PR or modify and merge accordingly :)**

tb-killa, 2018-08-17:
@Koenkk nice and thanks for your work on getting it finished.
Do you need my github permission actually?
We doesnt work with Dynamic file Name?

Koenkk, 2018-08-17:
Yes I needed your permission in order to push to your branch.

The dynamic file name causes confusion.

First file looks like:

log-2018-04-03-19:30:20.txt

Second:
log-2018-04-03-19:30:20**1**.txt

Third:
log-2018-04-03-19:30:20**2**.txt

So I thought it would be better to just have a separate directory per run.

Can this be closed?

tb-killa, 2018-08-17:
Okay.
Yes close it.

Koenkk, 2018-08-17:
Ok, great, thanks for the effort you put into it!

closed 2018-08-17 by Koenkk

tb-killa, 2018-08-17:
Thank you, the honour is yours ;)