Koenkk/zigbee2mqtt #183: Connecting coordinator and router

kaelsaes, 2018-07-06:
Sorry for this noob question :-(

But how can I connect the CC2531 router to the CC2531 coordinator and what is then happening with the already paired Xiaomi devices?
Do they automatically switch to the router or do I have to manually delete the devices from coordinator and then pair with the router?

Thanks

wilfredsmit, 2018-07-06:
I also have no idea how to pair the router to the coordinator. The instruction to press SW2 for 5 seconds doesn't work for me.

kaelsaes, 2018-07-07:
After flashing the router (why don't we call it repeater) and connecting it with an USB charger the red light begins flashing every 4 seconds what means normal operation.
In FHEM automatically a new Xiaomi device is created. The model is CC2530.router (although I have an CC2531) and the type was COORD yesterday evening and is now called CHILD_RFD

So the router CC2531 connected with the coordinator CC2531.

But it seems that all Xiaomi devices are still paired with the coordinator CC2531.

I guess that I have to delete the devices in the coordinator and pair them again with the router?!?!?

Please confirm.

ciotlosm, 2018-07-11:
I assume not, because a router being in the same network should just route packages to the correct coordinator.

talondnb, 2018-07-15:
I've been testing this myself and tried to reinitiate a pairing today. I'm receiving this message in the logs (CC2531 flashed with router firmware):

```WARN No converter available for 'CC2530.ROUTER' with cid 'genOnOff' and type 'devChange'```

Also, is there a way to have a Status for the router in HA? I've added a binary_sensor into HA but when i turn off the router it still says online..

tazzz, 2018-07-22:
I am having the same issue : 3 routers (CC2531 flashed with router firmware). New device discovery near one of these routers result in
_2018-7-22 10:12:58 WARN No converter available for 'CC2530.ROUTER' with cid 'genOnOff' and type 'devChange'
2018-7-22 10:12:58 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices._

That device itself then times out:
2018-7-22 10:14:12 ERROR Cannot get the Node Descriptor of the Device: 0x00158d00022cb37d (Error: Timed out after 10000 ms)

1of16, 2018-07-24:
I got the same Problem:
```
WARN No converter available for 'CC2530.ROUTER' with cid 'genOnOff' and type 'devChange'
WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```
additional every minute there is an interesting payload:
```
INFO MQTT publish, topic: 'zigbee2mqtt/0x00<id>, payload: '{"state":false}'
```
so I guess there is some black voodoo magic missing? ;)

one of my sensors isn't answering, since it is kinda far away from the coordinator. so I guess this router isn't working as it should?!?

1of16, 2018-07-24:
@kirovilya you missed my point ;)
I am confused about the "state":false message.
I would guess, that the router doesn't work as it should, otherwise he shouldn't send "false"?!?

talondnb, 2018-07-24:
Have you enabled the router? Press sw1 so the green LED is lit. Then give it some time to update the state in HA.

talondnb, 2018-07-24:
http://ptvo.info/cc2531-based-router-firmware-136/

1of16, 2018-07-25:
@talondnb  thank you! that was a noob-mistake: I missinterpreted  "long blinks - normal operation" for the the right state. but you're right, after pressing the button the router now lights green and sends a "true".

tb-killa, 2018-07-25:
This Handling (Howto Router) should be documented on a own Page inside the Wiki!
I'm afraid this topic could often get into the discussion.

closed 2018-09-30 by stale[bot]

kiwijunglist, 2018-12-09:
Do o have to have the router led green for our to work? I thought that the green led didn't do anything it was just for testing.

h4nc, 2018-12-29:
There are no buttons on the CC2530. How do I know that my routers work fine?

chaertl1, 2019-01-12:
You don't need the button.
I think the state and the led and the button are some rests of the zigbee light example from TI.
So pressing the switch means swtich on light, the led goes on and the state goes to true.
If you want to see the state of the cc2530 you can connect a LED on pin 1_4 using a 220Ohm resistor.
I also have data activity led on pin 1_0.
The disadvantage of the state LED is that it is not always indicating a correct function of the router.

First a flashed the router with clean firmware.
Router starts blinking every second (pairing mode).
I've paired the router to a coordinator.
Router starts blinking once every three seconds (paired/connected).
I powerd off the coordinator and the router.
I only powered the router again and it started blinking once every three seconds (paired/connected)
although there was no coordinator where it could connect to.
So this feature is not working correctly.

kariofillis, 2019-04-30:
is there a way to distinguish if the messages are routed through a router, or if the connection is directly to the coordinator? i recently added a router and would like to know if i have to repair the remote devices to make them talk to the new router.