Koenkk/zigbee2mqtt #184: request unsuccess: 17

todfox, 2018-07-07:
It would be grand if this bug could be squashed. 

 {"message":"request unsuccess: 17","stack":"Error: request unsuccess: 17                                                                                                         \n    at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:46:                                                                                                         21\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:                                                                                                         257:13)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js                                                                                                         :116:13)\n    at CcZnp.emit (events.js:211:7)\n    at CcZnp._mtIncomingDataHdlr                                                                                                          (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:357:14)\n    at /opt/zigbee2m                                                                                                         qtt/node_modules/cc-znp/lib/ccznp.js:333:22\n    at _combinedTickCallback (inter                                                                                                         nal/process/next_tick.js:131:7)\n    at process._tickCallback (internal/process/                                                                                                         next_tick.js:180:9)"}

Koenkk, 2018-07-07:
Are you running the latest firmware from: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin ?

todfox, 2018-07-07:
Yup. But I can reflash it if you think that that is the reason.

Koenkk, 2018-07-07:
If you are on the latest version it is ok, does zigbee2mqtt still work after this error?

todfox, 2018-07-07:
Yes. That said, I have noticed that pairing becomes near impossible
thereafter.

todfox, 2018-07-09:
After a couple days of faffing around, I re-flashed the sniffer. All seems to be working now.

todfox, 2018-07-09:
It's back!

todfox, 2018-07-09:
2018-7-9 13:21:32 INFO
        {"message":"request unsuccess: 17","stack":"Error: request unsuccess: 17\n    at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:257:13)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:357:14)\n    at /opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:333:22\n    at _combinedTickCallback (internal/process/next_tick.js:131:7)\n    at process._tickCallback (internal/process/next_tick.js:180:9)"}

Koenkk, 2018-07-10:
Ok, lets gather more information on when this happens. 

Do you still receive update of the devices after this error?

todfox, 2018-07-16:
I have found that after flashing the device the error goes away and all is well.

Sonsi79, 2018-08-02:
Today im also getting this error during start of zigbee2mqtt. 
I have completely update zigbee2mqtt on dev branch.

Should i try to reflash again? Im on newest version of firmware

Sonsi79, 2018-08-02:
strange. during try of pair there is absolutely no message on screen or log... I have try to pair 2 new devices after start zigbee2mqtt.
Here is my debug log.

Maybe i should go back to master branch and try?
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2254802/log.txt)

Sonsi79, 2018-08-02:
Ok. I have reflash the stick and everything works. The Error message is gone and im able to pair new devices... sorry for confusing you

kirovilya, 2018-08-03:
Because it is fixed by a reflash, then I think this is due to the memory in the chip. And maybe with the number of connected devices.

tb-killa, 2018-08-03:
What the Status about move Devices to zigbee shepherd alone and have maximum of availabel memory ?

Koenkk, 2018-08-04:
@tb-killa no update on that yet, if I remember correctly we also need Z-stack 3.0 for that. #211

ryanbeaton, 2018-08-06:
@Koenkk you said you changed NWK_MAX_DEVICE_LIST=0
in issue https://github.com/Koenkk/zigbee2mqtt/issues/134 from @kirovilya find. Did that not work?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Koenkk, 2018-08-06:
@ryanbeaton no that didn't work.

Koenkk, 2018-08-13:
@cmille34 how many devices have you paired now?

cmille34, 2018-08-13:
@Koenkk 15 devices now. I can't add a 16th because i get that request unsuccess 17 error.  I was able to add these 3 devices fine yesterday , then I went to add another one and noticed it wouldn't pair.  Reflash and start over or is there any debugging/troubleshooting you would like me to do first?

Sonsi79, 2018-08-13:
I was also getting these error for a 2nd time. Reflash fixed again. 
I have 18 devices running at the Moment.

Koenkk, 2018-08-13:
Found out that error code 17 means that the buffer of the CC2531 is full. Honestly I'm a bit lost on how to fix this issue.

@jackchased do you know more about this issue?

cmille34, 2018-08-13:
@koenkk I just reflashed and repaired a bunch of devices again and after the 16th one I get the error 17.  Once I get that error 17 I have to reflash and start all over. I guess Ill repair only 15 devices and just stay there until more information comes around about why this keeps happening.

Koenkk, 2018-08-14:
Lets also see what #211 brings us.

cmille34, 2018-08-14:
I got error 17 after only pairing 13 things yesterday. I just went back to using the Xiaomi gateway for now until this matures a little bit more. I have the flashing hardware and a few cc2531 I can try when this gets a little further along.

arteck, 2018-08-14:
i have 27 devices online with error code 17...

Koenkk, 2018-08-14:
I've found some more details about this error. [The buffers can be increased in the firmware](https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/t/390286?Is-it-possible-and-how-to-increase-the-messages-buffers-in-a-CC2538-ZNP- ).

@arteck let me know if this issue still appears after trying the solution from my previous post. If not, I will compile a firmware with increased buffers.

arteck, 2018-08-15:
i come from iobroker and we use the libs  zigbee-shepherd and zigbee-shepherd-converters only.
without polling.... 
I would recommend you compile a new firmware an i can test it.. I'm looking forward to reconnect 27 devices.. or it is possible reflashing the new fiirmware without losing the devices ??

tb-killa, 2018-08-15:
Will the error also occur if you add another router (e.g. Ikea Light) to the mesh network?

arteck, 2018-08-15:
is the question for me ?? by the way i have here hue and osram router (bulbs, gu10 and plugs)

Koenkk, 2018-08-15:
@arteck firmware with increased buffer sizes: https://github.com/Koenkk/Z-Stack-firmware/tree/increased_buffer/coordinator/CC2531/bin (4 times more buffers).

arteck, 2018-08-15:
Ok ... I test it Friday

arteck, 2018-08-15:
well.. new status.. 
setting from programmer
[image]

all router devices are reconnected after OFF and then ON.. all xiaomi battery devices reconnected automatically with the routers.. only hue switches i had to reconnect manually

kirovilya, 2018-08-15:
@arteck It is very good!
Interesting, if there were sensors connected to the coordinator, would they work or should they be repairing?

kirovilya, 2018-08-15:
The important part of the flashing it is to save ieee address of coordinator!

Koenkk, 2018-08-15:
28 devices! :smile: lets monitor this for a few days

arteck, 2018-08-15:
I still have here 4 router devices..  :-)
until now not connectd

arteck, 2018-08-16:
problem
[image]

Koenkk, 2018-08-16:
@arteck could this be a caused due to an offline router/connection problems?

arteck, 2018-08-16:
nope all routers are online

arteck, 2018-08-16:
and i have pairing problem with the new firmware and xiaomi NEW devices.. old devicess works

arteck, 2018-08-16:
hmm all 31 devices works... 
[image]

Sonsi79, 2018-08-16:
@arteck Sehe gerade du nutzt die Lampen als Router. Wie kann ich das einstellen? Finde deine Lösung genial :-) Habe mehrere HUE Lampen und würde diese gerne genauso nutzen wie Du

arteck, 2018-08-16:
die router werden automatisch erzegt.. es sind alle stromgebundenen Geräte wie hue Lampen oder Stripes sowie Osram Plugs oder auch hier Lampen egal ob gu10 oder e27 oder oder.. alles was an Strom hängt wird zur Router.. Zigbee spannt das Netz dann automatisch auf

Sonsi79, 2018-08-16:
Ok, danke für die Antwort. Eine letzte Frage, wie hast Du dir diese tolle Übersicht anzeigen lassen in der auch sichtbar ist welches Gerät an welchem Router hängt?

arteck, 2018-08-16:
die Map wird durch den zigbee Adapter aus iobroker generiert

Koenkk, 2018-08-16:
Ok great, so the firmware is still working OK 😄

arteck, 2018-08-18:
i think we have a problem

zigbee.0 | 2018-08-18 12:08:08.594 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - off - {} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the n
zigbee.0 | 2018-08-18 12:08:07.640 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - on - {} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the ne
zigbee.0 | 2018-08-18 12:08:01.597 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - off - {} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the n
zigbee.0 | 2018-08-18 12:07:47.930 | info | new dev 0x841826000001170d 397 PAR16 50 TW
zigbee.0 | 2018-08-18 12:07:31.991 | error | Zigbee publish to '0x0017880102de40ac', lightingColorCtrl - moveToColorTemp - {"colortemp":10,"transtime":0} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Pl
zigbee.0 | 2018-08-18 12:07:31.887 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - on - {} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the ne
zigbee.0 | 2018-08-18 12:07:22.559 | error | Zigbee publish to '0x7cb03eaa00a747b1', genOnOff - on - {} - 3 failed with error Error: AF data request fails, status code: 183. APS no ack.
zigbee.0 | 2018-08-18 12:07:15.315 | error | Cannot get the Node Descriptor of the Device: 0x0017880102ca68f3 (Error: Timed out after 10000 ms)
zigbee.0 | 2018-08-18 12:07:14.692 | info | Succesfully configured 0x7cb03eaa00a747b1
zigbee.0 | 2018-08-18 12:07:05.326 | error | Zigbee publish to '0x0017880102de40ac', lightingColorCtrl - moveToColorTemp - {"colortemp":10,"transtime":0} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Pl
zigbee.0 | 2018-08-18 12:07:05.291 | error | Cannot get the Node Descriptor of the Device: 0x7cb03eaa00a747b1 (Error: request unsuccess: 16)
zigbee.0 | 2018-08-18 12:07:05.248 | error | Cannot get the Node Descriptor of the Device: 0x0017880103dcc9d2 (Error: request unsuccess: 16)
zigbee.0 | 2018-08-18 12:07:05.203 | error | Cannot get the Node Descriptor of the Device: 0x0017880103648612 (Error: request unsuccess: 16)
zigbee.0 | 2018-08-18 12:07:05.123 | error | Cannot get the Node Descriptor of the Device: 0x0017880102f054a7 (Error: request unsuccess: 16)
zigbee.0 | 2018-08-18 12:07:05.014 | error | Cannot get the Node Descriptor of the Device: 0x84182600000b911f (Error: request unsuccess: 16)
zigbee.0 | 2018-08-18 12:07:04.991 | error | Failed to configure 0x0017880103caa673
zigbee.0 | 2018-08-18 12:07:04.970 | error | Failed to configure 0x0017880102e251de
zigbee.0 | 2018-08-18 12:07:04.949 | error | Zigbee publish to '0x0017880103648612', genOnOff - off - {} - 11 failed with error Error: rsp error: 16
zigbee.0 | 2018-08-18 12:07:04.928 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - off - {} - 11 failed with error Error: rsp error: 16
zigbee.0 | 2018-08-18 12:07:04.905 | error | Zigbee publish to '0x0017880102de40ac', lightingColorCtrl - moveToColorTemp - {"colortemp":10,"transtime":0} - 11 failed with error Error: rsp error: 16
zigbee.0 | 2018-08-18 12:07:04.883 | error | Zigbee publish to '0x0017880103648612', lightingColorCtrl - moveToColorTemp - {"colortemp":10,"transtime":0} - 11 failed with error Error: rsp error: 16
zigbee.0 | 2018-08-18 12:07:04.859 | error | Zigbee publish to '0x0017880103648612', genOnOff - on - {} - 11 failed with error Error: rsp error: 16
zigbee.0 | 2018-08-18 12:07:04.837 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - on - {} - 11 failed with error Error: rsp error: 16
zigbee.0 | 2018-08-18 12:07:04.743 | error | Zigbee publish to '0x0017880102de40ac', genOnOff - on - {} - 11 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the ne
zigbee.0 | 2018-08-18 12:07:04.662 | info | Succesfully configured 0x84182600000c8b2f
zigbee.0 | 2018-08-18 12:07:04.355 | error | Failed to configure 0x84182600000e4d12
zigbee.0 | 2018-08-18 12:06:50.107 | error | Cannot get the Node Descriptor of the Device: 0x84182600000e4d12 (Error: Timed out after 10000 ms)

arteck, 2018-08-19:
1 error found,, its a hue switch
[image]

but it works.. the other one is ok..

tb-killa, 2018-08-20:
This could be the fault of Power save (sleep Mode).
Please check if you pressed All four Buttons simultan short, this could bring device up and finish your fault check.

arteck, 2018-08-20:
press one Button help also..

matchett808, 2018-08-20:
Prior to the new firmware - was it producing error messages like this one?

```
2018-8-20 12:31:57 - error: Zigbee publish to '0x001e5e0902293394', genOnOff - off - {} - null failed with error Error: AF data request fails, status code: 183. APS no ack.    
```

My stick seems to get itself into a state where I can't recover from occasionally (it normally coincides with me messing about with new devices) - I get the error 17 and then things work for 5 minutes before all the devices stop working.

I'm trying to not go for new firmware as I'd need to repair all my devices - but if I have to I'll just wait for the NV_RESTORE firmware (or for it to be ruled out)

matchett808, 2018-08-20:
That's just a plug

arteck, 2018-08-20:
which plug ?? hue ?? osram ??

dzungpv, 2018-08-27:
I have this problem, try all type of firmware from original TI ZNP to the increase buffer size. One reason for this problem is when signal low, for me it often occurs if the signal lower than -60dbm

Koenkk, 2018-08-27:
@dzungpv do you have this issue with a CC2531 flashed with https://github.com/Koenkk/Z-Stack-firmware/tree/increased_buffer/coordinator/CC2531/bin ?

dzungpv, 2018-08-27:
@Koenkk yes, it has some more bug like it hang and must reboot PI, I don't know it is zigbee2mqtt or the hardware/firmware

worf0r, 2018-08-28:
With the iobroker solution error 17 never occurs with the first command sent to a lightbulb. But very often with the second or third command. Even with good signal...

arteck, 2018-08-29:
only combined with TRADFRI bulb ... did you forget

worf0r, 2018-08-29:
Not only Tradfri, also Osram, see here: https://forum.iobroker.net/viewtopic.php?f=36&t=14819&start=80#p171563

arteck, 2018-08-29:
Osram Stripe maybe, only.... my Osram Bulb E14 works..

[12 more replies]

closed 2018-11-09 by Koenkk