Koenkk/zigbee2mqtt #185: Xiaomi RTCGQ11LM devices not updating after some time

cmille34, 2018-07-07:
I've been tinkering with this for a few weeks in an effort to get rid of using the Xiaomi gateway.    I have everything working for the most part but after some restarts, I am noticing that Xiaomi RTCGQ11LM and MCCGQ11LM  devices will intermittently stop updating sometimes. I thought it might be a range issue but one of the motion sensors that is only 25 feet away in the other room is not longer updating now and it had been working the most reliably for the past couple weeks. 

If I bring the sensor close to the flashed CC2531 stick running in a docker container in my office here and re-pair it, it will likely work for a while but within a week after some reboots I expect it to stop reporting again.  I have (3) different RTCQ11LM devices that are doing this now.  I don't want to have to re-pair them every week if possible.  Is there something I can do to help troubleshoot this?  Below is a beginning of my latest log file to help provide a better picture of my setup.  Currently I am using the builtin MQTT broker in Home Assistant.

> 2018-07-07T14:40:10.434Z - info: Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-07-07T14:40:10.441Z - info: Starting zigbee-shepherd
2018-07-07T14:40:10.906Z - info: zigbee-shepherd started
2018-07-07T14:40:10.910Z - info: Currently 19 devices are joined:
2018-07-07T14:40:10.911Z - info: attic (0x00158d000201c2ce): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-07-07T14:40:10.911Z - info: chrisoffice (0x00158d0001e7c471): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-07-07T14:40:10.911Z - info: ericaoffice (0x00158d0001fa510c): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-07-07T14:40:10.912Z - info: bathroomdoor (0x00158d00022c7c23): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
2018-07-07T14:40:10.912Z - info: downstairsbathroommotion (0x00158d00022788b9): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
2018-07-07T14:40:10.912Z - info: laundryroommotion (0x00158d0002013370): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-07-07T14:40:10.912Z - info: loftsensor (0x00158d000201c204): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-07-07T14:40:10.913Z - info: parentsbedroom (0x00158d000201a06b): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-07-07T14:40:10.913Z - info: stairwaybutton (0x00158d00016bdf39): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2018-07-07T14:40:10.913Z - info: laundryroomdoor (0x00158d0002019aa9): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
2018-07-07T14:40:10.913Z - info: chrisofficemotion (0x00158d000200cb1d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-07-07T14:40:10.914Z - info: kitchenmotion (0x00158d0001f9d541): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-07-07T14:40:10.914Z - info: tybutton (0x00158d0001f3ae71): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2018-07-07T14:40:10.914Z - info: ericabutton (0x00158d0001f3ae74): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2018-07-07T14:40:10.915Z - info: waterheaterleaksensor (0x00158d0002112505): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
2018-07-07T14:40:10.915Z - info: frontdoor (0x00158d0001bf47fc): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-07-07T14:40:10.915Z - info: parentscloset (0x00158d0001e5ce2d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-07-07T14:40:10.916Z - info: ericachair (0x00158d000201b8cd): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
2018-07-07T14:40:10.916Z - info: ericaofficemotion (0x00158d0001ff3f23): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
2018-07-07T14:40:10.917Z - info: Connecting to MQTT server at mqtt://192.168.1.104:1883
2018-07-07T14:40:10.972Z - info: zigbee-shepherd ready
2018-07-07T14:40:11.835Z - info: Connected to MQTT server
2018-07-07T14:40:11.837Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'

Any help you can provide to help get to the bottom of this would be appreciated.  I apologize if I left out any crucial information to help troubleshoot.

cmille34, 2018-07-07:
In case it helps, I went ahead and just tried to repair one of the problem devices and this is what it shows in the log when I repair it:

> 2018-7-7 15:56:28 [32mINFO[39m Zigbee: allowing new devices to join.
spinlock: false []
device already in network
2018-7-7 15:56:43 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/kitchenmotion', payload: '{"battery":"100.00","voltage":3045}'
2018-7-7 15:56:45 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/kitchenmotion', payload: '{"battery":"100.00","voltage":3045,"illuminance":9}'
2018-7-7 15:56:45 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/kitchenmotion', payload: '{"battery":"100.00","voltage":3045,"illuminance":9,"occupancy":true}'

After doing this it starts to report again.  I hope I can just avoid having to do this over and over.

Koenkk, 2018-07-07:
Are you running the latest firmware from: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin?

cmille34, 2018-07-08:
@Koenkk I don't know Im afraid. I bought it pre-flashed from someone on reddit. Is there a way I can check the version from the ubuntu host I have it plugged into?

Koenkk, 2018-07-09:
At the moment not, when did you order the stick?

cmille34, 2018-07-09:
I ordered it on June 15th, so it would have been flashed sometime between June 1st and June 15th.  I went ahead and ordered hardware to flash it again.  should be here in a couple weeks from china.  ill keep you posted if that helps.   Thanks for all your time and efforts making thus great solution.

Sonsi79, 2018-07-29:
I have same problem with newest Version of your firmware and the RTCGQ11LM. After Restart zigbee2mqtt both of my 2 RTCGQ11LM Motion Sensor didn't update the state. I have to push the button one time or pair again.
All other Xiaomi devices working fine after restart but the RTCGQ11LM didn't.

Koenkk, 2018-07-31:
@Sonsi79 can you check the battery of your 2 motion sensors.

I tried reproducing by (on zigbee2mqtt 0.1.2 and last CC2531 firmware).
- Pair RTCGQ11LM
- Reports motion and illuminance
- Restart zigbee2mqtt
- Still reports motion and illuminance

Sonsi79, 2018-07-31:
@Koenkk it seems you're right. It seems to work now.
I have installed zigbee2mqtt two days before, but i have 0.1.1. How to install the 0.1.2? Is it dev branch?

Koenkk, 2018-08-01:
@Sonsi79 you can update using: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version

123trs, 2018-08-01:
Same issue here. I have 5 RTCGQ11LM. 2 pices worked very well. But the other 3 ones lost the connection time for time. I use the problemematic 3 ones for 2 weeks now. The othes runs about for 10 weeks without problems. Is it possible, that there are different revisions on the market? But all looks equal.

I use the 0.1.1 version, and the latest firmware on the CC2531 stick.

Koenkk, 2018-08-01:
@123trs have you also tried replacing the batteries?

cmille34, 2018-08-01:
For those of us having problems, is range an issue at all? How many devices do you have joined to your stick?  All the devices in my home that are having the problems are fairly far away from my CC2531 stick,  Its strange that they work for a while then stop working .  I hope to have my programming hardware in the next few days so I can flash another CC2531 stick to act as a router upstairs to extend the range.

Currently i have 20 devices joined to my single CC2531 stick, I wonder if its getting swamped?  I'm currently having problems with the following devices:

WSDCGQ11LM
RTCGQ11LM
MCCGQ11LM

If I rejoin them they will work for a couple days then just stop publishing their status.  I was hoping just flashing to the latest firmware would be the fix but it sounds like it's more then that.

123trs, 2018-08-01:
@Koenkk : Yes, batteries brand new, good Varta

123trs, 2018-08-01:
@cmille34 : I dont think, the range is the problem. For test, I place on in 50cm range. It works for about 24h. After that, same problem. I paired 5 sensors of that.

clockbrain, 2018-08-02:
For those with this problem do you have permit_join set to true or false?

With permit_join set to false I was also having heaps of problems with devices not responding after some time and needing to reboot daily and reflash the firmware weekly. I even started from scratch with a new image on my Pi and installing zigbee2mqtt again but still had same problem.

 I changed permit_join back to true and am finding the whole system is much more stable.

123trs, 2018-08-02:
"permit_join" ist set to "true" from the beginning. I never changed it.

123trs, 2018-08-02:
Again no responding from device 0x00158d000204aab1. New paired jesterday 16:30. Last response:

state | no_motion | 2018-08-02 07:19:27

Sonsi79, 2018-08-02:
My Setting was set to „false“ i have changed it and will try now. Thank for your hint.

cmille34, 2018-08-02:
My setting was set to false. I was using the toggle switch in Home Assistant to allow things to join when I needed.  Ill change it to allow join = true all the time and see if that makes a difference.

cmille34, 2018-08-04:
I finally received my flashing hardware and additional 2531 boards. I flashed the router firmware onto a 2531 and joined it to my zigbee2mqtt docker instance and have it plugged in upstairs to eliminate any chance of range being an issue. I then went ahead and re-flashed my coordinator 2531 stick to the firmware here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin  to make sure I am on the latest.

I then went through the trouble of removing and re-pairing all 20 devices.  I noticed one of the devices wasn't reporting afterwards so I tried to remove it again and re-pair it and it wouldn't go through the joining process at all. I looked at the logs and found:
"message":"request unsuccess: 17","stack":"Error: request unsuccess: 17
    at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21

I checked on Github here and saw the fix of reflashing the firmware again, so I went ahead and did that and now I can pair that one pesky Xiaomi temp sensor.  An hour after flashing, most of the devices I paired prior to reflashing the stick aren't reporting though, only 3 of the 20 devices are. They all show up in the log at boot up of being pared but no updates are being received it seems.  

When I flash firmware to the stick do I have to repair the devices everytime?
Is there anyway to force the Xiaomi sensors to check back in with the zigbee2mqtt or force the zigbee2mqtt to contact the sensors?

I really don't want to have to keep re-pairing these devices.

Koenkk, 2018-08-05:
1. Yes, clear the complete device section.
2. Yes
3. No (not yet)

cmille34, 2018-08-06:
@Koenkk OK - thanks for the guidance.  I've gone ahead and reflashed the stick 1 more time, deleted the database and cleared out the devices and repaired the 11 most important devices for me.  If I run into any trouble of devices no longer reporting in the zigbee2mqtt log I will post back here.

Is the C2531 not a good stick? Is there a stick with more memory that handles many devices better?

Koenkk, 2018-08-06:
@cmille34 CC2531 has limited memory, but there are no other options :(

kirovilya, 2018-08-06:
@cmille34 cc2531 has old chip with limited memory (8K). On TI forum recomended CC26x2 or CC2538 chip https://e2e.ti.com/support/wireless_connectivity/proprietary_2-4_ghz/f/964/t/633997

I try to find usb-dongle with this chip and found only this https://www.alibaba.com/product-detail/CC2538-CC2592-USB-Dongle-ZigBee-adapter_60426940995.htm but I can not buy it :(

ryanbeaton, 2018-08-06:
@kirovilya how about this one
http://s.aliexpress.com/iQNzUnIn

The U2ZB-3219 option

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

kirovilya, 2018-08-06:
@ryanbeaton I bought and wait this board https://www.aliexpress.com/item/CC2538-CC2592-zigbee-6-lowpan/32879578178.html

cmille34, 2018-08-06:
Thanks for all the additional information and for all the countless hours you guys put into this solution on your own time!  If I end up needing to order a newer chip from China and flashing it to allow more devices I certainly will order it.  

Is anyone else running a 2531 stick with 20+ end devices?  Ill stick with just the 11 devices I have right now and see if things continue to report reliably for a couple weeks.  @Koenkk if there are any other specific tests you would like me to run to help troubleshoot anything please just let me know.  
Thanks again

kirovilya, 2018-08-06:
@cmille34 Even on cc2531 you can obtain more than 30 end-devices in network - you need to connect the end-devices not to the coordinator (cc2531), but through routers (devices with a constant power supply). In this case, you can get 20 devices on the coordinator + 20 devices on each router.

So I want to see how the devices are connected to each other and start pairing on a specific router #252

cmille34, 2018-08-06:
@kirovilya I have been able to pair 21 devices on my 2531 but it only takes a few days before devices stop reporting in in the zigbee2mqtt log.  There doesn't appear to be an obvious cause for it either, I thought range might be an issue but one of my farthest devices was working last time but then a motion sensor very close just stopped reporting in and I was forced to remove it and repair it.  

I was hoping to see if anyone else out there was reliably using 20+ Xiaomi devices for an extended period of time?

cmille34, 2018-08-06:
@123trs How many devices do you have joined to your CC2531?

123trs, 2018-08-06:
Hello,
I have joined:
11  WSDCGQ01LM
5     RTCGQ11LM
1     WXKG03LM
1     MCCGQ01LM

Greets

Koenkk, 2018-08-07:
@cmille34 my network contains +-20 xiaomi devices and 5 bulbs (innr/hue/tradfri), working stable for months.

yfaway, 2018-09-29:
Hi guys,

I am wondering if there is any recent update on this issue. I am using CC2531, and recently two Xiaomi motion sensors and a temperature sensor just stop sending data. I try to bring it close to the CC2531 to rejoin, but it doesn't work. Nothing shows up in the zigbee2mqtt log. The motion sensors have been working for weeks. It's not a range issue since it is quite close to the CC2531. I have a total of 8 Xiaomi devices in my network.

Thanks,

yfaway, 2018-09-29:
I got them working again. It seems like the CC2531 is in a bad state somehow. I unplug it and then plug in again (without restarting the Pi), and re-pair. All is fine afterward. The only problem is that the device interface moves from ACM0 to ACM1. Let's see how long the Xiaomi devices remain in the network.

nikolaykasyanov, 2018-09-29:
I have a somewhat similar issue with RTCGQ11LM. First of all, it had been quite hard to pair it, then it was sending illuminance data with a noticeable delay (like, switch a light on, get increased illuminance in 10 minutes) but I ignored it because occupancy was working pretty well. Sometimes it simply stops reporting occupancy for some time and then gets back to normal on its own.

Could it be the battery? I'm getting `"battery":"100.00","voltage":3095` in response though.

Not sure how related it is to this issue, please let me know if it deserves another one.

Koenkk, 2018-09-29:
Could you publish your zigbee networkmap? https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgenetworkmap

nikolaykasyanov, 2018-09-30:
@Koenkk sure, I edited labels a bit though to have devices names here.

`0x000b57fffe2eb83c` is an IKEA dimer that I experimented with but then removed from zigbee2mqtt network using the respective MQTT command, but somehow it still resides in the network map.

```graphviz
digraph G {
  "0x00158d0001fa6f1b" [label="WSDCGQ11LM 1 (online)"];
  "0x00158d0001fa6f1b" -> "0x00124b0014bdeb51" [label="126"]
  "0x00158d0002049ded" [label="MCCGQ11LM (online)"];
  "0x00158d0002049ded" -> "0x00124b0014bdeb51" [label="63"]
  "0x00158d0002349639" [label="SJCGQ11LM (online)"];
  "0x00158d0002349639" -> "0x00124b0014bdeb51" [label="6"]
  "0x7cb03eaa00ad3563" [label="AB3257001NJ 1 (online)"];
  "0x7cb03eaa00ad3563" -> "0x84182600000d2ca1" [label="34"]
  "0x90fd9ffffe8287ab" [label="LED1536G5 (online)"];
  "0x90fd9ffffe8287ab" -> "0x84182600000d2ca1" [label="24"]
  "0x84182600000d2ca1" [label="AB3257001NJ 2 (online)"];
  "0x84182600000d2ca1" -> "0x90fd9ffffe8287ab" [label="40"]
  "0x000b57fffe2eb83c" [label="0x000b57fffe2eb83c (offline)"];
  "0x000b57fffe2eb83c" -> "0x7cb03eaa00ad3563" [label="16"]
  "0x00158d000245db08" [label="WSDCGQ11LM 2 (online)"];
  "0x00158d000245db08" -> "0x7cb03eaa00ad3563" [label="52"]
  "0x00158d0002437729" [label="WSDCGQ11LM 3 (online)"];
  "0x00158d0002437729" -> "0x7cb03eaa00ad3563" [label="96"]
  "0x00124b0014bdeb51" [label="0x00124b0014bdeb51 (online)"];
  "0x00124b0014bdeb51" -> "0x84182600000d2ca1" [label="68"]
  "0x00158d0002b490f2" [label="RTCGQ11LM (online)"];
  "0x00158d0002b490f2" -> "0x84182600000d2ca1" [label="48"]
}
```

[image]

SuperTux101, 2018-10-08:
Hi all, I like to join this discussion. I am testing the zigbee2mqtt for some time and noticed the same problem. I have the door sensor and the temperature sensor. It seems that the door sensor is immediately working again after a startup. The temperature sensor, however, seems to quit reporting. It becomes active again after short pressing the button on the device.  (no repairing, but just a short press). Is it possible that the device is in a sleep state and needs to be awakened? Maybe this is a mechanism to save some energy of the battery. To be sure, sniffing the traffic between the sensor and the original hub could provide some insight.

tb205gti, 2018-10-11:
Hmm I have the same issue with amotion sensor - completely dead, not reporting. Tried removing it and re-pairing it. Pairs fine, but does not send any data. :/ Poor Xiaomi quality or zigbee2mqtt issue?

SuperTux101, 2018-10-11:
I don't think Xioami delivers poor quality. If the original bridge is used, reporting does not stop. I've seen that Openhab has an addon for Zigbee, able to use the CC2531 (using the same firmware). It works and devices are recognized. It's maybe a good idea to test if reporting continues on OH.

Koenkk, 2018-10-12:
I thought that maybe this issue happens when the RTCGQ11LM is not directly connected to the coordinator. For those having these problems.

- Could you publish your zigbee network map
- If it's not directly connected to the coordinator, try repairing it when having it close to the coordinator (a few cm's).
- Check you zigbee network map again to see if it's directly connected now.

yfaway, 2018-10-14:
I think a lot of this is related to the range issue. If it is near the
edge, it might work for sometimes, and then one day it just stops. This
happens to one of my Xiaomi temperature sensor and couple of my Xiaomi
motion sensors. The only way to fix it it to repair again. I also find that
sometimes you might have to restart the Rasp PI board, especially if the
board is used for development as well, whereby the stick or other sticks
such as ZWave are plugged and unplugged often.

What I don't know is the recovery process in Zigbee protocol. Sometimes the
device can't reach the coordinator purely due to environment factor that
hinder the wireless range. If it tries again later, then it might work
again. But perhaps to conserve battery, the sensor just goes into sleep
mode immediately and won't try again.

yfaway, 2018-10-14:
I am using OpenHab as well; however, AFAIK, the Zigbee addon for OpenHab
has a lot of issues with Xiaomi devices. Chris, the addon developer,
believes that Xiaomi devices aren't fully Zigbee compliant. I jumped
directly to Zigbee2mqtt after doing the research. And for the most part,
it's quite stable. Several of my sensors do kind of stop communicating over
time, but I think it is a range issue. I am ordering another CC2531 to use
as a router.

yfaway, 2018-10-18:
I see couple of interesting log entries today. One of the Xiaomi motion
sensors on my 2nd floor stopped triggering today. This is the one farthest
away from the Rasp Pi containing the sniffer. I took it off and re-paired
right besides the Pi. But still no mqtt message is sent. But the sensor
does work. Every time I waved my hand in front of it, I get 3 log entries:

Oct 17 22:39:41 raspberrypi npm[5241]: 2018-10-17 22:39:41 DEBUG Recieved
zigbee message with data
{"cid":"msIlluminanceMeasurement","data":{"measuredValue":9}}
Oct 17 22:39:41 raspberrypi npm[5241]: 2018-10-17 22:39:41 DEBUG Recieved
zigbee message with data
{"cid":"msIlluminanceMeasurement","data":{"measuredValue":9}}
Oct 17 22:39:41 raspberrypi npm[5241]: 2018-10-17 22:39:41 DEBUG Recieved
zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}

Notice that no device id displayed in the log. So the sensor is working,
and zigbee2mqtt does receive the message, but it somehow doesn't know the
device or lost track of the device.

yfaway, 2018-10-18:
I restarted the device, and noticed in the log that that particular sensor
is in Unknown state. This is strange because it has been working in the
last couple of days. I had to remove it from zigbee2mqtt database using the
command "mosquitto_pub -d -t zigbee2mqtt/bridge/config/remove -m 0x..." and
re-pair again. It works afterward.

Does anyone know under what condition would zigbee2mqtt changes a known and
work device to "Unknown"?

Koenkk, 2018-10-18:
@yfaway could you post your zigbee2mqtt startup log?

yfaway, 2018-10-19:
Koen,

I am attaching the log containing the last timestamp the sensor was
working fine. The log then contains various attemps to re-pair, a restart,
a device removal, and then the final re-pair that get it back to working
state. The sensor id is: 0x00158d000200eb10.

At some point, I see this message in the log:
Oct 17 22:18:20 raspberrypi npm[5241]: 2018-10-17 22:18:20 DEBUG Recieved
zigbee message with data "0x00158d000200eb10"
Oct 17 22:18:21 raspberrypi npm[5241]: 2018-10-17 22:18:20 WARN Message
without device!
Oct 17 22:18:24 raspberrypi npm[5241]: spinlock: true []

After the restart:
Oct 17 22:42:22 raspberrypi npm[27439]: 2018-10-17 22:42:22 INFO Starting
zigbee2mqtt version 0.0.0 (commit #2ba2864)
Oct 17 22:42:22 raspberrypi npm[27439]: 2018-10-17 22:42:22 INFO Starting
zigbee-shepherd
Oct 17 22:42:25 raspberrypi npm[27439]: 2018-10-17 22:42:25 INFO
zigbee-shepherd started
Oct 17 22:42:25 raspberrypi npm[27439]: 2018-10-17 22:42:25 INFO Currently
8 devices are joined:
Oct 17 22:42:25 raspberrypi npm[27439]: 2018-10-17 22:42:25 INFO
0x00158d0001fa45fd (0x00158d0001fa45fd): RTCGQ11LM - Xiaomi Aqara human
body movement and illuminance sensor (EndDevice)
Oct 17 22:42:25 raspberrypi npm[27439]: 2018-10-17 22:42:25 INFO
0x00158d000200ed5e (0x00158d000200ed5e): RTCGQ11LM - Xiaomi Aqara human
body movement and illuminance sensor (EndDevice)
Oct 17 22:42:25 raspberrypi npm[27439]: 2018-10-17 22:42:25 INFO
0x00158d000236b6c0 (0x00158d000236b6c0): RTCGQ11LM - Xiaomi Aqara human
body movement and illuminance sensor (EndDevice)
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:25 INFO
0x00158d000236b4cd (0x00158d000236b4cd): RTCGQ11LM - Xiaomi Aqara human
body movement and illuminance sensor (EndDevice)
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 INFO
0x00158d000204c7ef (0x00158d000204c7ef): RTCGQ11LM - Xiaomi Aqara human
body movement and illuminance sensor (EndDevice)
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 INFO
0x00158d000245f049 (0x00158d000245f049): WSDCGQ11LM - Xiaomi Aqara
temperature, humidity and pressure sensor (EndDevice)
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 INFO
0x00158d000239725a (0x00158d000239725a): RTCGQ11LM - Xiaomi Aqara human
body movement and illuminance sensor (EndDevice)
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 INFO
0x00158d000200eb10 (0x00158d000200eb10): unkown - undefined unknown
(EndDevice)
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 WARN
`permit_join` set to  `true` in configuration.yaml.
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 WARN Allowing
new devices to join.
Oct 17 22:42:26 raspberrypi npm[27439]: 2018-10-17 22:42:26 WARN Set
`permit_join` to `false` once you joined all devices.

Note the very last device: undefined unknown (EndDevice)

Thank you for looking at this,

Koenkk, 2018-10-19:
Can you stop zigbee2mqtt, remove all entires of `0x00158d000200eb10` from the `database.db` start zigbee2mqtt and pair the sensor?

tb205gti, 2018-11-11:
Funny I do have some of the same issue with the 0.1.8 version and the motion sensor. It works a few times, then it stops updating the state from the sensor.

reverting to 0.1.6 solves it, then everything is working again.

enboig, 2018-11-11:
I have the same error; when trying to rejoin I get:

nov 11 22:12:37 servidor7 npm[3930]:   zigbee2mqtt:error 2018-11-11 22:12:37 Cannot get the Node Descriptor of the Device: 0x00158d000201358f (Error: Timed out after 10000 ms)

I cannot get it rejoined.
I am using 0.1.6

Koenkk, 2018-11-12:
@enboig make sure you correctly follow the pairing procedure from: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi

enboig, 2018-11-12:
After upgrading to 0.1.8 (maybe not necessary) and deleting database.db I could join it. Does deleting database.db remove all the devices? is there a "less drastic" way to remove one device by id?

Koenkk, 2018-11-12:
@enboig you can also remove a single entry from it.

enboig, 2018-11-12:
The database is looks quite simple, to remove an entry I could use a cat & grep, is there a preferred way?

I have a motion sensor and two temperature sensors, yesterday:
1) I removed my database.db
2) Synced my motion sensor and checked it was working (didn't check the temperatures ones)
3) At night my server crashed
4) I booted it, and motion sensor is working, but temp senors are missing (don't know if they were working before the crash).
This is related to removing database.db, or is another problem?

lolorc, 2018-11-12:
> @ryanbeaton I bought and wait this board https://www.aliexpress.com/item/CC2538-CC2592-zigbee-6-lowpan/32879578178.html

Hi,

What are you using to program the cc2538 ? does it come with SBL ?
does it need something like XDS110/XDS100 to program it ?

closed 2019-01-18 by stale[bot]

djpollum, 2019-02-05:
I would like to add a comment to this topic.
I bought the equipment and installed it as described in the documentation. Then I bought a Xiaomi aqara Hub and motion sensors. 
At first the installation went fine, but the motion sensor soon gave no updates after some time.
I thought this was a bug and tried to reinstall the motion sensor. All got soon very worse...
Nothing worked (Zigbee2mqtt saw no new devices) and I tried everything.

- reinstalling Zigbee2mqtt v1.0
- installing newest version of Zigbee2mqtt v1.1.1
- reflashing CC2531 with old and new firmware (CC2531ZNP-Prod_20181024 and CC2531ZNP-Prod_20190109)
- re-pairing 100x times (also 2 cm close to the CC2531)
- reset button on the CC2531
- etc. 

Then I saw a line somewhere that the Xiaomi devices do not lose their connection with a paired coordinator / router.  This was the solution which I was looking for. 
I re-paired the Xiaomi motion sensor to the original Xiaomi Aqara Hub, so the sensor loses it's connection to the CC2531 in it's "memory".
After successful pairing with the original Hub I was able to re-pair the motion sensor to the CC2531 !

I guess you can put this solution in your documentation also?

Koenkk, 2019-02-06:
@djpollum did you reflash without the `Retain IEEE address checked`?

djpollum, 2019-02-06:
> 
> 
> @djpollum did you reflash without the `Retain IEEE address checked`?

Yes @Koenkk , I did exactly the same as shown in the picture on https://koenkk.github.io/zigbee2mqtt/getting_started/flashing_the_cc2531.html.

Koenkk, 2019-02-07:
@djpollum at the time trying to pair the motion sensor to zigbee2mqtt, was the hub still running?

djpollum, 2019-02-07:
> 
> 
> @djpollum at the time trying to pair the motion sensor to zigbee2mqtt, was the hub still running?

@Koenkk no, I thought these would interfere with each other. So before pairing to zigbee2mqtt I removed the hub from the wall socket.

[7 more replies]