Koenkk/zigbee2mqtt #192: Add delay to execute

tb-killa, 2018-07-11:
We should add configurable delay inside the execute function of the zigbee converter to give end device time for finish.

Could be fix some Bugs where reporting doesnt work.

Koenkk, 2018-07-12:
Correct me if I'm wrong, but this is already possible via:

```js
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => setTimeout(() => cb(false), 3000),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];
```

tb-killa, 2018-07-12:
Yes, but if we Integrate it into function it would be easier and work for every cb.
But maybe you have another simple idea :)?

Koenkk, 2018-07-14:
The default delay is now 300ms. Can this be closed?

tb-killa, 2018-07-14:
Looks great, work great.
I remove the manually ```(cb) => setTimeout(() => cb(false), 3000),``` and let the new function work.

closed 2018-07-14 by tb-killa