Koenkk/zigbee2mqtt #193: Could not open secific file

jonydebu, 2018-07-13:
Hi , all goes well until last step . When i run the "perform action" the CC2531 greenled turns off and Smart Flash programm indicate "could not open sepecif file" . TI led Debugger is green .
[image]
Some idea ? Thanks.

Koenkk, 2018-07-13:
What are the contents of the .hex file? (open it with e.g. notepad)

jonydebu, 2018-07-13:
Hi a very long file (I think too large to paste here) . I´ve download it from here : 
[image]

jonydebu, 2018-07-13:
oops you´re the owner , great job and thanks for your help.

jonydebu, 2018-07-13:
Hi , i´ve install this one from this repository and this one seems to flash well .
[image]
Thanks 4 all .

closed 2018-07-13 by jonydebu