Koenkk/zigbee2mqtt #195: TRADFRI bulb E27 W opal 1000lm

jonydebu, 2018-07-14:
Hi , cc2531 seems works well , but cant find the Ikea tradfri bulb . Dont think it´s a distance pairing (is a 5cms) . In HA just configure discovery : true 
[image]
Apreciate any help , thanks .

Koenkk, 2018-07-14:
So which firmware did you flash? (asking because of #193)

jonydebu, 2018-07-14:
[image]
This One .

Koenkk, 2018-07-14:
Can you try running with `DEBUG=* npm start`

jonydebu, 2018-07-14:
Hi Koenkk ,Hi, I've tried it but nothing's happening yet. Is there any way to check if the zigbee2mqtt CC2531 coordinator is working?

Koenkk, 2018-07-15:
Can you pair other devices?

jonydebu, 2018-07-17:
Hi , i buy another Ikea Tradfri model(LED1650R5) and it pairing well . 
the problem seems to be in the bulb that for some reason according to this:           
( https://github.com/athombv/com.ikea.tradfri/issues/4  ) , need a dimmer .

jonydebu, 2018-07-17:
Finally I´ve got it . Just running ```DEBUG=*npm start``` and the bulb goes on . Perfect , close this isue . 
Thanks all.

closed 2018-07-17 by jonydebu