Koenkk/zigbee2mqtt #196: Implement Reset (FactoryReset) for EndDevice

tb-killa, 2018-07-14:
I read that inside http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf we have an FactoryReset Attribute.

I implemented ``toZigbee.js`` Converter.

```
genBasic_resetFactDefault: {
		key: 'reset',
		attr: [],
		convert: (value, message) => {
			return {
				cid: 'genBasic',
				cmd: 'resetFactDefault',
				zclData: {},
			};
		},
	},
```
has to integrate in specific converter inside ``devices.js`` inside ```toZigbee: [tz.genBasic_resetFactDefault],```.

Please do some Test, maybe i missread it or we have to integrate more .?

PS: Testable via ```zigbee2mqtt/<DEVICE_ID>/set``` and Message ```{"reset":""}```

Koenkk, 2018-07-14:
What would be the use case of this?

tb-killa, 2018-07-14:
```
3.2.2.3 Commands Received
The command IDs for the Basic cluster are listed in Table 3-13.
Table 3-13. Received Command IDs for the Basic Cluster
Command Identifier Description M/O
0x00 Reset to Factory Defaults O
3.2.2.3.1 Reset to Factory Defaults Command
This command does not have a payload.
3.2.2.3.1.1 Effect on Receipt
On receipt of this command, the device resets all the attributes of all its clusters to their factory defaults.
Note that ZigBee networking functionality,bindings, groups, or other persistent data are not affected by this
command.
```

tb-killa, 2018-07-14:
I use this successfully for eCozy (https://github.com/Koenkk/zigbee2mqtt/issues/129).

tb-killa, 2018-07-14:
Sometimes this could help if your device to strange things and you dont want to give the big prodecure via remove and repair.
If for this is no need we should close it ;)

tb-killa, 2018-07-15:
Yes will do it later ;)

tb-killa, 2018-07-21:
@Koenkk Thanks for Merge this. 
We should write some hint inside the Wiki.?
I will close it now, if needing we could reopen.

closed 2018-07-21 by tb-killa

Koenkk, 2018-07-21:
Yes, could you add it here: https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttdevice_idset ?