Koenkk/zigbee2mqtt #197: 2 State Messages after set message

wakille, 2018-07-15:
Hey,
I have the problem, that when im changing the brightness of an ikea tradfri bulb, I always get 2 state messages back. The first one contains the old brightness and in the second one the one I sended. I am not sure if is this behavior is intended. It makes it quite complicated if I parse the state message because I always have to ignore the first one, otherwise my value gets set back.

Koenkk, 2018-07-17:
Each attribute in the payload is set separately, so `ON` and `brightness` will both cause a state change -> 2 messages, one for `ON` and one for `brightness`.

wakille, 2018-07-20:
okay, didnt know that. But I think thats not very useful. Would be nice to get just one state message with the resulting state. But I also got the problem with just chainging the on/off state:

```
2018-07-20T23:17:04.530Z - info: Zigbee publish to '0x90fd9ffffe0cb0d4', genOnOff - on - {} - null
2018-07-20T23:17:04.692Z - info: Zigbee publish to '0x000b57fffed66a03', genOnOff - on - {} - null
2018-07-20T23:17:04.891Z - info: MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"ON"}'
2018-07-20T23:17:05.200Z - info: MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"ON"}'
2018-07-20T23:17:06.205Z - info: MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"ON"}'
2018-07-20T23:17:06.353Z - info: MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"ON"}'
2018-07-20T23:17:07.156Z - info: MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"ON"}'
2018-07-20T23:17:07.230Z - info: MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"ON"}'
2018-07-20T23:17:27.229Z - info: Zigbee publish to '0x90fd9ffffe0cb0d4', genOnOff - off - {} - null
2018-07-20T23:17:27.337Z - info: Zigbee publish to '0x000b57fffed66a03', genOnOff - off - {} - null
2018-07-20T23:17:27.501Z - info: MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"OFF"}'
2018-07-20T23:17:27.887Z - info: MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"OFF"}'
2018-07-20T23:17:29.211Z - info: MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"OFF"}'
2018-07-20T23:17:29.317Z - info: MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"OFF"}'
2018-07-20T23:17:30.076Z - info: MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"OFF"}'
2018-07-20T23:17:30.388Z - info: MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"OFF"}'
```

the set message is just {"state": "OFF"}/ {"state": "ON"}.

ciotlosm, 2018-07-21:
Maybe explore this configuration:

```
# Optional: advanced settings
advanced:
  # Optional: ZigBee pan ID
  pan_id: 0x1a62
  # Optional: ZigBee channel
  channel: 11
  # Optional: state caching 
  # https://github.com/Koenkk/zigbee2mqtt/commit/9396bde1f3b022e0f634487d1a37d2a5127c8cb3#diff-f68567477d803b49930337bf7fe1556bR16
  cache_state: true
  # Optional: Logging level, options: debug, info, warn, error
  log_level: info
  # Optional: Baudrate for serial port
  baudrate: 115200
  # Optional: RTS / CTS Hardware Flow Control for serial port
  rtscts: true
```

wakille, 2018-07-21:
Not sure if I unstand the debugging log right, but for me it looks like the state is requested 3 times. Is this intended ? And if yes, why ?

```
2018-7-21 19:35:03 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/0x90fd9ffffe0cb0d4/set' with data '{"state": "ON"}'
2018-7-21 19:35:03 INFO Zigbee publish to '0x90fd9ffffe0cb0d4', genOnOff - on - {} - null
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 79 +453ms
2018-7-21 19:35:03 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/0x000b57fffed66a03/set' with data '{"state": "ON"}'
2018-7-21 19:35:03 INFO Zigbee publish to '0x000b57fffed66a03', genOnOff - on - {} - null
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 80 +115ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +95ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +6ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 79 +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +24ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +37ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 81 +71ms
2018-7-21 19:35:03 INFO MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"ON"}'
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +198ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +10ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 80 +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +6ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +28ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +24ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 82 +12ms
2018-7-21 19:35:04 INFO MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"ON"}'
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +78ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +6ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 81 +8ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +52ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +65ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +14ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 83 +46ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +68ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +3ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 84 +101ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +70ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 82 +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +71ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +33ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 85 +13ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 86 +51ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +67ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 83 +8ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +25ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +24ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 87 +14ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +35ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +44ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +25ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +7ms
2018-7-21 19:35:05 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-7-21 19:35:05 INFO MQTT publish, topic: 'zigbee2mqtt/0x90fd9ffffe0cb0d4', payload: '{"state":"ON"}'
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +105ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 84 +2ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +48ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +84ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +14ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 85 +6ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +4ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +65ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +35ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 88 +24ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +137ms
[... 20 lines cut ...]
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 88 +3ms
[... cut ...]
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 90 +17ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +50ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +5ms
2018-7-21 19:35:06 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-7-21 19:35:06 INFO MQTT publish, topic: 'zigbee2mqtt/0x000b57fffed66a03', payload: '{"state":"ON"}'

```

tb205gti, 2018-09-24:
I have the same issues - apart from I get a null value for level.

Zigbee publish to '0x90fd9ffffee0f460', genOnOff - on - {} - null
  zigbee2mqtt:info 2018-9-24 21:30:58 Zigbee publish to '0x90fd9ffffee0f460', genLevelCtrl - moveToLevel - {"level":null,"transtime":0} - null
  zigbee2mqtt:info 2018-9-24 21:30:58 Zigbee publish to '0x90fd9ffffee0f460', genOnOff - on - {} - null
  zigbee2mqtt:info 2018-9-24 21:30:58 Zigbee publish to '0x90fd9ffffee0f460', genLevelCtrl - moveToLevel - {"level":186,"transtime":0} - null
  zigbee2mqtt:info 2018-9-24 21:30:58 Zigbee publish to '0x90fd9ffffee0f460', genOnOff - on - {} - null
  zigbee2mqtt:info 2018-9-24 21:30:58 Zigbee publish to '0x90fd9ffffee0f460', genLevelCtrl - moveToLevel - {"level":82,"transtime":0} - null

..So using the dimming functions of the Tradfri is more or less useless. :/

closed 2018-11-30 by stale[bot]