Koenkk/zigbee2mqtt #198: Unable to Pair Xiaomi MiJia wireless switch (WXKG01LM)

hahrens, 2018-07-16:
I'm unable to pair my first Device with zigbee2mqtt, a MiJia Switch (WXKG01LM). Once i've start the bridge in my console and allow pairing, i've started the button-pairing by pressing the config button on the switch until the led is blinking (~4sec) and continue it by pressing the button every second according to the [wiki documentation](https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi).

This is a repeatable error, i get it every time i try to pair the button.

```
~/zigbee2mqtt> sudo node index.js                                                                                                         master!
2018-7-16 11:49:02 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/home/pi/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-16 11:49:02 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-16 11:49:02 INFO Starting zigbee-shepherd
2018-7-16 11:49:03 INFO zigbee-shepherd started
2018-7-16 11:49:03 INFO Currently 0 devices are joined:
2018-7-16 11:49:03 WARN `permit_join` set to  `true` in configuration.yaml.
2018-7-16 11:49:03 WARN Allowing new devices to join.
2018-7-16 11:49:03 WARN Set `permit_join` to `false` once you joined all devices.
2018-7-16 11:49:03 INFO Zigbee: allowing new devices to join.
2018-7-16 11:49:03 INFO Connecting to MQTT server at mqtt://localhost
2018-7-16 11:49:03 INFO zigbee-shepherd ready
2018-7-16 11:49:03 INFO Connected to MQTT server
2018-7-16 11:49:03 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []
/home/pi/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^

TypeError: Cannot read property 'key' of undefined
    at ensureDataTypeString (/home/pi/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297:47)
    at getDataType (/home/pi/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:307:16)
    at Dissolve.<anonymous> (/home/pi/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:689:28)
    at Dissolve._transform (/home/pi/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
```

I've found the issue #107 with the same error message and already tried the commit referenced in [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/107#issuecomment-395981296) - doesn't fix it for me.

Here is my configuration.yml:
```
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

advanced:
  log_level: debug
```

tb-killa, 2018-07-16:
@hahrens see: https://github.com/Koenkk/zigbee2mqtt/issues/199

Koenkk, 2018-07-18:
Can you try running with `DEBUG=* npm start`?

hahrens, 2018-07-19:
Sure.

```
pi@fhem ~/zigbee2mqtt> DEBUG=* node index.js                                                 master
2018-7-19 22:35:47 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/home/pi/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-19 22:35:47 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-19 22:35:47 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM0 is opened. +46ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +9ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +7ms
  serialport write 8 bytes of data +32ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 51003, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 47, securityuse: 0, timestamp: 12524456, transseqnumber: 0, len: 29, data: <Buffer 1c 34 12 57 0a 02 ff 4c 06 00 10 01 21 f6 0b 21 a8 13 24 01 00 00 00 00 21 b1 00 20 57> } +60ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  cc-znp:SRSP <-- SYS:osalNvRead, { status: 0, len: 1, value: <Buffer 55> } +2ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +3ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +21ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 131, len: 2, value: [ 98, 26 ] } +1ms
  serialport write 6 bytes of data +3ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 131, len: 2, value: <Buffer 62 1a> } +13ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 132, len: 4, value: [ 0, 8, 0, 0 ] } +0ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 132, len: 4, value: <Buffer 00 08 00 00> } +19ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +4ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 98, len: 16, value: [ 1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13 ] } +3ms
  serialport write 6 bytes of data +8ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 98, len: 16, value: <Buffer 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d> } +24ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 99, len: 1, value: [ 0 ] } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 99, len: 1, value: <Buffer 00> } +18ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +14ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +0ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 01 00 06 00 00 00 2f> } +21ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +2ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +2ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +4ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 01 00 06 00 00 00 2f> } +10ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +3ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 1 } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 1, value: <Buffer be 2d 02 12 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 2 } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 2, value: <Buffer 00 00 02 12 00 4b 12 00> } +10ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 5 } +10ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 5, value: <Buffer 0b 00 02 12 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +3ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 6 } +1ms
  serialport write 6 bytes of data +5ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 6, value: <Buffer 62 1a 02 12 00 4b 12 00> } +16ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +3ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 7 } +0ms
  serialport write 6 bytes of data +2ms
[... 138 lines cut ...]
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +6ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 13986, status: 0, nwkaddr: 13986, len: 30, endpoint: 1, profileid: 260, deviceid: 260, deviceversion: 1, numinclusters: 4, inclusterlist: <Buffer 00 00 03 00 ff ff 19 00>, numoutclusters: 7, outclusterlist: <Buffer 00 00 04 00 03 00 06 00 08 00 05 00 19 00> } +2s
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +2ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +33ms
[... cut ...]
    at doWrite (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/home/pi/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
```

kirovilya, 2018-07-21:
Hmm, last buffer is good, but not parsed:
<Buffer 18 06 0a 05 00 42 12 6c 75 6d 69 2e 73 65 6e 73 6f 72 5f 73 77 69 74 63 68> - it is string 'lumi.sensor_switch'
it is necessary to dig deeper

hahrens, 2018-07-22:
Pairing successful with https://github.com/Koenkk/zcl-packet/commit/3ea8912f9d228b59fad554064d15ed9be7897a7f

kirovilya, 2018-07-22:
@hahrens So you used old version? And all working now?

hahrens, 2018-07-25:
@kirovilya Yes. I've manually cloned the zcl-packet in node_modules after `npm install` and checked out the https://github.com/Koenkk/zcl-packet/commit/3ea8912f9d228b59fad554064d15ed9be7897a7f commit. After that the pairing was successful.

kirovilya, 2018-07-25:
@hahrens So, you need install/update to last version of zigbee2mqtt with updated version of zcl-packet lib with this changes.

kirovilya, 2018-07-26:
yes, it looks like I was wrong somewhere :(

Sonsi79, 2018-07-29:
@kirovilya I have check the smoke battery function. It works great again. thnx

Koenkk, 2018-07-29:
@hahrens can you check if this issue is still present in the dev branch?

FrankZZ, 2018-07-29:
I see the same error when the RTCGQ01LM reports occupancy. Just switched to dev branch and testing if it still crashes @Koenkk

hahrens, 2018-07-29:
I can confirm fix in dev branch.

Koenkk, 2018-07-30:
Ok, great, thanks all!

closed 2018-07-30 by Koenkk

Baseje, 2018-08-09:
I still receive this errors;
2018-8-9 16:49:20 INFO Device incoming...
2018-8-9 16:49:20 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-8-9 16:51:08 INFO Zigbee: allowing new devices to join.
spinlock: false []
2018-8-9 16:52:10 ERROR Cannot get the Node Descriptor of the Device: 0x00158d00022435f7 (Error: Timed out after 10000 ms)
spinlock: false []
2018-8-9 16:53:08 ERROR Cannot get the Node Descriptor of the Device: 0x00158d00022435f7 (Error: Timed out after 10000 ms)

Koenkk, 2018-08-09:
@Baseje do you keep pressing the reset button while pairing?

Baseje, 2018-08-09:
I’ve pushed the normal button, also tried to hold that button no result in zigbee2mqtt log.
After that i tried the small button with an paperclip on the back to connect. No result.

Maybe i do something wrong…

Baseje, 2018-08-09:
I've tested again and receive the error after holding the small button with the paperclip for 3 seconds.
In the manual they tell its an reset commando.

How did you connect the sensor?
When i push or hold the big button i see no response in zigbee2mqtt..

Koenkk, 2018-08-09:
The procedure is described here: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi

Baseje, 2018-08-09:
what a hell to connect, every second push the reset button...
Sorry i didnt read the manual good enough...

Many thanks!