Koenkk/zigbee2mqtt #201: Xiaomi Zigbee Power Plug Update Frequency

kiwijunglist, 2018-07-18:
The xiaomi power plug reports wattage used. In the mi home app changes to wattage used is updated almost instantly. In the zigbee2mqtt the update is 1 minute or so.

Can the update frequency be increased or manually set?

Koenkk, 2018-07-29:
Unfortunately I've lost my EU plug adapter, just ordered a new one and will check when it arrives.

subzero79, 2018-08-01:
How would you work on this feature? Do you have to capture xiaomi gw and plug zigbee traffic? Using the same usb dongle?

I ask this because was of the features of the app was also to restore the power state of the plug after power outage. Don’t know if this was controlled by the gw or is a setting of the actual plug.

subzero79, 2018-08-08:
Is it as simply as adding the lines to the definition in converters for the xiaomi plug? or needs additional manual configuration in the yaml file?

If that method doesn't work will you then have to start sniffing traffic?

Koenkk, 2018-08-08:
A `configure` method has to be added in `devices.js`. Similar to: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L559

subzero79, 2018-08-11:
Sure, i've added the same lines and works as expected, energy report is instantaneous, but my impression is that only reports instantly if going up on power maybe more than 15W, less than that takes about 10 seconds which is still good. What i see also it doesn't report fast when power goes down it takes the normal time like a minute or so.

subzero79, 2018-08-12:
I think i spoke too soon. I see no difference with the function there. I just moved to the dev branch to test this, and comparing this to the stable branch i am pretty sure i see faster report times, but report function makes no difference. 
In my case the plug is attached to a fan. So basically turn on the fan light (20W) makes the report almost instant. Any speed setting and report comes back after 8-10 seconds. Must be due to current increasing slowly until reaches speed.
The slowest reports comes when consumption goes down. Takes something like 50-60 seconds.

subzero79, 2018-08-23:
Here is a wireshark capture of the zigbee traffic being captured while i interact via the mi app.

https://www.dropbox.com/s/v1st24npnos1mdv/xiaomi_gw_capture.tar.bz2?dl=0

The file readPowerfromState has the info in line 322. It requests read attribute to the endpoint

The file setRestorePowerState is when the restore power boolean is toggled in the app in line 354, it writes attributes these are the values i saw

Off restore state aa:80:05:d1:47:2d:01:10:00
On restore state aa:80:05:d1:47:2c:01:10:01

As for the file that turns the led on/off, is in the setLedStatusOnOff file, line 255 (led off), line 354 (led on). 

The master key is the same as listed everywhere Zigbee Trust Center Key. You should see the transport key in the capture to decrypt traffic below the joining.

Hope it helps

Koenkk, 2018-08-24:
@subzero79 thanks!

From the logs, does the gateway constantly poll the plug or does the plug send its state on a change?

subzero79, 2018-08-24:
It does not poll constantly. The request come from when I click the device within the app. 
From within the device in the app if power changes, the report is not immediately unless I go back and then click the device again.

Koenkk, 2018-08-25:
Ok, so does this mean:
- When in app the Xiaomi gateway polls -> changes are almost instantly
- When not in app behaviour is the same as of zigbee2mqtt.

subzero79, 2018-08-25:
From what I saw changes are not instant using the app. You have to force (sort of a refresh) go back to the devices list and click the plug again to get the power consumption, if it has changed. If you’re within the device window and there is power change it doesn’t show up in the app unless you do what I described before. 

I didn’t monitor the zig bee traffic without the app. I can test again to see how the energy reports when consumption changes. But my guess is probably similar to what we have in zigbee2mqtt atm. Big increments are instant, smaller ones take ~10 seconds, and go back to zero or reductions take around 50-60 seconds. I’ll test this again with the gw

I’ll do some more testing.

subzero79, 2018-08-28:
> Do you capture like mentioned here: https://blog.jimmo.id.au/zigbee-packet-capture-724cf9346b47 ??

Yes, difference i did not build the contiki fw (good luck trying that), i use the sniffer_fw_cc2531.hex(which is shipped by TI), and whisniff (in linux). So basically use the trust center key, then pair a device to obtain a transport key.

As i mentioned before, you have to add the well known trust center key to the zigbee protocol, then the join process will appear. In the lines of the join process grab the transport key and add it again to the protocol, that should have the rest of the traffic de-crypted.

subzero79, 2018-08-28:
I did not find a way with wireshark to save as decrypted traffic

subzero79, 2018-09-04:
@tb-killa Did you manage to read the pcap file or still having trouble de-crypting it?

mattkerrison, 2018-12-17:
This would be really useful to update more than once a min

subzero79, 2018-12-17:
https://github.com/Koenkk/zigbee2mqtt/issues/518

closed 2019-02-22 by stale[bot]