Koenkk/zigbee2mqtt #203: Digi XBee compatibility?

Diaoul, 2018-07-21:
I have a some [XBee Series 2](https://www.sparkfun.com/products/8665) modules from Digi and a [Serial to USB interface](https://www.sparkfun.com/products/11697). Basically as a coordinator, frames are forwarded to the Serial interface.

Can I run Zigbee2MQTT with that? With or without flashing it?

tb-killa, 2018-07-21:
As Cordinator sadly no because actually only TI Stack is supported.
Theoretically you could use the XBEE as EndDevice and support them via Converter (we have one DIY acutally). 
Maybe to build some simple IO EndDevice ;) ?

Diaoul, 2018-07-21:
Well, too bad 😢 I can certainly use them as EndDevice for battery-powered DIY stuff.

closed 2018-07-21 by Diaoul

StefanD986, 2018-12-18:
Could one of the core-devs outline what would have to be done to support other Zigbee radio interface chips (like the Xbee)?

ptvoinfo, 2018-12-18:
@StefanD986 If you mean to use these chips/boards as coordinators, then the coordinator firmware for this chip should support a wide set of MT commands. These commands are built-in to Z Stack by Texas Instruments.

sergey-suloev, 2022-01-13:
Still no support for this excellent adapter as coordinator ???