Koenkk/zigbee2mqtt #204: no Devices Join

Paddyirl, 2018-07-21:
Hi I have hassio and have setup your hassio-zigbee2mqtt and have the following in the log

zigbee2mqtt@0.1.0 start /app
node index.js
2018-7-20 10:41:00 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-20 10:41:00 INFO Starting zigbee-shepherd
2018-7-20 10:41:04 INFO zigbee-shepherd started
2018-7-20 10:41:04 INFO Currently 0 devices are joined:
2018-7-20 10:41:04 WARN permit_join set to true in configuration.yaml.
2018-7-20 10:41:04 WARN Allowing new devices to join.
2018-7-20 10:41:04 WARN Set permit_join to false once you joined all devices.
2018-7-20 10:41:04 INFO Zigbee: allowing new devices to join.
2018-7-20 10:41:05 INFO Connecting to MQTT server at mqtt://homeassistant
2018-7-20 10:41:05 INFO zigbee-shepherd ready
2018-7-20 10:41:05 INFO Connected to MQTT server
2018-7-20 10:41:05 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'

I have reflashed the usb sniffer and uninstalled the hassio-zigbee addon and have installed the zigbee2mqtt-edge addon and get the same messages as above .I have put the ikea led 1622g12 e27 right beside the usb sniffer in rpi2 .Any help or suggestions would be much appreciated

ciotlosm, 2018-07-21:
Are you using the correct pairing method? https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices

Paddyirl, 2018-07-21:
I have tried the pairing method and i have tried both versions of the hassio-zigbee2mqtt  and i will revert to the one you recommend but still no devices join .

ciotlosm, 2018-07-21:
Do you have some logs output when trying to join devices? Try starting in DEBUG mode to see at bit more of what's going on.

Paddyirl, 2018-07-21:
I have reverted to  hassio-zigbee2mqtt ver (0.1.0) and this is the log with debug 

[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
[2018-07-21T13:14:05.253Z] PM2 log: Launching in no daemon mode
[2018-07-21T13:14:06.585Z] PM2 log: Starting execution sequence in -fork mode- for app name:npm id:0
[2018-07-21T13:14:06.680Z] PM2 log: App name:npm id:0 online
> zigbee2mqtt@0.1.0 start /app
> node index.js
2018-7-21 13:14:21 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-21 13:14:22 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-21 13:14:22 INFO Starting zigbee-shepherd
2018-7-21 13:14:26 INFO zigbee-shepherd started
2018-7-21 13:14:26 INFO Currently 0 devices are joined:
2018-7-21 13:14:26 WARN `permit_join` set to  `true` in configuration.yaml.
2018-7-21 13:14:26 WARN Allowing new devices to join.
2018-7-21 13:14:26 WARN Set `permit_join` to `false` once you joined all devices.
2018-7-21 13:14:26 INFO Zigbee: allowing new devices to join.
2018-7-21 13:14:26 INFO Connecting to MQTT server at mqtt://homeassistant
2018-7-21 13:14:27 INFO zigbee-shepherd ready
2018-7-21 13:14:27 INFO Connected to MQTT server
2018-7-21 13:14:27 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'

ciotlosm, 2018-07-22:
@Paddyirl you should have seen at least a message with `spinlock[]` in the logs if you were pressing the pair button. I don't have any ikea. I assume you already watched this video: https://www.youtube.com/watch?v=npxOrPxVfe0. Maybe someone else can help out either on this issue or maybe discord or some other chat if you have.

Paddyirl, 2018-07-25:
I have  watched the video and just to be sure i have tried 4 gu 10 ikea tradfri and even a Innr bulb and the log is the same as i posted . Can anybody help with this  ?

Paddyirl, 2018-08-04:
Hi I have found my problem and discovered that  i was running mqtt ver 1.0 and when i update to mqtt 2.0 then everything works and i can discover all my bulbs . hope this helps someone else . I have a bulb that comes up as unrecognized and it is Innr by 165 b22 which is identical to Innr rb 165 e27 except for the base .Can this be added to supported list ?

Koenkk, 2018-08-04:
@Paddyirl if you provide the output when connecting that bulb we can add support for it.

Paddyirl, 2018-08-05:
Many thanks here is the output i get in log 
2018-08-05T11:08:37.778Z - info: innr_rb165 (0x00158d0001c791be): unkown - undefined unknown (Router)
2018-08-05T11:09:15.794Z - warn: Device with modelID 'BY 165' is not supported
2018-08-05T11:09:15.799Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-08-05T11:09:15.804Z - debug: Recieved zigbee message with data "online"
2018-08-05T11:09:15.808Z - warn: Device with modelID 'BY 165' is not supported.
2018-08-05T11:09:15.812Z - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

Koenkk, 2018-08-05:
Is it this one: https://www.amazon.co.uk/Innr-dimmable-Compatible-Philips-165/dp/B01MR92SN0 ?

Paddyirl, 2018-08-05:
yes it is

Koenkk, 2018-08-05:
@Paddyirl could you add the following to `node_modules/zigbee-shepherd-converters/devices.js`

```js
    {
        zigbeeModel: ['BY 165'],
        model: 'BY 165',
        vendor: 'Innr',
        description: 'B22 Bulb dimmable',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: generic.light_onoff_brightness().fromZigbee,
        toZigbee: generic.light_onoff_brightness().toZigbee,
    },
```

Paddyirl, 2018-08-08:
Thank you for your zigbeeModel BY 165 but can anybody help me to find node_modules/zigbee-shepherd-converters/devices.js

Koenkk, 2018-08-09:
@Paddyirl should be supported in the dev branch now, can you verify?

Paddyirl, 2018-08-09:
Yes can confirm it shows in dev branch and thank you for a great addon and all your hard work

closed 2018-08-10 by Koenkk

Eeeeeediot, 2018-10-17:
```
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0002391a06","nwkAddr":4841,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,1026,1027,1029,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msPressureMeasurement":{"dir":{"value":1},"attrs":{}},"msRelativeHumidity":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"AzZZogjoeY7sSnhS"}
{"id":7,"type":"Router","ieeeAddr":"0x00158d0001d0c6f6","nwkAddr":27479,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"BY 265","epList":[1,242],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"9":5,"10":{},"11":"www.innr.com","65533":1,"zclVersion":2,"appVersion":16,"stackVersion":21,"hwVersion":1,"manufacturerName":"innr","modelId":"BY 265","dateCode":"20180810-1","powerSource":1,"appProfileVersion":0,"swBuildId":"2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":178,"identifyCommissionState":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"65533":1,"count":1,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0x0000000000000000"}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":255,"65533":1,"currentLevel":254,"remainingTime":0}},"genOta":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":1},"attrs":{"65533":1}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{"genGreenPowerProxy":{"dir":{"value":2},"attrs":{}}}}},"_id":"QhWFhUBLwWxDuVQ0"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0002b43011","nwkAddr":55131,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"UKugvelcGwJCdcqy"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00025298f4","nwkAddr":48186,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,6,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"WlyiKHGm8kG3HtPL"}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d00025297ed","nwkAddr":4801,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,6,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"YnDN6lXktNJAcOuJ"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018ecde65","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1539730875,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"wsFi5Ubr2lZUlZCj"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018ecde65","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1539731413,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outCluster [... cut ...]
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d00025297ed","nwkAddr":4801,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,6,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"YnDN6lXktNJAcOuJ"}
```

I tried to upload file but it didnt work... is this good enough?

Eeeeeediot, 2018-10-17:
Thanks @Koenkk!

If i install the edge version just now will it work?

Koenkk, 2018-10-17:
@Eeeeeediot yes

Eeeeeediot, 2018-10-17:
Thank you so much!

Eeeeeediot, 2018-10-18:
Hi @Koenkk  i have installed one of the bulbs and it is working perfectly how ever i cant install the second one. I have tried uninstalling the addon, removing the first bulb and trying to install the second one.

Is it possible that it will not install because the product ID may be identical and 2 devices with the same properties cannot be paired together?

Koenkk, 2018-10-18:
Product ID shouldn't matter, did you properly reset the bulb?

Eeeeeediot, 2018-10-18:
Yup, several times. I have now managed to loose connection with all my devices. I only have 5(xiaomi Aqara) so i don't mind setting them back up manually but the pairing method doesnt seems to be working now.

Is there anyway i can factory restet the addon/devices to trick hass.io into thiking that i am running it for the first time?

Koenkk, 2018-10-18:
I would recommend reflashing your stick and clearing your `database.db`.

Eeeeeediot, 2018-10-18:
That didnt work @Koenkk  :(. Do you know how I can delete the moquitto.db file on hassio and then ill give that a try?

thanks

Eeeeeediot, 2018-10-18:
Got it working! flashed it 4 times and both connected on the 4th! This is a great system you made and customer service was even better! Thanks again