Koenkk/zigbee2mqtt #207: Location of log file

SargonofAssyria, 2018-07-22:
Tnx, nice job.
I managed to get zigbee2mqtt working with Domoticz on a Raspberry Pi.
I would like to have the logging to the tmpfs '/var/log'
Could it be possible to set the location of the logging in the config file?

Koenkk, 2018-07-22:
Not yet, but this can be added as an feature.

SargonofAssyria, 2018-07-22:
Perfect, I wait for it.

tb-killa, 2018-07-28:
We could combine this via https://github.com/Koenkk/zigbee2mqtt/issues/182.

Koenkk, 2018-08-05:
This is now possible (currently dev branch only).

```yaml
advanced:
  log_directory: /var/log
```

tb-killa, 2018-08-08:
Works great, could be merged :)

Koenkk, 2018-08-08:
I try to merge the dev branch into master every +- 2 weeks. However currently my time is very limited due to moving houses.

closed 2018-08-08 by Koenkk