Koenkk/zigbee2mqtt #208: ERROR Failed to start

HessyJames, 2018-07-22:
Dear All,

after starting the service with "npm start" I got the following error:

2018-7-22 12:05:46 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-22 12:05:46 INFO Starting zigbee-shepherd
2018-7-22 12:05:46 INFO Error while starting zigbee-shepherd, attemping to fix..                                     . (takes 60 seconds)
2018-7-22 12:06:46 INFO Starting zigbee-shepherd
2018-7-22 12:06:46 ERROR Error while starting zigbee-shepherd!
2018-7-22 12:06:46 ERROR Press the reset button on the stick (the one closest to                                      the USB) and start again
2018-7-22 12:06:46 ERROR Failed to start
        {"message":"Error Resource temporarily unavailable Cannot lock port","st                                     ack":"Error: Error Resource temporarily unavailable Cannot lock port"}

I already pressed the reset button, but without any changes. Can anybody help me?

Thanks an BR
Volker

Koenkk, 2018-07-22:
It seems that another application is using the port or that the specified port is not ok. What is your `configuration.yaml` and what is the output of `ls /dev`?

HessyJames, 2018-07-22:
Hi Koenkk,
thanks you for your quick reply!

I hope the following infomration can help you.

pi@FHEM:~ $ **lsusb**
Bus 001 Device 006: ID 03eb:204b Atmel Corp. LUFA USB to Serial Adapter Project
Bus 001 Device 005: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 004: ID 0451:16a8 Texas Instruments, Inc.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
-------------------------------------------------------------------------------------------------------------------

pi@FHEM:~ $ **ls -l /dev/ttyACM0**
crw-rw---- 1 root dialout 166, 0 Jul 22 18:34 /dev/ttyACM0

-------------------------------------------------------------------------------------------------------------------
**configuration.yaml**

# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://192.168.178.41'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer

HessyJames, 2018-07-22:
Hi,
I found out, that zigbee-shepherd is working on anoter system (PI2b)  where i've running my MQTT and HAbridge and Alexa-FEHM.

BR
Hessy

aptonline, 2018-08-23:
I’m having the same issue on my Synology. Gettting the same error that @hessyjames reports.

The output is identical too.

Any help welcomed.

closed 2018-10-29 by stale[bot]

konstantin999, 2019-12-01:
It seems that I have the same issue :( in docker "koenkk/zigbee2mqtt:latest"

zigbee2mqtt:info  2019-12-01T12:08:12: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
zigbee2mqtt:info  2019-12-01T12:08:12: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-12-01T12:08:13: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-01T12:08:13: Failed to start zigbee
zigbee2mqtt:error 2019-12-01T12:08:13: Exiting...
zigbee2mqtt:error 2019-12-01T12:08:13: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot loc
k port'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:109:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.serialPort.open (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:107:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:197:14)
    at binding.open.then.err (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:241:12)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-12-01T12_08_13_015Z-debug.log

Any help welcomed.

DaveMe81, 2020-02-10:
Same problem here:

`Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.serialPort.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at binding.open.then.err (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12)
`

Funny, as I installed the device, I found that MQTT broker was missing. After installing mosquitto mqtt all seams good. Then installed node-red-contrib-zigbee tried to do something, like adding a zigbee shepherd. After deploying already a error drops. restarting the zigbee2mqtt was anwers with the above message.

re-plugging the stick and pressing the button closer to the usb port didn't helped.

maxpd1, 2020-08-02:
Same here. I only made sudo reboot.

```
zigbee.0 | 2020-08-02 21:10:34.730 | error | at /opt/iobroker/node_modules/iobroker.zigbee/node_modules/@serialport/stream/lib/index.js:242:12
-- | -- | -- | --
zigbee.0 | 2020-08-02 21:10:34.730 | error | at SerialPort._error (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/@serialport/stream/lib/index.js:198:14)
zigbee.0 | 2020-08-02 21:10:34.730 | error | at SerialPort.<anonymous> (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
zigbee.0 | 2020-08-02 21:10:34.730 | error | at __awaiter (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
zigbee.0 | 2020-08-02 21:10:34.730 | error | at new Promise (<anonymous>)
zigbee.0 | 2020-08-02 21:10:34.730 | error | at /opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
zigbee.0 | 2020-08-02 21:10:34.730 | error | at Generator.next (<anonymous>)
zigbee.0 | 2020-08-02 21:10:34.730 | error | at Znp.<anonymous> (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
zigbee.0 | 2020-08-02 21:10:34.730 | error | (2639) Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
zigbee.0 | 2020-08-02 21:10:34.727 | error | (2639) Failed to start Zigbee

```

scaldov, 2020-08-28:
rebooted and have had the same problem.

> zigbee2mqtt@1.14.3 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-08-28 06:51:39: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-28.06-51-39' filename: log.txt
Zigbee2MQTT:info  2020-08-28 06:51:39: Starting Zigbee2MQTT version 1.14.3 (commit #0db3f3c)
Zigbee2MQTT:info  2020-08-28 06:51:39: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-08-28 06:51:44: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-28 06:51:44: Failed to start zigbee
Zigbee2MQTT:error 2020-08-28 06:51:44: Exiting...
Zigbee2MQTT:error 2020-08-28 06:51:44: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/zigbee/.npm/_logs/2020-08-28T02_51_44_882Z-debug.log
scaldov@scyld:/opt/zigbee2mqtt$ Zigbee2MQTT:error 2020-08-28 06:52:07: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-28 06:52:07: Failed to start zigbee
Zigbee2MQTT:error 2020-08-28 06:52:07: Exiting...
Zigbee2MQTT:error 2020-08-28 06:52:07: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

then 

> zigbee2mqtt@1.14.3 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-08-28 06:58:41: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-28.06-58-41' filename: log.txt
Zigbee2MQTT:info  2020-08-28 06:58:42: Starting Zigbee2MQTT version 1.14.3 (commit #0db3f3c)
Zigbee2MQTT:info  2020-08-28 06:58:42: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-08-28 06:59:23: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-08-28 06:59:23: Failed to start zigbee
Zigbee2MQTT:error 2020-08-28 06:59:23: Exiting...
Zigbee2MQTT:error 2020-08-28 06:59:23: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/zigbee/.npm/_logs/2020-08-28T02_59_58_947Z-debug.log

Stefano0042, 2020-09-02:
Same issue here:
new coordinator: CC2652RB (Slaesh's iot) mounted on /dev/tty/USB0 (previous CC2531 was on /dev/tty/ACM0)
- USB permission has been given with udev rules:
   ACTION=="add", KERNEL=="dialout", MODE="0666"
   ACTION=="add", KERNEL=="ttyUSB0", MODE="0666"
- new pan ID in configuration.yaml

> zigbee2mqtt@1.14.3 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-09-02 10:26:49: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-09-02.10-26-49' filename: log.txt
Zigbee2MQTT:info  2020-09-02 10:26:51: Starting Zigbee2MQTT version 1.14.3 (commit #a057f3c)
Zigbee2MQTT:info  2020-09-02 10:26:51: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-09-02 10:26:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-09-02 10:26:53: Failed to start zigbee
Zigbee2MQTT:error 2020-09-02 10:26:53: Exiting...
Zigbee2MQTT:error 2020-09-02 10:26:53: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.14.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-02T08_26_53_308Z-debug.log

Stefano0042, 2020-09-02:
Just to update you.
I got it working! I will further investigate on the root problem.

slaesh, 2020-09-03:
Hey @Stefano0042 , nice to see you here ;)
So I know, we will get this https://github.com/ioBroker/ioBroker.zigbee/issues/749 running too =)

Could you share you experiences? :)

Juanuil, 2020-10-17:
Hi there,
same problem here. Would be very interested how to solve it

BigMike71, 2020-10-24:
have this problem too :(
using IObroker+Zigbee Adapter+TI CC26X2R1 Launchpad

slaesh, 2020-10-26:
@Stefano0042 can you help here? =)

Stefano0042, 2020-10-26:
my specific problem was related to the forgotten presence of the daemon service zigbee2mqtt running in the background,  so the manual command "npm start" on the bash was overlapping the already running service.

belichuk, 2021-04-02:
Faced the same issue. I tried to run `npm start` from the project folder.  But got an error
```
Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialDriver.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/uart.js:201:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/uart.js:27:71
```

This happens due to already executed `zigbee2mqtt.service` via `systemctl`.

wmarchesi123, 2022-09-02:
> Faced the same issue. I tried to run `npm start` from the project folder. But got an error
>
> ```
> Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock [...]

Using `systemctl stop zigbee2mqtt.service` then `npm start` solved my issue, thank you!