Koenkk/zigbee2mqtt #210: Device support: CentraLite 3305

Mdleal, 2018-07-22:
I am getting no warnings minus the original 'is not supported'.
warn: Device with modelID '3305' is not supported.

devices.js
`    { 
        zigbeeModel: ['3305'], 
        model: '3305', 
        vendor: 'CentraLite', 
        description: 'CentraLite motion & temperature sensor',
        supports: 'occupancy and temperature', 
        fromZigbee: [], // We will add this later
        toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
     },
`

Mdleal, 2018-07-24:
`2018-7-24 13:55:23 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-24 13:55:23 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-24 13:55:23 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM0 is opened. +23ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +4ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +6ms
  serialport write 8 bytes of data +15ms
  cc-znp:SRSP <-- SYS:osalNvRead, { status: 0, len: 1, value: <Buffer 55> } +36ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +3ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +247ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 131, len: 2, value: [ 98, 26 ] } +2ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 131, len: 2, value: <Buffer 62 1a> } +17ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 132, len: 4, value: [ 0, 8, 0, 0 ] } +1ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 132, len: 4, value: <Buffer 00 08 00 00> } +31ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 98, len: 16, value: [ 1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13 ] } +0ms
  serialport write 6 bytes of data +3ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 98, len: 16, value: <Buffer 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d> } +40ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 99, len: 1, value: [ 0 ] } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 99, len: 1, value: <Buffer 00> } +35ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +13ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +0ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 01 00 05 00 00 00 82> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +1ms
  zigbee-shepherd:request REQ --> SYS:version +1ms
  cc-znp:SREQ --> SYS:version, {} +1ms
  serialport write 5 bytes of data +1ms
  cc-znp:SRSP <-- SYS:version, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:version +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +11ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +1ms
  serialport write 6 bytes of data +2ms
  cc-znp Error: parse timeout
  cc-znp     at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/cc-znp/lib/zpiObject.js:117:26)
  cc-znp     at Object.onceWrapper (events.js:315:30)
  cc-znp     at emitOne (events.js:116:13)
  cc-znp     at Dissolve.emit (events.js:211:7)
  cc-znp     at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/cc-znp/lib/zpiObject.js:103:24)
  cc-znp     at ontimeout (timers.js:498:11)
  cc-znp     at tryOnTimeout (timers.js:323:5)
  cc-znp     at Timer.listOnTimeout (timers.js:290:5) +3ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 01 00 05 00 00 00 82> } +8ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +2ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 1 } +1ms
  serialport write 6 bytes of data +1ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 1, value: <Buffer d3 13 45 01 00 4b 12 00> } +10ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 2 } +1ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 2, value: <Buffer 00 00 45 01 00 4b 12 00> } +9ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 5 } +1ms
  serialport write 6 bytes of data +2ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 5, value: <Buffer 0b 00 45 01 00 4b 12 00> } +9ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +1ms
[... 640 lines cut ...]
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 37 +0ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 25, srcaddr: 47104, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 0, securityuse: 0, timestamp: 6824125, transseqnumber: 0, len: 32, data: <Buffer 08 25 01 05 00 00 21 ff ff 06 00 00 30 00 07 00 00 21 dc 10 08 00 00 21 ff ff 09 00 00 21 00 00> } +43ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +17ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
[... cut ...]
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
2018-7-24 13:55:46 DEBUG Recieved zigbee message with data "0x7ce524000012d433"
  zigbee-shepherd Device: 0x7ce524000012d433 join the network. +12ms
2018-7-24 13:55:46 DEBUG Recieved zigbee message with data "0x7ce524000012d433"
2018-7-24 13:55:46 INFO Device incoming...
2018-7-24 13:55:46 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-7-24 13:55:46 DEBUG Recieved zigbee message with data "online"
`

tb-killa, 2018-07-24:
Please use for long logfiles services like pastbin or equal.

We need also your database.db entrys, please post this.

Mdleal, 2018-07-24:
Thank you. I've been trying to remember what that site was all morning. 
[DEBUG](https://pastebin.com/Q9MuQ7xD)
[database.db](https://pastebin.com/DwuQfvTC)

tb-killa, 2018-09-01:
Please provide your ``database.db`` content.