Koenkk/zigbee2mqtt #212: Swann One Motion Detector

mattclar, 2018-07-23:
I am trying to get this motion detector woring, I have two of them..
https://www.swann.com/au/swo-mos1pa
I am following the wiki documentation and have got up to the point at which I have added the device to devices.js
```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-7-23 11:45:53 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-23 11:45:53 INFO Starting zigbee-shepherd
2018-7-23 11:45:59 INFO zigbee-shepherd started
2018-7-23 11:45:59 INFO Currently 2 devices are joined:
2018-7-23 11:45:59 INFO 0x00124b0006223e40 (0x00124b0006223e40): SWO-MOS1PA - Swann Swann One Motion Sensor (EndDevice)
2018-7-23 11:45:59 INFO 0x00124b0006223e8b (0x00124b0006223e8b): SWO-MOS1PA - Swann Swann One Motion Sensor (EndDevice)
2018-7-23 11:45:59 WARN `permit_join` set to  `true` in configuration.yaml.
2018-7-23 11:45:59 WARN Allowing new devices to join.
2018-7-23 11:45:59 WARN Set `permit_join` to `false` once you joined all devices.
2018-7-23 11:45:59 INFO Zigbee: allowing new devices to join.
2018-7-23 11:45:59 INFO Connecting to MQTT server at mqtt://192.168.1.2
2018-7-23 11:46:01 INFO zigbee-shepherd ready
2018-7-23 11:46:01 INFO Connected to MQTT server
2018-7-23 11:46:01 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []
2018-7-23 11:47:55 INFO Device incoming...
2018-7-23 11:47:55 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
spinlock: false []
device already in network
spinlock: false []
device already in network

```
however I am not seeing any activity, the only thing I can reliably trigger is if I press the 'add to network' button on the sensor I get the following:
```
zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +13ms
  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 23015, nwkaddr: 23015, ieeeaddr: '0x00124b0006223e40', capabilities: 128 } +14s
spinlock: false []
device already in network
```

as you can see I have tried starting it with the DEBUG=* option, without any luck
Is there something I am missing or that is missing from the wiki?

my database.db is below:
```
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0006223e40","nwkAddr":23015,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList"[1],"status":"online","joinTime":1532346455,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,2,3,1280,64512],"outClusterList":[3],"clusters":{"64512":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs{"hwVersion":1,"manufacturerName":"SwannONe","modelId":"SWO-MOS1PA","dateCode":"20150306        ","powerSource":3}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryAlarmState":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":17}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":0,"zoneStatus":49,"iasCie
Addr":"0xffffffffffffffff","zoneId":0}}}}},"_id":"5iy3Y31PC3bHCURs"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0012013fe1","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1532346358,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"e
pId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"
epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"Hg1DgMX96SOV1N3F"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00124b0006223e8b","nwkAddr":729,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":
[0,1,2,3,1280,64512],"outClusterList":[3],"clusters":{"64512":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":1,"manufacturerName":"SwannONe","modelId":"SWO-MOS1PA","dateCode":"20150306        ","powerSource":3}},"genPowerCfg":{"dir":{"va
lue":1},"attrs":{"batteryAlarmState":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":13}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":0,"zoneStatus":48,"iasCieAddr":"
0xffffffffffffffff","zoneId":0}}}}},"_id":"TM0STyoT0tMIascp"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0012013fe1","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1532348955,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"e
pId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"
epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"Hg1DgMX96SOV1N3F"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0006223e40","nwkAddr":23015,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList
":[0,1,2,3,1280,64512],"outClusterList":[3],"clusters":{"64512":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":1,"manufacturerName":"SwannONe","modelId":"SWO-MOS1PA","dateCode":"20150306        ","powerSource":3}},"genPowerCfg":{"dir":{"
value":1},"attrs":{"batteryAlarmState":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":17}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":0,"zoneStatus":49,"iasCieAddr"
:"0xffffffffffffffff","zoneId":0}}}}},"_id":"5iy3Y31PC3bHCURs"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00124b0006223e8b","nwkAddr":729,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":
[0,1,2,3,1280,64512],"outClusterList":[3],"clusters":{"64512":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":1,"manufacturerName":"SwannONe","modelId":"SWO-MOS1PA","dateCode":"20150306        ","powerSource":3}},"genPowerCfg":{"dir":{"va
lue":1},"attrs":{"batteryAlarmState":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":13}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":0,"zoneStatus":48,"iasCieAddr":"
0xffffffffffffffff","zoneId":0}}}}},"_id":"TM0STyoT0tMIascp"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0006223e40","nwkAddr":23015,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList":[1],"status":"online","joinTime":1532348960,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inCluste
rList":[0,1,2,3,1280,64512],"outClusterList":[3],"clusters":{"64512":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":1,"manufacturerName":"SwannONe","modelId":"SWO-MOS1PA","dateCode":"20150306        ","powerSource":3}},"genPowerCfg":{"di
r":{"value":1},"attrs":{"batteryAlarmState":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":17}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":0,"zoneStatus":49,"iasCie
Addr":"0xffffffffffffffff","zoneId":0}}}}},"_id":"5iy3Y31PC3bHCURs"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b0006223e40","nwkAddr":23015,"manufId":0,"manufName":"SwannONe","powerSource":"Battery","modelId":"SWO-MOS1PA","epList":[1],"status":"online","joinTime":1532348960,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inCluste
[... cut ...] "value":1},"attrs":{"hwVersion":1,"manufacturerName":"SwannONe","modelId":"SWO-MOS1PA","dateCode":"20150306        ","powerSource":3}},"genPowerCfg":{"di
r":{"value":1},"attrs":{"batteryAlarmState":0}},"genDeviceTempCfg":{"dir":{"value":1},"attrs":{"currentTemperature":19}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":0,"zoneStatus":49,"iasCie
Addr":"0xffffffffffffffff","zoneId":0}}}}},"_id":"5iy3Y31PC3bHCURs"}

```

tb-killa, 2018-07-24:
Please insert for first tests this type of segment into ``devices.js`` and restart via ```DEBUG=zigbee-shepherd* npm start ```

```
{ 
    zigbeeModel: ['SWO-MOS1PA'], 
    model: 'TODO', // Vendor model number, look on the device for a model number
    vendor: 'SwannONe', // Vendor of the device (only used for wiki and startup logging)
    description: 'TODO ', // Description of the device, copy from vendor site. (only used for wiki and startup logging)
    supports: 'TODO', // Actions this device supports (only used for wiki)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```
Because it seems to use ```ssIasZone``` we have to look deeper.

mattclar, 2018-07-24:
I have added a similar section to devices.js already :/

tb-killa, 2018-07-25:
Please test this...
Maybe we could add later reporting for ``genDeviceTempCfg`` and ``currentTemperature``.

```
{ 
    zigbeeModel: ['SWO-MOS1PA'], 
    model: 'TODO', // Vendor model number, look on the device for a model number
    vendor: 'SwannONe', // Vendor of the device (only used for wiki and startup logging)
    description: 'TODO ', // Description of the device, copy from vendor site. (only used for wiki and startup logging)
    supports: 'TODO', // Actions this device supports (only used for wiki)
    fromZigbee: [fz.SwannONe_ssIasZone],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```

and inside ``node_modules/zigbee-shepherd-converters/converters/fromZigbee.js``

```
SwannONe_ssIasZone: {
   cid: 'ssIasZone',
    type: 'attReport',
    convert: (model, msg, publish, options) => {
        console.log(msg.data)
    },
},
```
and provide log via ```DEBUG=zigbee-shepherd* npm start```

Maybe we got Reporting if Motion is detected ??

mattclar, 2018-07-25:
```
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-shepherd* npm start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-7-25 22:32:12 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-7-25 22:32:13 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-25 22:32:13 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +200ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +359ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +63ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +412ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +21ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +38ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +19ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +50ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +21ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +42ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +32ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +38ms
  zigbee-shepherd:init Start the ZNP as a coordinator... +8ms
  zigbee-shepherd:request REQ --> ZDO:startupFromApp +7ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:stateChangeInd +848ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +10ms
  zigbee-shepherd:request REQ --> SYS:version +7ms
  zigbee-shepherd:request RSP <-- SYS:version +3s
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +46ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +34ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +31ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +30ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +8ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +29ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +29ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +10ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +28ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +37ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +75ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +30ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +57ms
  zigbee-shepherd:request REQ --> AF:register +55ms
  zigbee-shepherd:request RSP <-- AF:register, status: 0 +36ms
  zigbee-shepherd:init Register delegator, epId: 1, profId: 260  +19ms
  zigbee-shepherd:request REQ --> AF:register +5ms
  zigbee-shepherd:request RSP <-- AF:register, status: 0 +34ms
  zigbee-shepherd:init Register delegator, epId: 2, profId: 257  +17ms
  zigbee-shepherd:request REQ --> AF:register +4ms
  zigbee-shepherd:request RSP <-- AF:register, status: 0 +34ms
  zigbee-shepherd:init Register delegator, epId: 3, profId: 261  +16ms
  zigbee-shepherd:request REQ --> AF:register +4ms
  zigbee-shepherd:request RSP <-- AF:register, status: 0 +35ms
  zigbee-shepherd:init Register delegator, epId: 4, profId: 263  +18ms
  zigbee-shepherd:request REQ --> AF:register +3ms
  zigbee-shepherd:request RSP <-- AF:register, status: 0 +35ms
  zigbee-shepherd:init Register delegator, epId: 5, profId: 264  +16ms
  zigbee-shepherd:request REQ --> AF:register +4ms
  zigbee-shepherd:request RSP <-- AF:register, status: 0 +33ms
  zigbee-shepherd:init Register delegator, epId: 6, profId: 265  +17ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +13ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +58ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +8ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +55ms
  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +35ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:permitJoinInd +72ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +4ms
  zigbee-shepherd:init Loading devices from database done. +558ms
[... 153 lines cut ...]
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +4ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +5m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +63ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +4ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +4m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +65ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +5ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +7m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +52ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +5ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms

```

mattclar, 2018-09-16:
still trying to get this working but havent made much progress

tb-killa, 2018-09-16:
Hi please test this and report via ``DEBUG=zigbee-shepherd* npm start `` (pastebin or other)

```
{ 
    zigbeeModel: ['SWO-MOS1PA'], 
    model: 'Swann SWO-MOS1PA One Motion Sensor',
    vendor: 'Swann', 
    description: '',
    supports: '', 
    fromZigbee: [],
    toZigbee: [],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
         const device = shepherd.find(ieeeAddr, 1);
         const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
            ];
             execute(device, actions, callback);

        }, 
},
```

closed 2018-11-22 by stale[bot]

loglux, 2019-12-05:
I've got a pack of Swann devices: two motion and temperature sensors, door/windows and temperature, smart plug with energy monitoring and key. I need a support for motion sensor and plug

I've put in devices.js
{
    zigbeeModel: ['SWO-MOS1PA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SWO-MOS1PA', // Vendor model number, look on the device for a model number
    vendor: 'Swann', // Vendor of the device (only used for documentation and startup logging)
    description: 'Swann Motion and Temperature Sensor ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'motion and temperature', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},

And now have a following message in the console output:
'''Dec 05 00:39:43 littlehelp npm[18124]: zigbee2mqtt:warn  2019-12-05T00:39:43: No converter available for 'SWO-MOS1PA' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"zonestatus":49,"extendedstatus":0}''''

I've no idea is that temperature or motion. Zones, statuses... I've got no clue....

loglux, 2019-12-05:
Changed to
```

{
    zigbeeModel: ['SWO-MOS1PA'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SWO-MOS1PA', // Vendor model number, look on the device for a model number
    vendor: 'SwannONe', // Vendor of the device (only used for documentation and startup logging)
    description: 'Swann Motion and Temperature Sensor ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'motion and temperature', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```

loglux, 2019-12-05:
I guess the device use the same set of parameters as other motion sensors. 
Could you help me to make it work?

loglux, 2019-12-06:
> I am trying to get this motion detector woring, I have two of them..
> https://www.swann.com/au/swo-mos1pa

Hello. I have progress, linking Swann Motion Sensor.
I've opened a new topic because your's is marked stale. 
https://github.com/Koenkk/zigbee2mqtt/issues/2470
Could you take a look?

MattEng0, 2020-06-06:
> 
> 
> > I am trying to get this motion detector woring, I have two of them..
> > https://www.swann.com/au/swo-mos1pa
> 
> Hello. I have progress, linking Swann Motion Sensor.
> I've opened a new topic because your's is marked stale.
> #2470
> Could you take a look?

Also linked in new issue: #3678