Koenkk/zigbee2mqtt #216: Option to change zigbee channel for routers

aidbish, 2018-07-26:
It would be good if the routers were able to operate on a different zigbee channel than the hard-coded channel 11 they use at the moment
this would allow you to set a channel that gets less interference with your or neighbours WIFI channels.

For me zigbee channel 11 is covered by WiFi channel 1, which i have to use to get decent Wifi in areas around the house that is not interfered with by neighbours WiFi signals.

See this article here for information on the zigbee and WiFi channels and how they overlap and interact.
https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

ptvoinfo, 2018-07-26:
@aidbish, I'll try to do this after my summer vacation time.

aidbish, 2018-07-26:
@ptvoinfo much appreciated mate

tb-killa, 2018-07-26:
If you look at the wiki page: https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration
There is the Option for ``data/configuration.yaml`` at 
```# Optional: advanced settings
advanced:
# Optional: ZigBee channel
  channel: 11
```

ryanbeaton, 2018-07-26:
@tb-killa yes but the router firmware by @ptvoinfo is hard coded to channel 11

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ptvoinfo, 2018-07-26:
@ryanbeaton, the channel number is not hard coded. But if a router cannot connect using the channel 11, it increments panId. This behavior of ZStack (not my code). It is necessary to dig deeply to the sources to find and fix this problem.

ryanbeaton, 2018-07-26:
@ptvoinfo oh that’s what I read on your page. That’s why I though @aidbish must have been asking. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ptvoinfo, 2018-07-26:
@ryanbeaton, I've found this after my comment :)

aidbish, 2018-10-04:
Hi, 
was there  any update on changing the zigbee channel for the routers?

ryanbeaton, 2018-10-04:
https://github.com/Koenkk/zigbee2mqtt/issues/252#issuecomment-424951388

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2018-12-10 by stale[bot]