Koenkk/zigbee2mqtt #220: Failed to start

jeroenterheerdt, 2018-07-27:
I am trying to get the zigbee2mqtt gateway to work. I am getting a consistent error though:

`
2018-7-27 13:44:24 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
2018-7-27 13:44:24 INFO Starting zigbee-shepherd
2018-7-27 13:44:32 INFO Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-7-27 13:45:32 INFO Starting zigbee-shepherd
2018-7-27 13:45:41 ERROR Error while starting zigbee-shepherd!
2018-7-27 13:45:41 ERROR Press the reset button on the stick (the one closest to the USB) and start again
2018-7-27 13:45:41 ERROR Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:272:13)\n    at CcZnp.emit (events.js:180:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:466:11)\n    at tryOnTimeout (timers.js:304:5)\n    at Timer.listOnTimeout (timers.js:267:5)"}
`

My config:
`
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://XXXXX'
  # MQTT server authentication, uncomment if required:
  user: XXXXX
  password: XXXXX
# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM2
`

My MQTT server is running https, maybe there is an issue here?
I checked the usb device by running `ls -l /dev/serial/by-id`:
` usb-Texas_Instruments_TI_CC2531_USB_CDC___0XD0B5C2ED60F3FFFF-if00 -> ../../ttyACM2`

I just flashed the usb stick today.

Koenkk, 2018-07-27:
CC2540 is a bluetooth chip not a Zigbee chip.

jeroenterheerdt, 2018-07-27:
I figured so much, apparently ordered the wrong one.

closed 2018-08-06 by jeroenterheerdt

jerrychong25, 2019-05-10:
> CC2540 is a bluetooth chip not a Zigbee chip. http://www.ti.com/product/CC2540

Thanks for the info.