Koenkk/zigbee2mqtt #225: Sercomm XHS2-SE Door/Window Sensor

carpenike, 2018-07-27:
I recently got a large number of these door/window sensors from a failed Xfinity install. Based on the supported devices it doesn't appear that anyone has attempted using these devices yet. Has anyone tried?

https://www.amazon.com/CELTRALITE-XHS2-SE-Door-Window-Sensor/dp/B01EEL80S4

I have the parts coming hopefully next week and will go through the process for an unsupported device to see if I can get it working.

Koenkk, 2018-08-05:
Good, let us know if you need any help :)

carpenike, 2018-08-18:
Alright finally got my CC Debugger... that took forever! So I started playing with this last night.

I'm using docker to run the service with the dongle passed through via --device. I submitted a PR to add support for the self signed MQTT certificate so still using my local copy until that gets pushed to docker hub. Sensor is around 2-3 feet from the CC2531 device.

`docker run -d --name zigbee2mqtt --hostname zigbee2mqtt -v /mnt/perf/docker/zigbee2mqtt:/app/data --device /dev/ttyACM0 zigbee2mqttcarp:latest`

Step 1: Pair the unknown device with zigbee2mqtt

```
ryan@adonis:/mnt/perf/docker/zigbee2mqtt$ docker logs -f zigbee2mqtt
Using '/app/data' as data directory

> zigbee2mqtt@0.1.2 start /app
> node index.js

2018-8-18 14:23:33 INFO Starting zigbee2mqtt version 0.1.2 (commit #)
2018-8-18 14:23:33 INFO Starting zigbee-shepherd
2018-8-18 14:23:33 INFO zigbee-shepherd started
2018-8-18 14:23:33 INFO Currently 0 devices are joined:
2018-8-18 14:23:33 WARN `permit_join` set to  `true` in configuration.yaml.
2018-8-18 14:23:33 WARN Allowing new devices to join.
2018-8-18 14:23:33 WARN Set `permit_join` to `false` once you joined all devices.
2018-8-18 14:23:33 INFO Zigbee: allowing new devices to join.
2018-8-18 14:23:33 INFO Connecting to MQTT server at mqtts://mqtt.holthome.net
2018-8-18 14:23:34 INFO zigbee-shepherd ready
2018-8-18 14:23:34 INFO Connected to MQTT server
2018-8-18 14:23:34 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-8-18 14:24:05 INFO Connected to MQTT server
2018-8-18 14:24:05 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
spinlock: false []
2018-8-18 14:24:26 WARN Message without device!
2018-8-18 14:24:29 INFO Device incoming...
2018-8-18 14:24:29 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-8-18 14:24:29 WARN Device with modelID 'XHS2-SE' is not supported.
2018-8-18 14:24:29 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-8-18 14:24:29 WARN Device with modelID 'XHS2-SE' is not supported.
2018-8-18 14:24:29 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-8-18 14:25:00 INFO Connected to MQTT server
2018-8-18 14:25:00 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
```

Step 2: Exec into the docker image and adjust the devices file:

```
ryan@adonis:/mnt/perf/docker/zigbee2mqtt$ docker exec -it zigbee2mqtt sh
/app # vi node_modules/zigbee-shepherd-converters/devices.js
```

Adjust the end of the file so that it looks like the below:

```
    // Gledopto
    {
        zigbeeModel: ['GLEDOPTO'],
        model: 'GL-C-008',
        vendor: 'Gledopto',
        description: 'Zigbee LED controller RGB + CCT',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
    },

    // SmartThings
    {
        zigbeeModel: ['PGC313'],
        model: 'STSS-MULT-001',
        vendor: 'SmartThings',
        description: 'SmartSense multi sensor',
        supports: 'contact',
        fromZigbee: [fz.smartthings_contact],
        toZigbee: [],
    },

    // Sercomm
    {
        zigbeeModel: ['XHS2-SE'],
        model: 'XHS2-SE',
        vendor: 'Sercomm',
        description: 'Door and Window Sensor',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
    },
];

module.exports = devices;
```

Step 4: Restart the docker image and attempt to trigger an action by moving the magnet infront of the device, but nothing registers in the log at all.

```
Using '/app/data' as data directory

> zigbee2mqtt@0.1.2 start /app
> node index.js

2018-8-18 14:32:45 INFO Starting zigbee2mqtt version 0.1.2 (commit #)
2018-8-18 14:32:45 INFO Starting zigbee-shepherd
2018-8-18 14:32:46 INFO zigbee-shepherd started
2018-8-18 14:32:46 INFO Currently 0 devices are joined:
2018-8-18 14:32:46 WARN `permit_join` set to  `true` in configuration.yaml.
2018-8-18 14:32:46 WARN Allowing new devices to join.
2018-8-18 14:32:46 WARN Set `permit_join` to `false` once you joined all devices.
2018-8-18 14:32:46 INFO Zigbee: allowing new devices to join.
2018-8-18 14:32:46 INFO Connecting to MQTT server at mqtts://mqtt.holthome.net
2018-8-18 14:32:46 INFO zigbee-shepherd ready
2018-8-18 14:32:46 INFO Connected to MQTT server
2018-8-18 14:32:46 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-8-18 14:33:17 INFO Connected to MQTT server
2018-8-18 14:33:17 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-8-18 14:33:48 INFO Connected to MQTT server
2018-8-18 14:33:48 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-8-18 14:34:19 INFO Connected to MQTT server
2018-8-18 14:34:19 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
```

Any thoughts?

carpenike, 2018-08-18:
Ah, realized there's a dev branch that got updated in docker hub yesterday! Updated my docker run to the following:

`docker run -d --name zigbee2mqtt --hostname zigbee2mqtt -v /mnt/perf/docker/zigbee2mqtt:/app/data --device /dev/ttyACM0 koenkk/zigbee2mqtt:latest-dev`

Cleaned out the devices listed in the configuration.yaml and went through the steps again.

Step 1: Reset device:

```
ryan@adonis:/mnt/perf/docker/zigbee2mqtt$ docker logs -f zigbee2mqtt
Using '/app/data' as data directory

> zigbee2mqtt@0.1.3 start /app
> node index.js

  zigbee2mqtt:info 2018-8-18 14:48:17 Logging to directory: '/app/data/log/2018-08-18.14:48:17'
  zigbee2mqtt:info 2018-8-18 14:48:18 Starting zigbee2mqtt version 0.1.3 (commit #4ee7845)
  zigbee2mqtt:info 2018-8-18 14:48:18 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-8-18 14:48:18 zigbee-shepherd started
  zigbee2mqtt:info 2018-8-18 14:48:18 Coordinator firmware version: '20180507'
  zigbee2mqtt:info 2018-8-18 14:48:18 Currently 0 devices are joined:
  zigbee2mqtt:warn 2018-8-18 14:48:18 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-8-18 14:48:18 Allowing new devices to join.
  zigbee2mqtt:warn 2018-8-18 14:48:18 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-8-18 14:48:18 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-8-18 14:48:18 Connecting to MQTT server at mqtts://mqtt.holthome.net
  zigbee2mqtt:info 2018-8-18 14:48:18 zigbee-shepherd ready
  zigbee2mqtt:info 2018-8-18 14:48:18 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 14:48:18 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 14:48:49 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 14:48:49 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 14:49:20 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 14:49:20 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 14:49:51 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 14:49:51 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 14:50:07 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:07 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:11 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:11 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2018-8-18 14:50:13 Message without device!
  zigbee2mqtt:info 2018-8-18 14:50:13 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:13 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:14 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:14 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:15 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:15 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:16 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:16 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:17 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:17 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:17 Connecting with device...
  zigbee2mqtt:info 2018-8-18 14:50:17 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-8-18 14:50:17 Device incoming...
  zigbee2mqtt:info 2018-8-18 14:50:17 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-8-18 14:50:17 New device with address 0x000d6f000fff6bd5 connected!
  zigbee2mqtt:info 2018-8-18 14:50:17 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000fff6bd5"}'
  zigbee2mqtt:warn 2018-8-18 14:50:17 Device with modelID 'XHS2-SE' is not supported.
  zigbee2mqtt:warn 2018-8-18 14:50:17 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:warn 2018-8-18 14:50:17 Device with modelID 'XHS2-SE' is not supported.
  zigbee2mqtt:warn 2018-8-18 14:50:17 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

However, after updating the devices.js file I don't see any feedback making it into the logs after initiating actions on the device (same as before). I have about 15 of these that I've tried and they all exhibit the same behavior.

tb-killa, 2018-08-18:
There must be some sort of error in your devices.js extension because in the log there is

14:50:17 New device with address 0x000d6f000fff6bd5 connected!
  zigbee2mqtt:info 2018-8-18 14:50:17 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000fff6bd5"}'
  zigbee2mqtt:warn 2018-8-18 14:50:17 Device with modelID 'XHS2-SE' is not supported.

Also please Start docker via debug environment as there we could Show what states are transported!

carpenike, 2018-08-18:
Sounds good. Deleted database file, cleaned up configuration.yaml (devices), deleted the docker container and started fresh.

1. New Docker Run:

`ryan@adonis:/mnt/perf/docker/zigbee2mqtt$ docker run -d --name zigbee2mqtt --hostname zigbee2mqtt -v /mnt/perf/docker/zigbee2mqtt:/app/data --device /dev/ttyACM0 -e DEBUG=true koenkk/zigbee2mqtt:latest-dev`

2. Paired device with zigbee2mqtt:

```
ryan@adonis:/mnt/perf/docker/zigbee2mqtt$ docker logs -f zigbee2mqtt
Using '/app/data' as data directory

> zigbee2mqtt@0.1.3 start /app
> node index.js

  zigbee2mqtt:info 2018-8-18 15:03:56 Logging to directory: '/app/data/log/2018-08-18.15:03:56'
  zigbee2mqtt:debug 2018-8-18 15:03:56 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/app/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2018-8-18 15:03:56 Loaded state from file /app/data/state.json
  zigbee2mqtt:info 2018-8-18 15:03:56 Starting zigbee2mqtt version 0.1.3 (commit #4ee7845)
  zigbee2mqtt:info 2018-8-18 15:03:56 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-8-18 15:03:57 zigbee-shepherd started
  zigbee2mqtt:info 2018-8-18 15:03:57 Coordinator firmware version: '20180507'
  zigbee2mqtt:debug 2018-8-18 15:03:57 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014d9e1be","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180507},"startTime":1534604637,"joinTimeLeft":0}
  zigbee2mqtt:info 2018-8-18 15:03:57 Currently 0 devices are joined:
  zigbee2mqtt:warn 2018-8-18 15:03:57 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-8-18 15:03:57 Allowing new devices to join.
  zigbee2mqtt:warn 2018-8-18 15:03:57 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-8-18 15:03:57 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-8-18 15:03:57 Connecting to MQTT server at mqtts://mqtt.holthome.net
  zigbee2mqtt:debug 2018-8-18 15:03:57 MQTT reject_unauthorized set false, ignoring certificate warnings.
  zigbee2mqtt:info 2018-8-18 15:03:57 zigbee-shepherd ready
  zigbee2mqtt:info 2018-8-18 15:03:57 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:03:57 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:error 2018-8-18 15:04:27 Not connected to MQTT server!
  zigbee2mqtt:info 2018-8-18 15:04:28 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:04:28 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:debug 2018-8-18 15:04:38 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:38 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:38 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:42 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:42 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:42 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:44 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:warn 2018-8-18 15:04:44 Message without device!
  zigbee2mqtt:debug 2018-8-18 15:04:44 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:44 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:44 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:45 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:45 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:45 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:45 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:45 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:45 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:45 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:45 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:45 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:46 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:46 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:46 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:47 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:47 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:04:47 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:04:47 Recieved zigbee message with data "0x000d6f000f8b127c"
  zigbee2mqtt:info 2018-8-18 15:04:47 Device incoming...
[... cut ...]
  zigbee2mqtt:info 2018-8-18 15:07:14 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:07:14 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:07:45 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:07:45 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:08:16 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:08:16 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
```

carpenike, 2018-08-18:
Also, pairing another one of the sensors does not pop up the same unsupported message:

```
Using '/app/data' as data directory

> zigbee2mqtt@0.1.3 start /app
> node index.js

  zigbee2mqtt:info 2018-8-18 15:07:13 Logging to directory: '/app/data/log/2018-08-18.15:07:13'
  zigbee2mqtt:debug 2018-8-18 15:07:13 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/app/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2018-8-18 15:07:13 Loaded state from file /app/data/state.json
  zigbee2mqtt:info 2018-8-18 15:07:13 Starting zigbee2mqtt version 0.1.3 (commit #4ee7845)
  zigbee2mqtt:info 2018-8-18 15:07:13 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-8-18 15:07:14 zigbee-shepherd started
  zigbee2mqtt:info 2018-8-18 15:07:14 Coordinator firmware version: '20180507'
  zigbee2mqtt:debug 2018-8-18 15:07:14 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014d9e1be","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180507},"startTime":1534604834,"joinTimeLeft":0}
  zigbee2mqtt:info 2018-8-18 15:07:14 Currently 0 devices are joined:
  zigbee2mqtt:warn 2018-8-18 15:07:14 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-8-18 15:07:14 Allowing new devices to join.
  zigbee2mqtt:warn 2018-8-18 15:07:14 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-8-18 15:07:14 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-8-18 15:07:14 Connecting to MQTT server at mqtts://mqtt.holthome.net
  zigbee2mqtt:debug 2018-8-18 15:07:14 MQTT reject_unauthorized set false, ignoring certificate warnings.
  zigbee2mqtt:info 2018-8-18 15:07:14 zigbee-shepherd ready
  zigbee2mqtt:info 2018-8-18 15:07:14 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:07:14 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:07:45 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:07:45 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:08:16 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:08:16 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:08:47 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:08:47 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:09:18 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:09:18 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:09:49 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:09:49 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-8-18 15:10:20 Connected to MQTT server
  zigbee2mqtt:info 2018-8-18 15:10:20 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:debug 2018-8-18 15:10:36 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:36 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:36 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:40 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:40 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:40 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:42 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:warn 2018-8-18 15:10:42 Message without device!
  zigbee2mqtt:debug 2018-8-18 15:10:42 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:42 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:42 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:43 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:43 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:43 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:44 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:44 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:44 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:44 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:44 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:44 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:45 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:45 Connecting with device...
  zigbee2mqtt:info 2018-8-18 15:10:45 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2018-8-18 15:10:45 Recieved zigbee message with data "0x000d6f000fff6bd5"
  zigbee2mqtt:info 2018-8-18 15:10:45 Connecting with device...
[... cut ...]
  zigbee2mqtt:info 2018-8-18 15:10:45 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000fff6bd5"}'
  zigbee2mqtt:debug 2018-8-18 15:10:45 Recieved zigbee message with data "online"
```

However, it's also quiet with any of the state changes.

Here's the devices section of the configuration.yaml:

```
devices:
  '0x000d6f000f8b127c':
    friendly_name: '0x000d6f000f8b127c'
    retain: false
  '0x000d6f000fff6bd5':
    friendly_name: '0x000d6f000fff6bd5'
    retain: false
```

tb-killa, 2018-08-18:
Please provide full log via pastebin or another Service.

Does some no convert messages comes up?

carpenike, 2018-08-18:
Nothing with convert in it...

```
ryan@adonis:/mnt/perf/docker/zigbee2mqtt/log$ grep -R convert *
ryan@adonis:/mnt/perf/docker/zigbee2mqtt/log$
```

https://pastebin.com/raw/LsEq1ShE

tb-killa, 2018-10-01:
Hi @carpenike 
Do you actually working on this ?
I think i found a way to get your stuff working.
Give me some hints if we both could do some tests.

carpenike, 2018-10-01:
I'm still interested in getting it working, but have been playing with other HomeAssistant components.

tb-killa, 2018-10-02:
Please test this and report.

```
{
        zigbeeModel: ['XHS2-SE'], 
        model: 'XHS2-SE',
        vendor: 'Sercomm', 
        description: '',
        supports: '', 
        fromZigbee: [],
        toZigbee: [],
	configure: (ieeeAddr, shepherd, coordinator, callback) => {
	const device = shepherd.find(ieeeAddr, 1);
	const writeRecArgs_ssIasZone = {
                attrId: 16,
                dataType: 'ieeeAddr',
                attrData: coordinator.device.getIeeeAddr()
        };
	const enrollRspArgs_ssIasZone = {
                enrollrspcode: 0x00,
                zoneid : 1 //This has to modify +1 for every new iaszone device
        };
	const actions = [
                (cb) => device.foundation('ssIasZone', 'write', [ writeRecArgs_ssIasZone ], { direction: 0}, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', [ enrollRspArgs_ssIasZone ], cb),
                (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
            ];
             execute(device, actions, callback);
        },
    },
```

carpenike, 2018-10-02:
Got further!

Now this pops up in the error logs. Attempted with multiple sensors```
.

  zigbee2mqtt:info 2018-10-2 11:52:14 New device with address 0x000d6f000f8b126e connected!
  zigbee2mqtt:info 2018-10-2 11:52:14 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000f8b126e"}'
  zigbee2mqtt:error 2018-10-2 11:52:14 Failed to configure 0x000d6f000f8b126e 0x000d6f000f8b126e
```

carpenike, 2018-10-03:
@tb-killa : I haven't touched the zoneid before. Do I increment that based on some other data?

tb-killa, 2018-10-03:
You have to define one zoneid for every single device.
sadly actually with this type of code you have to to them manually, i´m thinking about some sort of function who could do them automatic.

you could start with one device paired with zoneid: 20 and report back ..

carpenike, 2018-10-03:
Ahhh, so update the devices.js file each time a new device is paired?

carpenike, 2018-10-03:
Set the number to 21 and now get this error:

```
  zigbee2mqtt:debug 2018-10-3 19:17:12 Recieved zigbee message of type 'devIncoming' with data '"0x000d6f000fff6bd5"' of device 'XHS2-SE' (0x000d6f000fff6bd5)
  zigbee2mqtt:info 2018-10-3 19:17:12 Device incoming...
  zigbee2mqtt:info 2018-10-3 19:17:12 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-10-3 19:17:12 New device with address 0x000d6f000fff6bd5 connected!
  zigbee2mqtt:info 2018-10-3 19:17:12 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000fff6bd5"}'
  zigbee2mqtt:error 2018-10-3 19:17:12 Failed to configure 0x000d6f000fff6bd5 0x000d6f000fff6bd5
  zigbee2mqtt:debug 2018-10-3 19:17:12 Recieved zigbee message of type 'devStatus' with data '"online"' of device 'XHS2-SE' (0x000d6f000fff6bd5)
```

closed 2018-12-09 by stale[bot]

coolguymatt, 2019-09-16:
@carpenike - were you ever able to figure this out?