Koenkk/zigbee2mqtt #227: Getting current status of Tradfri bulb after Home Assistant restart

stratplayer, 2018-07-29:
I am still configuring my HA and having to restart many times, everytime I do so the Tradfri bulb I have connected in the lamp which is on before I restart HA shows up as off after I restart. The only way I can get it to appear in the correct state is to turn the switch on in HA where it then shows correctly. On a couple of other MQTT devices I set up an automation to ask for the status, I tried to do the same with this but it didnt work.

- alias: "Ikea Lamp status after HA restart"
  trigger:
    platform: homeassistant
    event: start
  action:
    - service: mqtt.publish
      data:
        topic: "zigbee2mqtt/0x90fd9ffffe8c6523/set"
        payload: ""

ryanbeaton, 2018-07-29:
Might have to retain status in MQTT.

https://github.com/Koenkk/zigbee2mqtt/issues/139#issuecomment-399661976

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Koenkk, 2018-07-29:
As @ryanbeaton mentioned, set `retain: true` for this device: https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration

stratplayer, 2018-07-29:
Hi thanks guys, sorry to have been such a numpty! Working fine now

closed 2018-07-29 by stratplayer