Koenkk/zigbee2mqtt #228: What about the alert function? (Pulsating lights for a few seconds)

jboeddeker, 2018-07-29:
First of all, thanks for all the work!

I'm using zigbee2mqtt for a few days and i like it very much. I already migrated my Osram Smart Plugs to the bridge and added a some other sensors. 

But one thing keeps me from removing the hue bridge altogether: i am missing the Alert function of the  light bulbs. I currently use it as an additional signal of the doorbell, this works pretty well. So well iwouldn't enjoy missing it.

Is this a feature of the hue bridge or a feature of the bulbs? Either way, could this be implemented? 

Thanks for any input.

tb-killa, 2018-07-30:
I think its same like https://github.com/Koenkk/zigbee2mqtt/issues/15
This scenario (Alert function) should be created in your home automation software.

tb-killa, 2018-07-30:
@jboeddeker Give some example how this "feature" works, maybe we could build some sort of this as a "feature" as tozigbee converter ;) ? I need some more detail informations.

jboeddeker, 2018-07-30:
Thanks @tb-killa. 
I found some groovy code related to smartthings, which looks rather technical (the line with "st cmd..."). Maybe that's the zigbee-message/register to be set?

`def alert(action) {`
`	def value = "00"`
`	def valid = true`
`	switch(action) {`
`		case "Blink":`
`			value = "00"`
`			break`
`		case "Breathe":`
`			value = "01"`
`			break`
`		case "Okay":`
`			value = "02"`
`			break`
`		case "Stop":`
`			value = "ff"`
`			break`
`		default:`
`			valid = false`
`			break`
`	}`
`	if (valid) {`
`		log.debug "Alert: ${action}, Value: ${value}"`
`		sendEvent(name: "alert", value: action)`
`		"st cmd 0x${device.deviceNetworkId} ${endpointId} 3 0x40 {${value} 00}"`
`	}`
`	else { log.debug "Invalid action" }`
`}`

jboeddeker, 2018-07-30:
Here is another link, which makes me believe, it's a bulb hardware feature (also smartthings):

[https://community.smartthings.com/t/home-monitor-alert-with-lights-blinking/45116/5](https://community.smartthings.com/t/home-monitor-alert-with-lights-blinking/45116/5)

tb-killa, 2018-07-30:
Its okay at this place.
I has to buy one bulb to test.
Will privide so Code and you have to test :)

jboeddeker, 2018-09-11:
@tb-killa, any news on that? 

Maybe could you lead me into the direction how to add such a message, then i could add it by myself. I know how to write code, but do not know zigbee, zigbee-shepherd & zigbee2mqtt. Does something similar exist?

d-EScape, 2018-11-11:
Would be great to see the alert effect implemented. I use it my alarm system with 8 lights (no way that's going unnoticed;). A script on the home automation side will choke on that many transitions, but with the Philips bulbs handling the blinking themselves it works fine.

ebaauw seems to have figured out the Zigbee side of this command:
_"On the Hue bridge, setting state.alert to select results in two Trigger effect (0x0003/0x40) commands. The first with parameters effectIdentifier Breathe (0x01) and effectVariant Default (0x00). The second with Finish effect (0xFE) and, again, Default (0x00). The second command is sent immediately (30ms) after the first. Setting state.alert to lselect results in only the first command, which causes the light to breathe for 15s. Setting state.alert to none results in only the second command."_

source:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/100#issuecomment-322172207

d-EScape, 2018-12-03:
Submitted a pull request to Zigbee-shepherd-converts to add support for alert
/Koenkk/zigbee-shepherd-converters#157

Koenkk, 2018-12-05:
@d-EScape is merged, closing this.

closed 2018-12-05 by Koenkk

augisbud, 2019-02-08:
> Thanks @tb-killa.
> I found some groovy code related to smartthings, which looks rather technical (the line with "st cmd..."). Maybe that's the zigbee-message/register to be set?
> 
> ```groovy
> def alert(action) {
>     def value = "00"
>     def valid = true
>     switch(action) {
>         case "Blink":
>             value = "00"
>             break
>         case "Breathe":
>             value = "01"
>             break
>         case "Okay":
>             value = "02"
>             break
>         case "Stop":
>             value = "ff"
>             break
>         default:
>             valid = false
>             break
>     }
>     if (valid) {
>         log.debug "Alert: ${action}, Value: ${value}"
>         sendEvent(name: "alert", value: action)
>         "st cmd 0x${device.deviceNetworkId} ${endpointId} 3 0x40 {${value} 00}"
>     }
>     else { log.debug "Invalid action" }
> }
> ```
> 
> source: https://community.smartthings.com/t/enhanced-zigbee-hue-bulb-direct-connect-to-st/33013
> 
> Update:
> Here is another link, which makes me believe, it's a bulb hardware feature ("blinking is handled at the hardware level") (also smartthings):
> 
> https://community.smartthings.com/t/home-monitor-alert-with-lights-blinking/45116/5
> 
> This looks like some official specification, chapter 7, page 112 describes the effects:
> 
> https://www.nxp.com/docs/en/user-guide/JN-UG-3103.pdf
> 
> From the effect description i use the "Breathe" effect. Just checked my lights are flashing 15 times.
> 
> All my lights support this, i own the following:
> 
> * innr RB185C
> * innr RB162
> * Osram Lightify Surface Light TW
> * Osram Lightify Indor Flex RGBW
> * Ikea Tradfri E27 opal 1000lm
> * Philips Hue LWB010
> 
> @tb-killa, is this the correct repository for the issue? Shouldn't it be moved to the converter repository https://github.com/Koenkk/zigbee-shepherd-converters?

how did you pair the Osram Lightify Surface Light TW to zigbee2mqtt?