Koenkk/zigbee2mqtt #229: router devices can't be managed by zigbee2mqtt

marcgarciamarti, 2018-07-30:
Hello,

in my setup, EndDevices work ok (wireless switch, accumpancy sensor, door sensor), but router devices won't. I receive information comming from those routers just fine (direction devices --> usb stick) but when I attempt to command them, they will just not act as I say, but rather Zigbee2mqtt responds with an error message:

```
2018-7-30 10:37:42 INFO Switching log level to 'debug'
2018-7-30 10:37:50 DEBUG Check online 0x000b57fffedc1058
2018-7-30 10:37:50 DEBUG Check online 0x90fd9ffffe4bf6fd
2018-7-30 10:37:50 DEBUG Check online 0x00158d00022f35ca
2018-7-30 10:37:56 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/power-plug.1.xiaomi/set' with data 'ON'
2018-7-30 10:37:56 WARN Cannot handle MQTT message with topic 'zigbee2mqtt/power-plug.1.xiaomi/set' and message 'ON'
2018-7-30 10:38:50 DEBUG Check online 0x000b57fffedc1058
2018-7-30 10:38:50 DEBUG Check online 0x90fd9ffffe4bf6fd
2018-7-30 10:38:50 DEBUG Check online 0x00158d00022f35ca
2018-7-30 10:39:50 DEBUG Check online 0x000b57fffedc1058
2018-7-30 10:39:50 DEBUG Check online 0x90fd9ffffe4bf6fd
2018-7-30 10:39:50 DEBUG Check online 0x00158d00022f35ca
2018-7-30 10:40:50 DEBUG Check online 0x000b57fffedc1058
2018-7-30 10:40:50 DEBUG Check online 0x90fd9ffffe4bf6fd
2018-7-30 10:40:50 DEBUG Check online 0x00158d00022f35ca
2018-7-30 10:40:55 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"3":34,"5":12,"7":"0x0000000000000000","8":4886,"9":256,"100":0,"149":0.0014568129554390907,"150":2240,"152":0,"154":16}}}
2018-7-30 10:40:55 INFO MQTT publish, topic: 'zigbee2mqtt/power-plug.1.xiaomi', payload: '{"state":"OFF","power":0,"voltage":224,"consumption":0,"temperature":34}'
2018-7-30 10:40:55 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,null,null,12,null,"0x0000000000000000",4886,256,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,2240,null,0,null,16]}}
2018-7-30 10:41:50 DEBUG Check online 0x000b57fffedc1058
```

The relevant information is 

**2018-7-30 10:37:56 WARN Cannot handle MQTT message with topic 'zigbee2mqtt/power-plug.1.xiaomi/set' and message 'ON'**

additionally, I find the message below a bit suspicious:

**2018-7-30 10:40:55 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,null,null,12,null,"0x0000000000000000",4886,256,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,2240,null,0,null,16]}}**

Thanks,

marcgarciamarti, 2018-07-30:
> Does it work when changing the friendly name to: powerplug1xiaomi ?

Do you mean by removing the dots?

This is what it looks like in my configuration file:

  '0x00158d00026765c4':
>     friendly_name: kitchen.door-sensor.xiaomi
>     retain: false

Will try to remove the dots in 1h and report back.

thanks

marcgarciamarti, 2018-07-30:
Sorry, wrong device

  '0x00158d00022f35ca':
>     friendly_name: power-plug.1.xiaomi
>     retain: false

I will now get to it with removing the dots :)

marcgarciamarti, 2018-07-30:
Man, I bow down to you!! You solved it!!!

Many thanks!!

tb-killa, 2018-07-30:
What do you do now in Detail?
Remove dots?

marcgarciamarti, 2018-07-30:
yes, I simplified and shortened the friendly names for all my devices,
making sure they do not have dots or "-".

marcgarciamarti, 2018-07-30:
@koenkk, you can tag this one as solved. Any chance I could mention this in the documentation? Either that or can you please harden the code?

Koenkk, 2018-07-31:
No, this bug should be solved :)

closed 2018-12-22 by stale[bot]