Koenkk/zigbee2mqtt #234: Multiple entries per device in database.db

wouterbulten, 2018-07-30:
I'm encountering some issues with my Tradfri lights not corresponding to updates. While debugging I found that my database.db has multiple entries (2-4) per device. Is this normal behaviour? And could this be a cause of my unstable system?

To me it seems that some of the entries are not complete. For example, the entries of a IKEA bulb:

```
{"id":22,"type":"Router","ieeeAddr":"0x000b57fffed6a107","nwkAddr":45810,"manufId":4476,"epList":[1],"status":"online","joinTime":1532868992,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{}}},"_id":"mkxtflzTZxkWHL0v"}
{"id":22,"type":"Router","ieeeAddr":"0x000b57fffed6a107","nwkAddr":45810,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"status":"online","joinTime":1532868992,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{}}},"_id":"mkxtflzTZxkWHL0v"}
{"id":22,"type":"Router","ieeeAddr":"0x000b57fffed6a107","nwkAddr":45810,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb GU10 W 400lm","epList":[1],"status":"online","joinTime":1532868992,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb GU10 W 400lm","dateCode":"20170302","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":4,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":72,"lastMessageRssi":-82}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"mkxtflzTZxkWHL0v"}
```

I'm running the latest release and re-flashed my cc2531 with the latest file tonight.

Koenkk, 2018-07-31:
If you update the firmware of your cc2531 you have to start from scratch. 

The multiple entries in `database.db` are normal.

wouterbulten, 2018-07-31:
Good to know. I got many "cannot get nodeDescriptor" errors which disappeared after cleaning the db file so I thought it was related. I'll close the issue as it could be simply solved by re-pairing all devices again.

closed 2018-07-31 by wouterbulten