Koenkk/zigbee2mqtt #235: Crash on device pairing

theupmachine, 2018-07-31:
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.2.0
3 info using node@v8.11.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@0.1.1~prestart: zigbee2mqtt@0.1.1
6 info lifecycle zigbee2mqtt@0.1.1~start: zigbee2mqtt@0.1.1
7 verbose lifecycle zigbee2mqtt@0.1.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@0.1.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@0.1.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@0.1.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@0.1.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@0.1.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@0.1.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:304:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:925:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@0.1.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.52-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.11.3
19 verbose npm  v6.2.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@0.1.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@0.1.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

theupmachine, 2018-07-31:
No, but it does give a different error.

```
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^

TypeError: Cannot read property 'key' of undefined
    at ensureDataTypeString (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:297:47)
    at getDataType (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:307:16)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:682:24)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-07-31T12_53_01_479Z-debug.log

```

Koenkk, 2018-07-31:
Are you on the dev branch? And which device are you pairing?

theupmachine, 2018-07-31:
This was not on the dev branch. It has crashed when trying to pair several devices. These two crashes I posted were from Xiaomi devices, but it has done the same with an Lightify bulb as well.

Koenkk, 2018-07-31:
Have you done a `rm -rf node_modules && npm install` after updating?

theupmachine, 2018-07-31:
This was on a completely new install.

Koenkk, 2018-07-31:
Can you checkout of this problem still exists in the dev branch?

Koenkk, 2018-07-31:
Update: maybe related to #236, can you check if zigbee2mqtt 0.1.2 solves the problem?

closed 2018-07-31 by Koenkk