Koenkk/zigbee2mqtt #236: Error in Log

jasimancas, 2018-07-31:
This is the error in 0.1.1:

```
/zigbee2mqtt-0.1.1/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^
TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/zigbee2mqtt-0.1.1/node_modules/zcl-packet/lib/foundation.js:683:21)
    at Dissolve._transform (/zigbee2mqtt-0.1.1/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/zigbee2mqtt-0.1.1/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/zigbee2mqtt-0.1.1/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR!
 zigbee2mqtt@0.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-07-31T10_50_57_391Z-debug.log
[2018-07-31T12:50:57.489Z] PM2 log: App [npm] with id [0] and pid [45], exited with code [1] via signal [SIGINT]
[2018-07-31T12:50:57.493Z] PM2 log: Starting execution sequence in -fork mode- for app name:npm id:0
[2018-07-31T12:50:57.498Z] PM2 log: App name:npm id:0 online
> zigbee2mqtt@0.1.1 start /zigbee2mqtt-0.1.1
> node index.js
```

dodacs, 2018-07-31:
Same here. It was working fine. I just update it to the latest and it stop working.

spinlock: true []
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683:21)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-07-31T11_59_44_090Z-debug.log

Koenkk, 2018-07-31:
Are you on the master or dev branch?

jasimancas, 2018-07-31:
@Koenkk Master, update to 0.1.1 today

Koenkk, 2018-07-31:
What devices do you have paired?

talondnb, 2018-07-31:
This is also happening with the Hassio addon.

jasimancas, 2018-07-31:
@Koenkk no, do this?

Koenkk, 2018-07-31:
@jasimancas yes, can you check if that solves the issue?

jasimancas, 2018-07-31:
@Koenkk I use Hassio Addons, where execute this?

mdroberts1243, 2018-07-31:
What I see is a bit different (using hass.io addon):

> /zigbee2mqtt-0.1.1/node_modules/zcl-packet/lib/foundation.js:297
> dataTypeStr = zclId.dataType(dataType).key;
> ^
> TypeError: Cannot read property 'key' of undefined
> at ensureDataTypeString (/zigbee2mqtt-0.1.1/node_modules/zcl-packet/lib/foundation.js:297:47)
> at getDataType (/zigbee2mqtt-0.1.1/node_modules/zcl-packet/lib/foundation.js:307:16)
> at Dissolve. (/zigbee2mqtt-0.1.1/node_modules/zcl-packet/lib/foundation.js:682:24)

Posted the whole log to the addon githttps://github.com/danielwelch/hassio-zigbee2mqtt/issues/42

danielwelch, 2018-07-31:
I’ll do my best to update the add-on tonight!

jasimancas, 2018-07-31:
Thanks guys, waiting the update :)

danielwelch, 2018-07-31:
see https://github.com/danielwelch/hassio-zigbee2mqtt/releases/tag/v0.1.2

dodacs, 2018-07-31:
@Koenkk It was just my initial test with the system before migrate everything. I had 2 Osram bulbs and 1 Lutron Remote Connected Bulb.

I can confirm that v0.1.2 solved my problem.

Thank you for the quick fix and the amazing work.

mdroberts1243, 2018-07-31:
I'm working again with the v0.1.2 fix.

Koenkk, 2018-08-01:
Thanks all!

closed 2018-08-01 by Koenkk