Koenkk/zigbee2mqtt #237: Issue with Hue Dimmer

Imperial-Guard, 2018-07-31:
After pressing on and off or some other buttons I got the following error: 

`2018-7-31 17:41:14 INFO MQTT publish, topic: 'zigbee2mqtt/0x001788011032d314', payload: '{"action":"off"}'
events.js:341
  this.on(type, _onceWrap(this, type, listener));
                ^
RangeError: Maximum call stack size exceeded
    at Dissolve.once (events.js:341:17)
    at Dissolve.Transform (/zigbee2mqtt-0.1.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:133:8)
    at new Dissolve (/zigbee2mqtt-0.1.1/node_modules/dissolve/index.js:25:20)
    at Dissolve (/zigbee2mqtt-0.1.1/node_modules/dissolve/index.js:17:45)
    at module.exports (/zigbee2mqtt-0.1.1/node_modules/dissolve-chunks/index.js:5:19)
    at FoundPayload._getObj (/zigbee2mqtt-0.1.1/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:216:14)
    at getObjCB (/zigbee2mqtt-0.1.1/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
    at Dissolve.<anonymous> (/zigbee2mqtt-0.1.1/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:238:13)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@0.1.1 start: `node index.js`
npm
 ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.1 start script.
npm 
ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!
     /root/.npm/_logs/2018-07-31T15_41_19_021Z-debug.log
[2018-07-31T17:41:19.766Z] PM2 log: App [npm] with id [0] and pid [157], exited with code [1] via signal [SIGINT]
[2018-07-31T17:41:19.792Z] PM2 log: Starting execution sequence in -fork mode- for app name:npm id:0
[2018-07-31T17:41:19.814Z] PM2 log: App name:npm id:0 online
> zigbee2mqtt@0.1.1 start /zigbee2mqtt-0.1.1
> node index.js
2018-7-31 17:41:29 INFO Starting zigbee2mqtt version 0.1.1 (commit #unknown)
2018-7-31 17:41:29 INFO Starting zigbee-shepherd`

Koenkk, 2018-07-31:
Are you on the latest CC2531 firmware?

Imperial-Guard, 2018-07-31:
I'm using the Hass.io addon, is this done with the lastest version zigbee2mqtt 0.1.1 ?

Koenkk, 2018-07-31:
The firmware of the CC2531 has to be updated manually, see: https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started#2-flashing-the-cc2531-usb-stick

Imperial-Guard, 2018-07-31:
The firmware is flashed with the alternative flashing method, this version is flashed to the CC2531, https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex

Koenkk, 2018-08-01:
That is not the correct one, you should have https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin. That firmware has increased memory which solves this problem.

closed 2018-10-11 by stale[bot]