Koenkk/zigbee2mqtt #240: Integrate Xiaomi MiJia Honeywell smoke detector in alarm system?

Smarti15, 2018-08-03:
Hello, is it possible to integrate the Xiaomi MiJia Honeywell smoke detector into an alarm system. So if alarm is triggered, the smoke detector should also beep. At the moment I can not integrate the smoke alarm (integrated via the MiHome adapter) into the alarm system in the ioBroker because I do not have access to the alarm. Would be great if that's possible.

Koenkk, 2018-08-03:
Its not possible to control the alarm manually.

Smarti15, 2018-08-03:
Oh well, but thanks for the quick answer.

tb-killa, 2018-08-03:
With original hub (Controller) you could trigger the Alarm.

Koenkk, 2018-08-03:
@tb-killa interesting would be indeed nice if we could also do this

tb-killa, 2018-08-04:
For this we have to sniff needing Information with original Controller and maybe we find the needing Code. I Google for it to check if groovy Code is availabel But didnt find it. Only test Button Code who could be implement. Sadly doesnt in Touch with smoke detector and Controller (gateway)

tb-killa, 2018-08-06:
I´v found some more interessted code segments and informations:
https://github.com/athombv/homey/issues/2005
https://github.com/bspranger/Xiaomi/issues/59
https://pastebin.com/59rUTJKv

As it seems to be that the remote control for alarm isn´t available anymore, because i didn´t find informations about them in the system of a friend of mine who gave me short access to his system.
Have to search deeper, maybe we could get them working.

Baseje, 2018-08-13:
Any progress?

Would be nice!

tb-killa, 2018-08-18:
We should close this because actually there is no known way to remote trigger Alarm.

tb-killa, 2018-08-18:
This seems to be no more available.
If i found some more Informations i will reopen.

Smarti15, 2018-08-20:
Okay, that would be really great, if you could use the smoke alarm simultaneously as an auditory alarm sound. That concludes the post.

AnhDuc85, 2018-08-20:
Any chance to set the density level for JTYJ-GD-01LM/BW?

tb-killa, 2018-08-27:
@AnhDuc85 Actually i don´t know ... because of lack of device themselve.
But i figure out that it seems to be posible to trigger the alarm by the use of the ``IAS WD`` Cluster ..
Sadly this seems to be some special stuff and actually i see that ``IAS`` themselve isn´t fully implemented on zigbee shepherd. 
Heimann Siren seems to be also remote enableable like mentioned with ``IAS WD`` ...

look at this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/404#issuecomment-384163525 there seems to be some more sort of siren who could be remote triggered ....

closed 2018-12-22 by stale[bot]

sascha777, 2019-01-02:
I am also looking for a way of using my Mijia smoke detectors as an alarm.
Did someone managed to set smoke density or trigger the alarm or the self test?