Koenkk/zigbee2mqtt #244: zigbee2mqtt service not starting in background

rolfz, 2018-08-03:
Hi,
I am able to start the zigbee2mqtt service in a shell but not launch the background service which returns with an error. I am using the openhabian installation which may create trouble as I have no root login!
so:
cd /opt/zigbee2mqtt
npm start
is working fine,
but
sudo systemctl start zigbee2mqtt
is not working :-(
to be mentionned that the configuration file has been created as requested.

Any idea what may be wrong ?

Koenkk, 2018-08-04:
Could you provide the log of: `sudo systemctl status zigbee2mqtt.service`

rolfz, 2018-08-04:
Here some more from the log

Aug 05 01:24:27 openHABianPi systemd[1]: Stopped zigbee2mqtt.
Aug 05 01:24:27 openHABianPi systemd[1]: Started zigbee2mqtt.
Aug 05 01:24:28 openHABianPi npm[7161]: > zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
Aug 05 01:24:28 openHABianPi npm[7161]: > node index.js
Aug 05 01:24:30 openHABianPi npm[7161]: 2018-8-5 01:24:30 INFO Starting zigbee2mqtt version 0.1.0 (commit #ae41b7b)
Aug 05 01:24:30 openHABianPi npm[7161]: 2018-8-5 01:24:30 INFO Starting zigbee-shepherd
Aug 05 01:24:30 openHABianPi npm[7161]: /opt/zigbee2mqtt/node_modules/objectbox/lib/db.js:10
Aug 05 01:24:30 openHABianPi npm[7161]:         if (err) throw err;
Aug 05 01:24:30 openHABianPi npm[7161]:                  ^
Aug 05 01:24:30 openHABianPi npm[7161]: Error: EACCES: permission denied, open '/opt/zigbee2mqtt/data/database.db'
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR! code ELIFECYCLE
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR! errno 1
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR! Exit status 1
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR!
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
Aug 05 01:24:30 openHABianPi npm[7161]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Aug 05 01:24:30 openHABianPi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Aug 05 01:24:30 openHABianPi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Aug 05 01:24:30 openHABianPi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Aug 05 01:24:30 openHABianPi systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Aug 05 01:24:30 openHABianPi systemd[1]: Stopped zigbee2mqtt.
Aug 05 01:24:30 openHABianPi systemd[1]: Started zigbee2mqtt.
Aug 05 01:24:32 openHABianPi npm[7187]: > zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
Aug 05 01:24:32 openHABianPi npm[7187]: > node index.js

rolfz, 2018-08-05:
Thank you for the tip. It seems that the problem came from the openhabian implementation that has no pi account. I changed the rights on database.db and replaced pi by openhabian inside zigbee2mqtt.services. The service now works !!! :-)

closed 2018-08-05 by rolfz

RIMperator, 2018-11-16:
I seem to have the same issue trying to get zigbee2mqtt.service running on my Loxberry. When cloning the zigbee2mqtt repository I've adapted the pi user and group to the loxberry user. However this doesn't seem to be enough. Can anyone indicate what modifications need to be done additionally? Many thanks!

Nov 16 01:20:46 loxberry npm[19442]: npm ERR!
Nov 16 01:20:46 loxberry npm[19442]: npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
Nov 16 01:20:46 loxberry npm[19442]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Nov 16 01:20:46 loxberry npm[19442]: npm ERR! A complete log of this run can be found in:
Nov 16 01:20:46 loxberry systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 01:20:46 loxberry systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Nov 16 01:20:46 loxberry systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov 16 01:20:47 loxberry systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Nov 16 01:20:47 loxberry systemd[1]: Stopped zigbee2mqtt.
Nov 16 01:20:47 loxberry systemd[1]: Started zigbee2mqtt.
Nov 16 01:20:48 loxberry npm[19465]: > zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
Nov 16 01:20:48 loxberry npm[19465]: > node index.js
Nov 16 01:20:48 loxberry npm[19465]:                 Zigbee2mqtt requires node version >=8.11 10, you are running v8.12.0!
Nov 16 01:20:49 loxberry npm[19465]: fs.js:885
Nov 16 01:20:49 loxberry npm[19465]:   return binding.mkdir(pathModule._makeLong(path),
Nov 16 01:20:49 loxberry npm[19465]:                  ^
Nov 16 01:20:49 loxberry npm[19465]: Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/data/log/2018-11-16.01-20-49'
Nov 16 01:20:49 loxberry npm[19465]:     at Object.fs.mkdirSync (fs.js:885:18)
Nov 16 01:20:49 loxberry npm[19465]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:199:16)
Nov 16 01:20:49 loxberry npm[19465]:     at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
Nov 16 01:20:49 loxberry npm[19465]:     at Object.mkdirSync (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:82:10)
Nov 16 01:20:49 loxberry npm[19465]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:17:4)
Nov 16 01:20:49 loxberry npm[19465]:     at Module._compile (module.js:653:30)
Nov 16 01:20:49 loxberry npm[19465]:     at Object.Module._extensions..js (module.js:664:10)
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! code ELIFECYCLE
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! errno 1
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! Exit status 1
Nov 16 01:20:49 loxberry npm[19465]: npm ERR!
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Nov 16 01:20:49 loxberry npm[19465]: npm ERR! A complete log of this run can be found in:
Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Nov 16 01:20:49 loxberry systemd[1]: Stopped zigbee2mqtt.

RIMperator, 2018-11-16:
> I seem to have the same issue trying to get zigbee2mqtt.service running on my Loxberry (based on Raspbian Stretch Lite). When cloning the zigbee2mqtt repository I've adapted the pi user and group to the loxberry user and group (loxberry:loxberry). However this doesn't seem to be enough. Can anyone indicate what modifications need to be done additionally? Many thanks!
> 
> Nov 16 01:20:46 loxberry npm[19442]: npm ERR!
> Nov 16 01:20:46 loxberry npm[19442]: npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
> Nov 16 01:20:46 loxberry npm[19442]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> Nov 16 01:20:46 loxberry npm[19442]: npm ERR! A complete log of this run can be found in:
> Nov 16 01:20:46 loxberry systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
> Nov 16 01:20:46 loxberry systemd[1]: zigbee2mqtt.service: Unit entered failed state.
> Nov 16 01:20:46 loxberry systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
> Nov 16 01:20:47 loxberry systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
> Nov 16 01:20:47 loxberry systemd[1]: Stopped zigbee2mqtt.
> Nov 16 01:20:47 loxberry systemd[1]: Started zigbee2mqtt.
> Nov 16 01:20:48 loxberry npm[19465]: > zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
> Nov 16 01:20:48 loxberry npm[19465]: > node index.js
> Nov 16 01:20:48 loxberry npm[19465]: Zigbee2mqtt requires node version >=8.11 10, you are running v8.12.0!
> Nov 16 01:20:49 loxberry npm[19465]: fs.js:885
> Nov 16 01:20:49 loxberry npm[19465]: return binding.mkdir(pathModule._makeLong(path),
> Nov 16 01:20:49 loxberry npm[19465]: ^
> Nov 16 01:20:49 loxberry npm[19465]: Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/data/log/2018-11-16.01-20-49'
> Nov 16 01:20:49 loxberry npm[19465]: at Object.fs.mkdirSync (fs.js:885:18)
> Nov 16 01:20:49 loxberry npm[19465]: at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:199:16)
> Nov 16 01:20:49 loxberry npm[19465]: at mkdirSyncRecursive (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:197:12)
> Nov 16 01:20:49 loxberry npm[19465]: at Object.mkdirSync (/opt/zigbee2mqtt/node_modules/mkdir-recursive/index.js:82:10)
> Nov 16 01:20:49 loxberry npm[19465]: at Object. (/opt/zigbee2mqtt/lib/util/logger.js:17:4)
> Nov 16 01:20:49 loxberry npm[19465]: at Module._compile (module.js:653:30)
> Nov 16 01:20:49 loxberry npm[19465]: at Object.Module._extensions..js (module.js:664:10)
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! code ELIFECYCLE
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! errno 1
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! Exit status 1
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR!
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> Nov 16 01:20:49 loxberry npm[19465]: npm ERR! A complete log of this run can be found in:
> Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
> Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Unit entered failed state.
> Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
> Nov 16 01:20:49 loxberry systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
> Nov 16 01:20:49 loxberry systemd[1]: Stopped zigbee2mqtt.

Thanks to a tip to someone from another forum the issue was solved using the following command:

`chmod -R 777 /opt/zigbee2mqtt`

Tontze, 2022-12-12:
I had everything running just fine, after i updated z2m to latest, this permission error came up .. @Koenkk