Koenkk/zigbee2mqtt #246: TRADFRI IKEA LIGHTS

davidhrbaty, 2018-08-04:
Hello,
First thank you for great project and work. 
I have one queston. How can I implement states of Tradfri lights? If I turn on the light via tradfri remote or by connecting the light to main power ( default state of these lights are, that they turn on if I connect them to main power), I cannot see updated state in HA.  I tried to add `retain: true`  on both sides: zigbee2mqtt and HA, but there is no change.  I think, there is a problem that the light doesn't announce its state after power up or if it is controlled via tradfri remote control.

davidhrbaty, 2018-08-05:
Thank you @Koenkk  for wiki page, now it is clear for me. 

I have zigbee2mqtt running in docker, so I have started it with debug environment enabled and there is a debug log when I turned on tradfri bullb via remote:

`2018-8-5 11:40:45 DEBUG Check online 0x90fd9ffffe752d76
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +60s
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 64236, nwkaddrofinterest: 64236 } +0ms
  serialport write 9 bytes of data +0ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +10ms
  cc-znp:AREQ <-- ZDO:srcRtgInd, { dstaddr: 64236, relaycount: 0, relaylist: <Buffer > } +16ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +0ms
  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 64236, status: 0, nwkaddr: 64236, logicaltype_cmplxdescavai_userdescavai: 1, apsflags_freqband: 64, maccapflags: 142, manufacturercode: 4476, maxbuffersize: 82, maxintransfersize: 82, servermask: 0, maxouttransfersize: 82, descriptorcap: 0 } +48ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +1ms
`

Koenkk, 2018-08-07:
With some report and binding of `genOnOff` I tried to get the bulb to report it's state on power on, but unfortunately this is not working. 

I'm wondering how you control the bulb through the remote when also having it paired with zigbee2mqtt, could you explain this?

Koenkk, 2018-08-09:
@davidhrbaty yes it indeed doesn't announce it's state on powering on. Only announces itself to the network. Have to check if we can act on that.

zsiti, 2018-08-29:
@Koenkk @davidhrbaty Hey guys, I'm new in this, my CC2531 is on the way. I'd very much like to see the IKEA bulbs' status in Home Assistant even if it's not switched by Home Assistant (via zigbee2mqtt) but instead the IKEA's own remote or by connecting mains power. Will it be possible?

Koenkk, 2018-08-30:
@zsiti I don't know if it is possible to still operate the bulbs with the remote once paired to zigbee2mqtt... That should be investigated first.

timvankatwijk, 2018-09-08:
@Koenkk I can answer that! Just installed zigbee2mqtt on my hass.io yesterday. You can connect a light to both the cc2531 and a remote. Key is that you first make sure that neither the remote or the bulb are connected to anything: Then connect the bulb to the cc2531 and after that you can connect the remote to the bulb as usual.

Now you can turn on/off the bulb using either zigbee2mqtt or the ikea remote.
But the disadvantage (as mentioned) is that the ikea side doesn't know what the zigbee2mqtt side does and vice versa.

RonnyAL, 2018-09-27:
I guess it's not so relevant to zigbee2mqtt, but would it be possible to use a Home Assistant automation to react on the bulb connecting to and disconnecting from the network? My bulbs are not in use right now (need to change a socket first), but I can test this once they are.

LuisCanossa, 2018-10-03:
No updates on this topic!
I have the same problem with IKEA bulbs...

tb-killa, 2018-10-04:
@Koenkk Dont know if this could help but deconz get reporting working.? :
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/751#issuecomment-416017332

wimpie007, 2018-10-14:
Started to use the CC2531 a few days ago: quit impressed with it! Nice work Koen!
What i'm stugling with: when i for example dimm or change the color of a tradfri lamp with the remote control, it doesnt reflect in HA. The lamp doesnt seem to report it's "changed state" back to HA. Any ideas??

Koenkk, 2018-10-14:
@tb-killa, indeed maybe we can solve the problem with that. @wimpie007 which bulb do you have? Then we can do some tests.

wimpie007, 2018-10-15:
These are my devices:
0x000b57fffea710af (0x000b57fffea710af): LED1624G9 - IKEA TRADFRI LED bulb E27 600 lumen, dimmable, color, opal white (Router)
0x000b57fffef35731 (0x000b57fffef35731): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
0x000b57fffeedf510 (0x000b57fffeedf510): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
0x000b57fffe2b8781 (0x000b57fffe2b8781): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
None of the 3 lamps (1lamp, 2 10watt led drivers) report 'back' when changing something with the remote. (the 10 watt drivers have the little dimmer ICTC-G1 wich is 'supported' somehow (problems anyway with loosing the pairing with zigbee2mqqt) and the bulb has the 'bigger' unsupported dimmer with the buttons)
Willing to test, no problem. Thanks in advance!

Koenkk, 2018-10-15:
Could you change the following entry in your `devices.js`. 

```js
    {
        zigbeeModel: ['TRADFRI bulb E27 CWS opal 600lm'],
        model: 'LED1624G9',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 600 lumen, dimmable, color, opal white',
        supports: generic.light_onoff_brightness_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colorxy().toZigbee,
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            execute(device, [(cb) => device.report('genLevelCtrl', 'currentLevel', 0, 1000, 0, cb)], callback);
        },
    },
```

wimpie007, 2018-10-15:
mmm, i feel stupid. Running hass.io in a container, so no direct access to devices.js.
Succeeded in connecting to port 22222, but then no means to edit the devices.js.
Any hint anyone??

Koenkk, 2018-10-15:
Probably `/0.1.6/node_modules/zigbee-shepherd-converters/devices.js`.

wimpie007, 2018-10-15:
took
https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js
Edited the file as asked, and put it in /share/zigbee2mqtt
Can't see anything that changed.

Koenkk, 2018-10-15:
Do you see a successfully configured with the device ID?

wimpie007, 2018-10-15:
heuh, do i have to re-pair it?
still says the same when starting zigbee2mqqt:
0x000b57fffea710af (0x000b57fffea710af): LED1624G9 - IKEA TRADFRI LED bulb E27 600 lumen, dimmable, color, opal white (Router)

Koenkk, 2018-10-15:
You should see this in the log, can you restart zigbee2mqtt and provide the log after letting it run for a minute?

wimpie007, 2018-10-15:
ok, done that:
https://pastebin.com/2aN2b5Sj

Koenkk, 2018-10-17:
Thanks, could you provide the log when running with `DEBUG=* npm start`

wimpie007, 2018-10-17:
as i  run it in hass.io container, debug is not easy.
is it possible to give a debug command in configuration.yaml?

Koenkk, 2018-10-17:
@danielwelch is there an easy way to set DEBUG=* for the hassio addon?

wimpie007, 2018-10-17:
@koenkk @danielwelch 
hehe, i just tried  "log_level": "debug" in the configuration.yaml and bingo;)
https://pastebin.com/VR1GPBJd
"Recieved zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI bulb E27 CWS opal 600lm'" Looks like it must be able to see the status online/offline at least!:)

danielwelch, 2018-10-17:
@wimpie007 yep, that’s how it’s done. Log level and directory can be configured via the add-on Hass.io front end (not actually configuration.yaml, it’s JSON). You can also set zigbee_shepherd_debug to true to (I think) achieve the same effect as “DEBUG=*”, although I need to document this.

wimpie007, 2018-10-17:
@danielwelch 
putting   "zigbee_shepherd_debug": true
in config spits out a lot of low-level debug.
How can i define the directory? (its not in the /share/ dir...

tmarquespt, 2018-10-22:
I also have IKEA bulbs and I'm also suffering with the lack of updated states. It happens both when we cut the AC from the bulb (it stays as 'on' in HA) and upon a HA restart they all appear as OFF regardless of the state they really have.
I'm using danielwelch addon on Hassio. If you need any debug info from me I'm glad to help.

wimpie007, 2018-10-22:
@tmarquespt Yes, it's not really usable without the states... :(

steinis123, 2018-11-12:
Is this only a problem with the Ikea bulbs or a generell problem? I also have some Philips light connected to hue hub and Philips dimmer switch connected to the same bulb that works just fine with or without the hue hub powered on. So it seems that hue sends status to the hub regardless if it is connected to a remote or not.

luismsandrade, 2018-11-12:
Philips Hue white and color (model: 9290012573A) with same problem as @tmarquespt:
" It happens both when we cut the AC from the bulb (it stays as 'on' in HA) and upon an HA restart they all appear as OFF regardless of the state they really have.", on HASS.IO with cc2531.

tmarquespt, 2018-11-12:
@luismsandrade Zigbee2Mqtt saves the last state on states.json. It's loaded upon the addon restart. When you reboot the HA cointainer, the zigbee2mqtt container doesn't know, therefore the info on states.jason it's not updated. You might solve this with an automation to restart the zigbee2mqtt addon upon a HA restart, so at least you can have the last known state of the lights passed to HA. (assuming you didn't change any status during the HA restart)

The case of a power outage is the worst in my opinion. As I have an UPS on HA but not on the bulbs, and the bulbs allways turn on after a power outage, I have no way to avoid zigbee bulbs to stay on for a long period.

closed 2019-03-20 by stale[bot]

bakkerv, 2020-04-12:
I still have this issue, anybody any clue?

elenril2, 2022-01-31:
> 

How do you set this option?

romlisrl, 2022-07-31:
@elenril2 
Might be here, but what setting he used not sure.
[image]