Koenkk/zigbee2mqtt #247: Help to add TRADFRI motion sensor?

fradducho, 2018-08-05:
Hello world!

I would like to know when we can have support for. Ikea TRADFRI motion sensor?

Is there a way to add it manually?

I understood that it worked with any product that I use zigbee.

Many Gracais, I wish someone could help me.

Is there a way to donate? to the creator of this?

Koenkk, 2018-08-05:
I have a TRADRI motion sensor myself. The device can be paired but the TRADFRI motion sensor (and 5 button remote) is a strange device. As it directly control bulbs (it doesn't send occupancy state to the gateway). See #102

dmak, 2018-12-27:
I am checking the list of [supported hardware by ZiGate](https://zigate.fr/produits-ikea-tradfri-compatibles-zigate/) and [Conbee](https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/compatibility/?L=1) and I see that IKEA motion sensor is in the list. So I wonder if those lists are lying about proper motion sensor support or there is some physical advantage comparing to CC2531 chip?

Koenkk, 2018-12-27:
This is because the device directly controls the bulbs or group and therefore is not sending anything to the coordinator. I don't know how Conbee and ZiGate fix this.

Please keep an eye on https://github.com/Koenkk/zigbee2mqtt/issues/765 which will probably make this a supported device.

saz, 2019-02-06:
@Koenkk Might the TRADFRI motion sensor work with group support added now?

wizzor, 2019-02-06:
Sorry to lift a closed ticket, but #765 is closed, what does that mean for the motion sensor?

Koenkk, 2019-02-06:
I will give the motion sensor another try soon (I also have one :smile:)

Sviests, 2019-02-09:
I'd also very much like to see the motion sensor supported :)

vke, 2019-02-09:
Hello, what I can do to help to add support for IKEA TRADFRI motion sensor, model E1525?

Exact device name is `'TRADFRI motion sensor'`, device sends group broadcast messages (group is set like for 5-button remote), Cluster is 6 (On/off), ZCL Command is: On with timed off (0x42), Source endpoint is 1

Koenkk, 2019-02-09:
@vke I just tried, but don't see any no converter available messages yet.. need to dig a bit deeper.

tynor88, 2019-02-17:
Has anyone tried the Motion Sensor with the new dev firmware? 
https://github.com/Koenkk/zigbee2mqtt/issues/102#issuecomment-464460726

vke, 2019-02-17:
@Koenkk can I help adding this sensor? I have such sensor, sniffer, zigbee2mqtt and cc2531 with latest firmware. But I don't know where to start.

Koenkk, 2019-02-17:
@vke i already tried, but it seems that the commands that are send are not correctly parsed, need to dive deeper.

vke, 2019-02-17:
@Koenkk I see differences, but I don’t understand whether I am going in the right direction:

Here I see two uint8 and two uint16 (six bytes total):
https://github.com/Koenkk/zcl-packet/blob/master/lib/defs/zcl_meta.json#L401
```            "onWithTimedOff":{
                "params":[
                    {"ctrlbits":"uint8"},
                    {"ctrlbyte":"uint8"},
                    {"ontime":"uint16"},
                    {"offwaittime":"uint16"}
                ],
"dir":0}
```

But in sniffer (Wireshark) I see **one** uint8 and two uint16 numbers (five total):
```ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 66
    Command: On with timed off (0x42)
    Payload
        On/Off Control Mask: 0x00
            .... ...0 = Accept Only When On: 0
            0000 000. = Reserved: 0
        On Time: 60.0 seconds
        Off Wait Time: 0.0 seconds

00 - On/Off Control Mask: 0x00
58 02 - On Time: 60.0 seconds
00 00 - Off Wait Time: 0.0 seconds
````

vke, 2019-02-17:
@Koenkk yes, after removing `ctrlbyte` var from `zcl_meta.json` I see right messages in debug log:

```
 zigbee-shepherd:af {"groupid":4567,"clusterid":6,"srcaddr":26455,"srcendpoint":1,"dstendpoint":1,
"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":14470033,"transseqnumber":0,
"len":8,"data":{"0":1,"1":118,"2":66,"3":0,"4":88,"5":2,"6":0,"7":0},
"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":0},
"manufCode":0,"seqNum":118,"cmdId":"onWithTimedOff","payload":
{"ctrlbits":0,"ontime":600,"offwaittime":0}}}
```

vke, 2019-02-17:
I added `'onWithTimedOff'` to this line: to https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L688

And now I see motion messages in logs

Koenkk, 2019-02-18:
@vke very good job! That means supporting this device is now just a matter of writing converters.

vke, 2019-02-18:
@Koenkk I used this to test:

```
    {
        zigbeeModel: ['TRADFRI motion sensor'],
        model: 'E1525',
        vendor: 'IKEA',
        description: 'TRADFRI motion sensor',
        supports: 'on, off',
        fromZigbee: [
            fz.genOnOffTimed,
			fz.generic_battery, fz.ignore_power_change
        ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {
                direction: 0, attrId: 33, dataType: 32, minRepIntval: 0, maxRepIntval: repInterval.MAX, repChange: 0,
            };

            const actions = [
                (cb) => device.bind('genOnOff', coordinatorGroup, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.foundation('genPowerCfg', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

and

```
    genOnOffTimed: {
        cid: 'genOnOff',
        type: 'cmdOnWithTimedOff',
        convert: (model, msg, publish, options) => {
            return {click: 'on'};
        },
    },
```
But there is no work with variables like `ontime` and `offwaittime`, also `click` should be renamed to something more appropriate like `occupancy`

Koenkk, 2019-02-18:
@vke could you make a pr for zcl-packet? Then I can continue with this.

vke, 2019-02-18:
@Koenkk done: https://github.com/Koenkk/zcl-packet/pull/9

tynor88, 2019-02-23:
@vke is it working with zigbee2mqtt ?

Koenkk, 2019-02-23:
Device is supported in the dev branch now, biggest part of the credits go to @vke. 

To increase the occupancy timeout and sensitivity of this device, the turning knob on the device itself can be used.

closed 2019-02-23 by Koenkk

timdonovanuk, 2019-03-10:
Just pulled the latest dev branch and although I can add the sensor, it does not work - the occupancy state never changes from "off":

```
zigbee2mqtt:info 3/10/2019, 5:34:38 PM Bedroom Motion Sensor (0x000b57fffe949fa3): E1525 - IKEA TRADFRI motion sensor (EndDevice)

zigbee2mqtt:info 3/10/2019, 5:34:15 PM MQTT publish: topic 'homeassistant/binary_sensor/0x000b57fffe949fa3/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/Bedroom Motion Sensor","json_attributes_topic":"zigbee2mqtt/Bedroom Motion Sensor","name":"Bedroom Motion Sensor_occupancy","unique_id":"0x000b57fffe949fa3_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffe949fa3","name":"Bedroom Motion Sensor","sw_version":"Zigbee2mqtt 1.2.1","model":"TRADFRI motion sensor (E1525)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'

zigbee2mqtt:info 3/10/2019, 5:34:15 PM MQTT publish: topic 'homeassistant/sensor/0x000b57fffe949fa3/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/Bedroom Motion Sensor","json_attributes_topic":"zigbee2mqtt/Bedroom Motion Sensor","name":"Bedroom Motion Sensor_battery","unique_id":"0x000b57fffe949fa3_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffe949fa3","name":"Bedroom Motion Sensor","sw_version":"Zigbee2mqtt 1.2.1","model":"TRADFRI motion sensor (E1525)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'

zigbee2mqtt:info 3/10/2019, 5:34:15 PM MQTT publish: topic 'homeassistant/sensor/0x000b57fffe949fa3/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/Bedroom Motion Sensor","json_attributes_topic":"zigbee2mqtt/Bedroom Motion Sensor","name":"Bedroom Motion Sensor_linkquality","unique_id":"0x000b57fffe949fa3_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffe949fa3","name":"Bedroom Motion Sensor","sw_version":"Zigbee2mqtt 1.2.1","model":"TRADFRI motion sensor (E1525)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'

zigbee2mqtt:info 3/10/2019, 5:34:23 PM Successfully configured Bedroom Motion Sensor (0x000b57fffe949fa3)

```

Thats it - nothing else ever appears in the log, despite movement happening.

timdonovanuk, 2019-03-10:
I tried adding another Tradfri motion sensor with the following errors:

  zigbee2mqtt:error 3/10/2019, 5:55:24 PM Cannot get the Node Descriptor of the Device: 0x000b57fffe935e6a (Error: Timed out after 10000 ms)

Koenkk, 2019-03-10:
Are you on the latest `20190223` firmware? This is required for this device. (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.2.0)

fradducho, 2019-03-11:
What are the steps to  pair it???

fradducho, 2019-03-11:
I have pair correctly but I see that I get this message. that is normal?

  zigbee2mqtt:warn 3/11/2019, 5:19:25 AM Failed to configure IkeaMotionSensor1 (0x000b57fffe99f62a) ('Error: Timed out after 10000 ms') (attempt #1)

  zigbee2mqtt:warn 3/11/2019, 5:19:25 AM This can be ignored if the device is working properly

Koenkk, 2019-03-11:
@fradducho depends, if your device works (detects occupancy) it can be ignored.

n1k0p0l, 2019-03-21:
> Just pulled the latest dev branch and although I can add the sensor, it does not work - the occupancy state never changes from "off":
>
> ```
> zigbee2mqtt:info 3/10/2019, 5:34:38 PM Bedroom Motion [...]

From my understanding, the sensor only goes to occupancy-false after the time you set on the dial inside it, passes. I've set mine to 1 minute and after triggering, it only goes to false after that minute.

timdonovanuk, 2019-05-04:
Just got this working with the latest firmware, thank you! These sensors have a much lower blind time compared to Aqara sensors..in fact they almost seem to have zero blind time!

dmak, 2019-07-14:
I see from this thread that someone had succeeded in pairing E1525. It was a bit of challenge for me. I have pressed 4 times the button to reset it (and it blinked several times with red LED) and then I have pressed & hold the button for 10 seconds and put it next to CC2531 stick. Nothing in zigbee2mqtt logs (loglevel is debug). I have repeated the same, but this time I did reset also close to CC2531 stick and it finally was paired! I still wonder how to do the pairing reliably.

RestOp, 2019-09-30:
Hi guys,

maybe someone knows how to change the delay in after "true" state?

oberlinatgithub, 2021-11-15:
Hello folks!  First of all, thank you for the great work.  I have several Ikea devices running via zigbee2mqtt in Hassio.  Unfortunately, the brightness display of the motion sensor does not work for me.  All but one are showing 254.  I am not quite clear why that is.  Thanks very much!
[image]