Koenkk/zigbee2mqtt #253: Provide Informations for usable Converter per Device as mqtt-log

tb-killa, 2018-08-06:
It would be great if we could extend our ``zigbee2mqtt/bridge/log`` -> ``devices`` list with an json like output of available converter (only ``toZigbee``) per devices.

With this informations we could give the Application behind zigbee2mqtt the possibility to parse and build automatic control.
Actually you have to check first the wiki and converter source code to know what control-type we could use per available device.

Koenkk, 2018-08-07:
Great! I think we can also use the home assistant mapping for this?

tb-killa, 2018-08-27:
@Koenkk what about use ``docgen.js`` (modified) to generate our devices_converter_output.json file who we provide if you push new version of converter?
With this we could onload them on zigbee2mqtt start for providing via mqtt topic ?
This would make converting simple and its like the same as we do for generate supported devices list.
If somebody like he could also use the file themselve for building 3party application stuff ;)

Koenkk, 2018-08-28:
That would mean that we need to move stuff for homeassistant.js to zigbee-shepherd-converters. Maybe it would be nicer if each converter specifies what values it can return.

closed 2018-10-22 by tb-killa