Koenkk/zigbee2mqtt #254: Problems with QBKG04LM

jasimancas, 2018-08-06:
When I push the button to turn of the QBKG04LM, this is the log:

> 2018-8-6 12:59:11 DEBUG Check online 0x00158d0001f4e65f
2018-8-6 12:59:11 DEBUG Check online 0x00158d00016d94bf
2018-8-6 12:59:11 DEBUG Check online 0x00158d0001b89324
2018-8-6 12:59:11 DEBUG Check online 0x00158d000200c16b
2018-8-6 12:59:11 DEBUG Check online 0x00158d000159a8b3
2018-8-6 12:59:11 DEBUG Check online 0x00158d0001b8939e
2018-8-6 12:59:11 DEBUG Check online 0x00158d000119911c
2018-8-6 12:59:39 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/interruptor_entrada/set' with data 'ON'
2018-8-6 12:59:39 INFO Zigbee publish to '0x00158d00016d94bf', genOnOff - on - {} - 2
2018-8-6 13:00:11 DEBUG Check online 0x00158d0001f4e65f
2018-8-6 13:00:11 DEBUG Check online 0x00158d00016d94bf
2018-8-6 13:00:11 DEBUG Check online 0x00158d0001b89324
2018-8-6 13:00:11 DEBUG Check online 0x00158d000200c16b
2018-8-6 13:00:11 DEBUG Check online 0x00158d000159a8b3
2018-8-6 13:00:11 DEBUG Check online 0x00158d0001b8939e
2018-8-6 13:00:11 DEBUG Check online 0x00158d000119911c

Is random, sometimes it works well, sometimes it fails, any idea why?
I have two QBKG04LM and same troubles.

Koenkk, 2018-08-07:
Could this be a range/interference issue?

jasimancas, 2018-08-07:
I doubt it, because when I push the switch the status changes to ON immediately and Zigbee2Mqtt reads the change of status, if it were interference I wouldn't read the change.
And the same thing happens to both of us, it's random.

jasimancas, 2018-08-07:
OK, although the problem is random, I install and test a few days to see if the problem repeats again.

jasimancas, 2018-08-08:
`2018-8-8 16:06:09 INFO MQTT publish, topic: 'zigbee2mqtt/pulsador_estudio', payload: '{"battery":"100.00","voltage":3062,"linkquality":23,"click":"single"}'
2018-8-8 16:06:09 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:09 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/interruptor_luz_estudio/set' with data 'OFF'
2018-8-8 16:06:09 INFO Zigbee publish to '0x00158d000159a8b3', genOnOff - off - {} - 2

SEND MESSAGE TO TURN OFF BUT DONT TURN OFF

2018-8-8 16:06:12 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":0.10000000149011612}}
2018-8-8 16:06:12 INFO MQTT publish, topic: 'zigbee2mqtt/enchufe_dormitorio_ja', payload: '{"power":0.1,"linkquality":84,"state":"ON","voltage":null,"consumption":82.12,"temperature":39}'
2018-8-8 16:06:12 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":0.10000000149011612}}
2018-8-8 16:06:15 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/interruptor_luz_estudio/set' with data 'OFF'
2018-8-8 16:06:15 INFO Zigbee publish to '0x00158d000159a8b3', genOnOff - off - {} - 2

AGAIN

2018-8-8 16:06:25 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-8 16:06:25 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-8 16:06:26 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:26 INFO MQTT publish, topic: 'zigbee2mqtt/pulsador_estudio', payload: '{"battery":"100.00","voltage":3062,"linkquality":2,"click":"single"}'
2018-8-8 16:06:26 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:26 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/interruptor_luz_estudio/set' with data 'OFF'
2018-8-8 16:06:26 INFO Zigbee publish to '0x00158d000159a8b3', genOnOff - off - {} - 2

AGAIN

2018-8-8 16:06:27 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-8 16:06:27 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-8 16:06:27 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":231.8300018310547}}
2018-8-8 16:06:27 INFO MQTT publish, topic: 'zigbee2mqtt/enchufe_pc', payload: '{"power":231.83,"linkquality":15,"state":"ON","voltage":null,"consumption":142.32,"temperature":37}'
2018-8-8 16:06:27 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":231.8300018310547}}
2018-8-8 16:06:28 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:28 INFO MQTT publish, topic: 'zigbee2mqtt/pulsador_estudio', payload: '{"battery":"100.00","voltage":3062,"linkquality":13,"click":"single"}'
2018-8-8 16:06:28 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:28 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/interruptor_luz_estudio/set' with data 'OFF'
2018-8-8 16:06:28 INFO Zigbee publish to '0x00158d000159a8b3', genOnOff - off - {} - 2

AGAIN

2018-8-8 16:06:28 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":0}}
2018-8-8 16:06:28 INFO MQTT publish, topic: 'zigbee2mqtt/enchufe_dormitorio_ja', payload: '{"power":0,"linkquality":84,"state":"ON","voltage":null,"consumption":82.12,"temperature":39}'
2018-8-8 16:06:28 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":0}}
2018-8-8 16:06:30 DEBUG Check online 0x00158d0001f4e65f
2018-8-8 16:06:30 DEBUG Check online 0x00158d00016d94bf
2018-8-8 16:06:30 DEBUG Check online 0x00158d0001b89324
2018-8-8 16:06:30 DEBUG Check online 0x00158d000200c16b
2018-8-8 16:06:30 DEBUG Check online 0x00158d000159a8b3
2018-8-8 16:06:30 DEBUG Check online 0x00158d0001b8939e
2018-8-8 16:06:30 DEBUG Check online 0x00158d000119911c
2018-8-8 16:06:47 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":210.32000732421875}}
2018-8-8 16:06:47 INFO MQTT publish, topic: 'zigbee2mqtt/enchufe_pc', payload: '{"power":210.32,"linkquality":84,"state":"ON","voltage":null,"consumption":142.32,"temperature":37}'
2018-8-8 16:06:47 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":210.32000732421875}}
2018-8-8 16:06:53 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-8 16:06:53 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-8 16:06:53 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:53 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":1}}
2018-8-8 16:06:53 INFO MQTT publish, topic: 'zigbee2mqtt/pulsador_estudio', payload: '{"battery":"100.00","voltage":3062,"linkquality":5,"click":"single"}'
2018-8-8 16:06:53 DEBUG Recieved mqtt message on topic 'zigbee2mqtt/interruptor_luz_estudio/set' with data 'OFF'

SEND MESSAGE

2018-8-8 16:06:53 INFO Zigbee publish to '0x00158d000159a8b3', genOnOff - off - {} - 2
2018-8-8 16:06:54 INFO MQTT publish, topic: 'zigbee2mqtt/interruptor_luz_estudio', payload: '{"state":"OFF","linkquality":92}'

NOW YES !!!! (MAYBE AFTER CHECK IF IS ONLINE?)

2018-8-8 16:06:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":117440697,"onOff":0}}
2018-8-8 16:06:54 INFO MQTT publish, topic: 'zigbee2mqtt/interruptor_luz_estudio', payload: '{"state":"OFF","linkquality":84}'
2018-8-8 16:06:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"61440":117440697,"onOff":0}}
2018-8-8 16:06:54 DEBUG Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}`

The devide is: zigbee2mqtt/interruptor_luz_estudio (0x00158d000159a8b3)

Koenkk, 2018-08-08:
I don't see any error which should appear once a device request fails https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/zigbee.js#L190.

Can you provide the start of your log?

jasimancas, 2018-08-08:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2271741/log.txt)

This is the log, Addons restarted and complete log, several tries but the device does not turn on

jasimancas, 2018-08-08:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2271755/log.txt)
The continuation of the log, I have turned it on manually (by clicking physically) and when you turn it off the problem is repeated.

jasimancas, 2018-08-08:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2271759/log.txt)
And now, randomly, it works

Koenkk, 2018-08-09:
I still don't see the error messages of the failed request, can you update to the latest dev branch?

jasimancas, 2018-08-09:
This?: 2018-8-9 19:01:50 INFO Starting zigbee2mqtt version 0.1.2 (commit #a36e5ab)

jasimancas, 2018-08-09:
OK, i install in docker, the DEV version, and see that!

> 2018-8-9 22:38:44 ERROR Zigbee publish to '0x00158d00016d94bf', genOnOff - on - {} - 2 failed with error Error: AF data request fails, status code: 183. APS no ack.
,2018-8-9 22:38:53 INFO MQTT publish, topic: 'zigbee2mqtt/enchufe_dormitorio_ja', payload: '{"power":0,"linkquality":76,"state":"ON","voltage":null,"consumption":82.13,"temperature":39}'
,2018-8-9 22:38:53 DEBUG Recieved zigbee message with data {"cid":"genAnalogInput","data":{"presentValue":0}}
,2018-8-9 22:38:46 DEBUG Check online 0x00158d0001f4e65f
,2018-8-9 22:38:46 DEBUG Check online 0x00158d00016d94bf
,2018-8-9 22:38:46 DEBUG Check online 0x00158d0001b89324
,2018-8-9 22:38:46 DEBUG Check online 0x00158d000200c16b
,2018-8-9 22:38:46 DEBUG Check online 0x00158d000159a8b3
,2018-8-9 22:38:46 DEBUG Check online 0x00158d0001b8939e
,2018-8-9 22:38:47 ERROR Zigbee publish to '0x00158d00016d94bf', genOnOff - on - {} - 2 failed with error Error: AF data request fails, status code: 183. APS no ack.
,2018-8-9 22:38:46 DEBUG Check online 0x00158d000119911c

Is this?

Koenkk, 2018-08-10:
Great that it! 

138 = `0Xb7` -> [`ZApsNoAck`](http://dev.ti.com/tirex/content/simplelink_zigbee_sdk_plugin_1_60_01_09/docs/zigbee_user_guide/html/zigbee/developing_zigbee_applications/znp_interface/znp_interface.html#) (aka no acknowledge received from device). This looks like a connection issue to me.

jasimancas, 2018-08-10:
In both devices? this issue only do in two QBKG04LM

Koenkk, 2018-08-10:
At least for `0x00158d00016d94bf`

jasimancas, 2018-08-10:
Currently 21 devices are joined:
(0x00158d0001b91f59): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d0001f4124f): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
(0x00158d00013ff5f8): WXKG03LM - Xiaomi Aqara single key wireless wall switch (EndDevice)
(0x00158d0001f4e65f): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
(0x00158d000183b81a): WXKG03LM - Xiaomi Aqara single key wireless wall switch (EndDevice)
(0x00158d00015f2443): WXKG03LM - Xiaomi Aqara single key wireless wall switch (EndDevice)
(0x00158d0001b91f39): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d0001b89324): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
(0x00158d00013fe2dc): WXKG03LM - Xiaomi Aqara single key wireless wall switch (EndDevice)
(0x00158d00017102ad): WXKG03LM - Xiaomi Aqara single key wireless wall switch (EndDevice)
(0x00158d0001b95e19): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d000186f722): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
(0x00158d000152e953): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
(0x00158d000200c16b): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
(0x00158d0001e119ae): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
(0x00158d000159a8b3): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)
(0x00158d0001b8939e): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
(0x00158d000119911c): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
(0x00158d00015f2692): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
(0x00158d0001838725): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
(0x00158d00016d94bf): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)

More? Its a stability problem then?

Koenkk, 2018-08-10:
How is this device positioned in the network, is it further away than other devices?

jasimancas, 2018-08-10:
(0x00158d0001f4e65f): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router) and (0x00158d00016d94bf): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router) are in the same wall, in the same room, side by side.

Koenkk, 2018-08-10:
Does controlling of the QBCZ11LM always work?

jasimancas, 2018-08-10:
All devices work fine all time, 0 issues, only fail (0x00158d00016d94bf): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router) and (0x00158d000159a8b3): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router) sometimes (both).

Koenkk, 2018-08-13:
I was thinking that maybe the wall switch is somehow not connected to the router but directly to the coordinator. 

To verify: could you put the switch in very close proximity of the coordinator and test if you still get this error?

Koenkk, 2018-08-13:
So it seems that you get this error when the router doesn't route the messages correctly (sometimes). If you pair it while close to the coordinator, does it keep working?

Koenkk, 2018-08-20:
Yes, but this functionality has not yet been implemented in zigbee2mqtt.

jasimancas, 2018-08-21:
Are you planning to implant it?

Koenkk, 2018-08-21:
Yes! (no ETA)

luci-11, 2018-08-21:
> I was thinking that maybe the wall switch is somehow not connected to the router but directly to the coordinator.

@Koenkk I need one clarification about this sentence (i have more or less same config at home as @jasimancas ).
I have a sensor which is outside CC2531 range and a xiaomi plug marked as Router in Log between the sensor and the CC2531 but I do not receive any updates.
The sensor was initially paired to the CC2531 and worked then I placed it in another room and discovered that in this new position it was too far. So later I added the plug in the middle but the sensor is still unavailable. Should I remove the sensor from coordinator and do the pairing again from the new position? It is automatic that it will now connect to router? 
Thanks in advance for clarification

Koenkk, 2018-08-21:
@luci-11 can you try repairing while close to the router?

luci-11, 2018-08-24:
@Koenkk I tried to pair the sensor close to the router but I can't manage to pair it, as soon as I go more close to coordinator I receive the pairing message.  Doing this I noticed that maybe door sensor have more range than temperature and switches.  
(Just to complete the scene: Plug(router) has linkquality=141, Door sensor has linkquality =2, for temp and switches I have no messages after some time) 
I'll wait for network map to check what's going on in my house.. At worse I think I will buy another CC2531 to flash as router. Thanks for help.

jasimancas, 2018-08-24:
The network map will be a great function.

Koenkk, 2018-08-29:
The network map has now been implemented in the dev branch, see #252 for more information.

jasimancas, 2018-08-29:
I generate 

digraph G {
  "0x00158d0001b91f59" [label="0x00158d0001b91f59 (online)"];
  "0x00158d0001b91f59" -> "0x00124b00120232cb" [label="61"]
  "0x00158d00017102ad" [label="0x00158d00017102ad (online)"];
  "0x00158d00017102ad" -> "0x00124b00120232cb" [label="49"]
  "0x00158d0001f4124f" [label="0x00158d0001f4124f (online)"];
  "0x00158d0001f4124f" -> "0x00124b00120232cb" [label="26"]
  "0x00158d00013fe2dc" [label="0x00158d00013fe2dc (online)"];
  "0x00158d00013fe2dc" -> "0x00124b00120232cb" [label="38"]
  "0x00158d00013ff5f8" [label="0x00158d00013ff5f8 (online)"];
  "0x00158d00013ff5f8" -> "0x00124b00120232cb" [label="24"]
  "0x00158d0001f4e65f" [label="0x00158d0001f4e65f (online)"];
  "0x00158d0001f4e65f" -> "0x00124b00120232cb" [label="94"]
  "0x00158d000183b81a" [label="0x00158d000183b81a (online)"];
  "0x00158d000183b81a" -> "0x00124b00120232cb" [label="74"]
  "0x00158d0001b91f39" [label="0x00158d0001b91f39 (online)"];
  "0x00158d0001b91f39" -> "0x00124b00120232cb" [label="49"]
  "0x00158d00015f2443" [label="0x00158d00015f2443 (online)"];
  "0x00158d00015f2443" -> "0x00124b00120232cb" [label="170"]
  "0x00158d0001b89324" [label="0x00158d0001b89324 (online)"];
  "0x00158d0001b89324" -> "0x00124b00120232cb" [label="34"]
  "0x00158d0001b95e19" [label="0x00158d0001b95e19 (online)"];
  "0x00158d0001b95e19" -> "0x00124b00120232cb" [label="77"]
  "0x00158d000186f722" [label="0x00158d000186f722 (online)"];
  "0x00158d000186f722" -> "0x00124b00120232cb" [label="63"]
  "0x00158d000152e953" [label="0x00158d000152e953 (online)"];
  "0x00158d000152e953" -> "0x00124b00120232cb" [label="43"]
  "0x00158d000200c16b" [label="0x00158d000200c16b (online)"];
  "0x00158d000200c16b" -> "0x00124b00120232cb" [label="75"]
  "0x00158d0001e119ae" [label="0x00158d0001e119ae (online)"];
  "0x00158d0001e119ae" -> "0x00124b00120232cb" [label="71"]
  "0x00158d000119911c" [label="0x00158d000119911c (online)"];
  "0x00158d000119911c" -> "0x00124b00120232cb" [label="77"]
  "0x00158d0001b8939e" [label="0x00158d0001b8939e (online)"];
  "0x00158d0001b8939e" -> "0x00124b00120232cb" [label="1"]
}

Umm i think thats something are wrong.

Koenkk, 2018-08-29:
Seems that all your devices are connected to the coordinator. Can you try repairing the QBKG04LM close to the router you want to pair it to?

APS no ack = no connection

jasimancas, 2018-08-29:
Is it possible that the router will not route correctly?
Strange that all the devices are in the coordinator, aren't they distributed?

Koenkk, 2018-08-29:
It depends on how they are paired, I believe they will pair to the device closest to them when paired.

jasimancas, 2018-08-29:
When I installed the USB, I did the following:
1 - Install USB and Z2M and configure HA
2 - Paired the Routers devices

- (0x00158d0001f4e65f): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
- (0x00158d0001b89324): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
- (0x00158d000200c16b): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
- (0x00158d0001b8939e): QBCZ11LM - Xiaomi Aqara socket Zigbee (Router)
- (0x00158d000119911c): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
- (0x00158d00016d94bf): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)
- (0x00158d000159a8b3): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)

3 - I restarted and saw that everyone was fine
4 - Paired all other devices (EndDevices)

That's why I find it strange

luci-11, 2018-08-29:
I really think that xiaomi routers do not act as expected :'( @jasimancas situation is too similar to mine. 
I'm switching to dev branch to see if I have all devices too connected to the coordinator.
Any recommendations on how to do the switch?

jasimancas, 2018-08-29:
@luci-11 Recommendations to swtich to dev version?

luci-11, 2018-08-29:
yeah it works :) great!

here is my map and yes, anything is connected to the xiaomi plug (router)

digraph G {
  "0x00158d00022da787" [label="0x00158d00022da787 (online)"];
  "0x00158d00022da787" -> "0x00124b0012023473" [label="82"]
  "0x00158d0002486ca2" [label="0x00158d0002486ca2 (online)"];
  "0x00158d0002486ca2" -> "0x00124b0012023473" [label="170"]
  "0x00158d00024783a0" [label="0x00158d00024783a0 (online)"];
  "0x00158d00024783a0" -> "0x00124b0012023473" [label="1"]
  "0x00158d00022c8a75" [label="0x00158d00022c8a75 (online)"];
  "0x00158d00022c8a75" -> "0x00124b0012023473" [label="170"]
  "0x00158d00023471a0" [label="0x00158d00023471a0 (online)"];
  "0x00158d00023471a0" -> "0x00124b0012023473" [label="1"]
  "0x00158d00023f03ea" [label="0x00158d00023f03ea (online)"];
  "0x00158d00023f03ea" -> "0x00124b0012023473" [label="1"]
  "0x00158d0001bb5289" [label="0x00158d0001bb5289 (offline)"];
  "0x00158d0001bb5289" -> "0x00124b0012023473" [label="6"]
  "0x00158d0001bf944a" [label="0x00158d0001bf944a (online)"];
  "0x00158d0001bf944a" -> "0x00124b0012023473" [label="21"]
  "0x00158d00023fef72" [label="0x00158d00023fef72 (online)"];
  "0x00158d00023fef72" -> "0x00124b0012023473" [label="0"]
  "0x00158d0002333ba9" [label="0x00158d0002333ba9 (online)"];
  "0x00158d0002333ba9" -> "0x00124b0012023473" [label="25"]
}

jasimancas, 2018-08-29:
I use Docker, but yes, good work.

Possible bug @Koenkk ?

Koenkk, 2018-08-30:
@kirovilya how did you manage to pair devices to the xiaomi power plugs?

kirovilya, 2018-08-30:
@Koenkk I wrote here https://github.com/Koenkk/zigbee2mqtt/issues/252#issuecomment-411507339
Need to send pairing signal to router (power plug).

Koenkk, 2018-08-30:
@kirovilya thanks, this would mean that zigbee2mqtt has to execute these permitJoin commands when `permit_join: true`. 

Does this only apply to the Xiaomi power plug, or should this also be done for other routers?

kirovilya, 2018-08-30:
@Koenkk  By default, when you set permit_join: true - you can connect through any router and coordinator! Therefore, most devices are connected through the coordinator, but if you bring it closer to the router, it will connect through it.

And the above method makes it possible to connect only through a specific router (without a coordinator and other routers).

Yes, you can connect to any routers.

luci-11, 2018-08-31:
@kirovilya @Koenkk so if I understood correctly, It should already work if `permit_join:true`.
For me no way :( 

Here is my situation-map:
coordinator ---------- plug + temp sensor ---- door sensor (paired to coordinator and properly working _furthest sensor I have)
No sensor can be paired to the plug, and the plug itself and the temp sensor sometimes cannot be reached sending commands from HA...maybe I have a small black hole of coverage..(which means luck in placing the sensors ahahahah) or some sensors have less range than others.
I Will update you as soon as I receive the new CC2531 to be flashed as router. I'll put it in the same position of the plug and see what will happen.
Thanks again

jasimancas, 2018-09-05:
Today...

`
zigbee2mqtt:debug 2018-9-5 09:54:18 Recieved mqtt message on topic zigbee2mqtt/interruptor_entrada/set' with data 'ON',
zigbee2mqtt:warn 2018-9-5 09:54:18 Device with modelID 'undefined' is not supported.,
zigbee2mqtt:warn 2018-9-5 09:54:18 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices`

Koenkk, 2018-09-05:
@jasimancas did this only occur one time?

jasimancas, 2018-09-05:
@Koenkk First time, restarted system, but same problem, when i return to home, i will repair the device.

jasimancas, 2018-09-09:
Repaired, but today, error again, APS no ACK, but if i turn on manually the device, Homeassistant say me that the device is on, if i push in the HA to turn off, dont turn off and the log write the error :-(

goshi0, 2018-11-12:
I am having similar problems with 2 elements the other 14 are fine, but these two show:

`Zigbee publish to '0x00158d0001a41923', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 3`

and 

` failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.`

any hints ?

Koenkk, 2018-11-13:
@goshi0 the new firmware should solve this: https://github.com/Koenkk/zigbee2mqtt/issues/533

jasimancas, 2018-11-19:
zigbee2mqtt:info 2018-11-19 17:45:26 Zigbee publish to '0x00158d000252543a', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 2,
zigbee2mqtt:error 2018-11-19 17:45:44 Zigbee publish to '0x00158d000252543a', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 2 failed with error Error: AF data request fails, status code: 183. APS no ack.

jasimancas, 2018-11-19:
Maybe too much devices? I have 26, and i flash the #533 firmware. And same problem.

Koenkk, 2018-11-19:
183 means that the device cannot be reached. (probably a range/network issue)

jasimancas, 2018-11-19:
But it's temporary, two minutes later it works.

Koenkk, 2018-11-19:
It helps to add some stronger routers (e.g. CC2530 with CC2591).

jasimancas, 2018-11-19:
New hardware @Koenkk ?
Can you share a link to this?

Koenkk, 2018-11-19:
https://nl.aliexpress.com/item/ZigBee-Wireless-Module-CC2530-CC2591-PA-Module/1831284083.html?spm=a2g0z.search0104.3.16.60884d400HRv1P&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_5729915_319_317_5733915_10696_10924_453_10084_454_10083_5733715_10618_10920_10921_10304_10922_10307_10820_10301_10821_5730815_537_536_5733815_10843_10059_10884_5734015_10887_100031_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=91282617-b224-4570-a260-936bb7a2ffc2-2&algo_pvid=91282617-b224-4570-a260-936bb7a2ffc2 

In my case it drastically reduced the number of no ack error.

jasimancas, 2018-11-19:
Thank! In arduino @Koenkk ?

Koenkk, 2018-11-19:
I have it connected to a 220v to 3.3v power converter.

jasimancas, 2018-11-19:
OK, buy this then, any tuto?

jasimancas, 2018-11-19:
@luci-11 you ok?

[8 more replies]

closed 2019-01-26 by Koenkk