Koenkk/zigbee2mqtt #255: Philips Hue bulbs not pairing

jeroenterheerdt, 2018-08-06:
My setup is working fine, I was able to join Tradfri lights. Somehow my Philips Hue lights are not pairing. I tried resetting it in multiple ways, but it seems that does not help. I have reset my bridge, removed the lights from the bridge as well. I do not own a Philips remote so I cannot follow the reset procedure that is sometimes mentioned online. Just rapidly turning the lights on and off did not help. Does anybody here have experience in resetting Philips lights/bulbs?

Koenkk, 2018-08-07:
I've reset my hue go with a tradfri remote (as mentioned on the wiki)

jeroenterheerdt, 2018-08-07:
thanks. I had to buy a remote just for this. If anyone is having trouble resetting their Hue lights: buy a tradfri or hue remote (I bought a hue remote), hold it very close the to light that you want to reset and hold the on/off buttons for about 10 seconds. The light flashes a couple of times and then dims. It will now join.

closed 2018-08-07 by jeroenterheerdt

Koenkk, 2018-08-07:
@jeroenterheerdt thanks for letting us know what solved the problem :smile: