Koenkk/zigbee2mqtt #256: RTCGQ11LM no mqtt messages

Mdleal, 2018-08-06:
I am getting info messages in the logs about state changes but nothing about publishing mqtt messages.  I am getting message from my ge lights.

Koenkk, 2018-08-07:
Can you post your log?

Mdleal, 2018-08-07:
Here is some of the log file showing motion. LMK if you need the entire log file.
[https://pastebin.com/2gD4QSTJ](LOGS)

Koenkk, 2018-08-07:
Looks strange indeed, can you provide the whole log?

Mdleal, 2018-08-07:
removed a bunch of check online lines. LMK if u need more. I have a 10k line file but don't know where to upload it.

[https://pastebin.com/SJ1d6xQM](log)

Mdleal, 2018-08-07:
This is docker if that maters.

Koenkk, 2018-08-07:
Could you also provide the log at the start? (+- first 50 lines)

Mdleal, 2018-08-07:
[http://gofile.me/6zbCh/fdsjFsdn1](entire log)

I'll share off my synology. You will have to download.  7.1mb

Koenkk, 2018-08-07:
@Mdleal the problem is that the device hasn't been paired correctly:

```
2018-08-05T22:52:30.366Z - info: kitchen (0x00158d0001e550a6): unkown - undefined unknown (EndDevice)
```

Can you try repairing?

Mdleal, 2018-08-07:
@Koenkk That was [https://github.com/Koenkk/zigbee2mqtt/issues/210](this) device which has been removed.

Mdleal, 2018-08-07:
I having this issue with 2 RTCGQ11LM  that I paired at the same time.

Mdleal, 2018-08-08:
[database.db](https://pastebin.com/dXmGdgzD)
If this will help.

Koenkk, 2018-08-08:
In your startup log I don't see that you have paired any motion sensors:
```
2018-08-07T13:45:41.874Z - info: Currently 5 devices are joined:
2018-08-07T13:45:41.875Z - info: ge_bulb (0x7ce524000004a8d6): 22670 - GE Link smart LED light bulb, BR30 soft white (2700K) (Router)
2018-08-07T13:45:41.876Z - info: patio (0x7c25240000156f5c): 22670 - GE Link smart LED light bulb, BR30 soft white (2700K) (Router)
2018-08-07T13:45:41.877Z - info: dining (0x7ce524000012d433): 22670 - GE Link smart LED light bulb, BR30 soft white (2700K) (Router)
2018-08-07T13:45:41.877Z - info: undefined (0x00158d0001e550a6): unkown - undefined unknown (EndDevice)
2018-08-07T13:45:41.878Z - info: 0x00158d00022516ec (0x00158d00022516ec): unkown - undefined unknown (EndDevice)
```

So it is either `0x00158d00022516ec` or `0x00158d0001e550a6` which both are not paired correctly.

closed 2018-08-11 by Mdleal