Koenkk/zigbee2mqtt #257: Building own sensors with Arduino?

klein0r, 2018-08-07:
Has anyone already build their own sensors with Arduino? I'm searching for a small ZigBee (802.15.4) module like the CC1101 for 868/433MHz communication.

Already found Xbee and CC2530F256 - but those options are pretty expensive for small sensors ($13+). There must be a cheaper option...

Maybe it will be quite easy to build your own little cheap sensors for a lot of usecases (instead of buying the expensive parts from big companies).

Any module/library recommendations?

tb-killa, 2018-08-07:
Hallo Matthias, ich nutze hierzu cc2530 Module von AliExpress.
Leider sind zigbee Module doch recht teuer.

Hi Matthias, i'v used cc2530 from AliExpress. Sadly zigbee modules in General are sadly expensiv.

tb-killa, 2018-08-07:
Das ist auch ganz nett aber der Preis ist auch entsprechend :

Its also nice But the price isnt cheap. 

http://wiki.microduinoinc.com/Microduino-Module_CoreRF

ryanbeaton, 2018-08-07:
And that makes the Xiaomi sensors so attractive, they are so cheap!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

klein0r, 2018-08-07:
Yes you're right. But Xiaomi is also able to sell those cheap sensors with free shipping world wide and they (hopefully) still earn some money 😄 Sure, they sell hundreds of them, but I thought there must be an option to get those communication modules separately 😢 

Maybe it's possible to teardown those sensors and just use the ZigBee interface?!

GadgetUK, 2018-08-07:
Xiaomi typically use the JN5169 chip from NXP (formally Jeenic), unit price for these is $6.5 dropping to $4 when buying 500 for the chip only.  Take into account PCB, battery, Case, sensor components etc and it's soon going to be over the $10 mark.

klein0r, 2018-08-07:
@GadgetUK Thanks for that hint! Of course, but you can build some sensors which are not available on the market or available by Xiaomi.

tb-killa, 2018-08-07:
I´v used this type: https://de.aliexpress.com/store/product/2-4G-CC2530F256-Zigbee-Intelligent-Home-Networking-Wireless-Module-WITH-SMD-type-IPEX-Antenna-Interface-CC2530/1089303_32803335623.html
In combination with some sort of electronic like cr2450 battery or two aa battery and simple pcb from https://www.itead.cc/open-pcb/pcb-prototyping.html you´v could build great own hardware.

@klein0r what do you have in mind about type of sensor you would build ?

klein0r, 2018-08-07:
@tb-killa Do you have any library recommendations to work with the CC2530F256? Or code examples?

I have no specific ideas. Just to play around a bit with zigbee and this amazing project 😄 Maybe a moisture sensor to start with. I also have some load cells to detect when I'm in bed. Would be great to communicate status changes with ZigBee.

tb-killa, 2018-08-07:
I´v used the sample code of -Stack Home 1.2.2a. (Projects\zstack\HomeAutomation\).
You need IAR Embedded workbench for 8051 to compile (trial version .. or ...) 

Maybe i´v see some cool stuff at your youtube channel ;) ?

MiloshCZ, 2018-11-01:
@tb-killa Hi, have you tried using E18 modules with preflashed FW? Is it possible to join and use this ZigBee network?
And (I know, that it is not "good" question) you have IAR full version, or trial, or "..." :-D If you have "..." could you help me? ;-)

khzd, 2018-11-02:
+1
Hi,
I got this pair : https://fr.aliexpress.com/item/CC2530-Zigbee-Core-Board-Development-Board-Kit-IOT-Smart-Home-Wireless-Module-Packet-24MHz-256KB/32767454609.html?spm=a2g0w.search0104.3.1.4c185218h9u8uz&ws_ab_test=searchweb0_0,searchweb201602_4_5731312_10065_10068_319_10892_317_10696_5728812_10084_10083_10618_10304_5731412_10307_10820_10821_10302_5731212_5731112_10843_328_5730112_5732712_10059_10884_5731512_10887_100031_5731612_321_322_10103_10913_10912_5729112_5731712-10892,searchweb201603_55,ppcSwitch_0&algo_expid=bd42a371-59d4-40c5-a9b9-78ae0c2e4182-0&algo_pvid=bd42a371-59d4-40c5-a9b9-78ae0c2e4182

and would like to add some Air quality sensor, CO sensor ... 
any way to code the cc2530 like arduino library ?
thanks

MiloshCZ, 2018-11-05:
I have [CC2530 module](http://www.aliexpress.com/item/Free-Shipping-2Pcs-Lot-CDEBYTE-E18-MS1-PCB-2-5mW-200m-2-4GHz-CC2530-low-cost/32803052003.html) which is cheap but create simple program is impossible for me (I am HW designer, I can write something easy for arduino, but this....).

closed 2019-01-11 by stale[bot]

jtomasrl, 2019-08-26:
Is it mandatory to use z stack on those devices (cc2530)? is there any arduino like library such as for the esp8266