Koenkk/zigbee2mqtt #261: Zigbee Utility Meter Support

ben423423n32j14e, 2018-08-08:
I have a RainForest Eagle that gets usage from my utility meter via Zigbee.

https://rainforestautomation.com/rfa-z114-eagle-200/

Is it possible for zigbee2mqtt to support utility meters?

I have to provide my utility company with the mac address and IN (installer number) of the rainforest for it to connect.

I am assuming there is a Zigbee power profile or something?

ben423423n32j14e, 2018-08-08:
I've tried multiple times entering the CC2531 devices MAC and copying the installer code off the rainforest since it's mandatory. I see absolutely nothing appear in the zigbee2mqtt logs.

I think Zigbee Smart Energy is a seperate profile that needs to be integrated.

https://www.zigbee.org/zigbee-for-developers/smart-energy/

tb-killa, 2018-08-08:
Could be Z-Stack 3.0 relevant https://github.com/Koenkk/zigbee2mqtt/issues/211

ben423423n32j14e, 2018-08-08:
tb-killa I am pretty sure you are correct, reading through: http://www.ti.com/tool/Z-STACK#

"Z-Stack 3.0 combines multiple previous Zigbee profiles into one unified standard"

"Based on Zigbee PRO 2015 stack, it provides new and improved security modes, including Install Codes for out-of-band key exchange"

closed 2018-10-14 by stale[bot]