Koenkk/zigbee2mqtt #263: SmartThings - two questions

PianSom, 2018-08-08:
Hi

The wiki [suggests](https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-devices) that at least one SmartThings device is supported (STSS-MULT-001). 

However, I have not been successful in pairing this, in spite of many attempts, restarts, factory resets, etc. First question - does anyone have any experience of success? My errors are always: `ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000b10d3db (Error: Timed out after 10000 ms)` in spite of every effort to keep it awake.

Second question - I have a SmartThing socket, which is not, I know, supported. Nonetheless I have paired it (and get `WARN Device with modelID '3200-Sgb' is not supported` when it starts). My question is - will this device, which is specified in configuration.yaml and was identified on pairing as a router, actually act as a zigbee router even if not supported?

Koenkk, 2018-08-08:
- @okeung can you explain how you paired the STSS-MULT-001?
- yes, it's already in the zigbee network so it will act as a router

PianSom, 2018-08-16:
Hey @Koenkk - well, no answer there and I am still not having any luck.

You may want to remove it from the supported devices ...

tb-killa, 2018-08-16:
@PianSom
Do you Follow this for joining??
https://youtu.be/7kAlHsWNntg

Please provide database.db content for
``modelID '3200-Sgb'`` maybe we could add this too. 

@Koenkk is actually busy so please wait as this is a Hobby project :)

PianSom, 2018-08-16:
Hi @tb-killa 

Yes, that is exactly the process (or at least one of them!) that I have tried. I just did it again, and got the same error as in my original post.

Below are the lines from database.db which refer to the SmartThings socket. For some reason there are four lines with `id:5`

Yes, I do know this is a hobby project. I don't think replying after 8 days can really be considered as unreasonable nagging! :) 

{"id":5,"type":"Router","ieeeAddr":"0x000d6f00057b44a4","nwkAddr":1458,"manufId":4174,"manufName":"CentraLite","powerSource":"Mains (single phase)","modelId":"3200-Sgb","epList":[1],"status":"online","joinTime":1534349858,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2820,2821],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}}}}},"_id":"l5BLPY9YbAP5ltYT"}

{"id":5,"type":"Router","ieeeAddr":"0x000d6f00057b44a4","nwkAddr":1458,"manufId":4174,"manufName":"CentraLite","powerSource":"Mains (single phase)","modelId":"3200-Sgb","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2820,2821],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}}}}},"_id":"l5BLPY9YbAP5ltYT"}

{"id":5,"type":"Router","ieeeAddr":"0x000d6f00057b44a4","nwkAddr":1458,"manufId":4174,"manufName":"CentraLite","powerSource":"Mains (single phase)","modelId":"3200-Sgb","epList":[1],"status":"online","joinTime":1534350303,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2820,2821],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}}}}},"_id":"l5BLPY9YbAP5ltYT"}

{"id":5,"type":"Router","ieeeAddr":"0x000d6f00057b44a4","nwkAddr":1458,"manufId":4174,"manufName":"CentraLite","powerSource":"Mains (single phase)","modelId":"3200-Sgb","epList":[1],"status":"online","joinTime":1534350303,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2820,2821],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"stackVersion":2,"hwVersion":3,"manufacturerName":"CentraLite","modelId":"3200-Sgb","dateCode":"����������������","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}}}}},"_id":"l5BLPY9YbAP5ltYT"}

PianSom, 2018-08-16:
PS For me, at least, zigbee control of the SmartThings socket is a "nice to have" as I don't really use it, having embraced Sonoff/Tasmota.  But it is great as a router, and I am sure others will be very keen to use it as a device.

Koenkk, 2018-08-16:
Could you provide your log after pairing the socket and toggling it with the local button on it?

PianSom, 2018-08-16:
Here is the entry at the time I tried at the post above (with the log entry before and after for context):

````
2018-08-16T12:39:42.317Z - info: MQTT publish, topic: 'zigbee2mqtt/occupancy1', payload: '{"illuminance":60,"occupancy":false,"battery":"100.00","voltage":3045}'
2018-08-16T12:44:03.507Z - error: Cannot get the Node Descriptor of the Device: 0x000d6f000b10d3db (Error: Timed out after 10000 ms)
2018-08-16T13:04:17.003Z - info: MQTT publish, topic: 'zigbee2mqtt/cube1', payload: '{"battery":"100.00","voltage":3005}'
````

Koenkk, 2018-08-16:
Is that for the `3200-Sgb`? Because reading your first post you managed to pair it (otherwise you wouldn't get `WARN Device with modelID '3200-Sgb' is not supported`?)

PianSom, 2018-08-16:
No, that is for the `STSS-MULT-001` (which is supported), not the `3200-Sgb` (which is not; database entries for this are as posted above). It is the former that I am most keen to get working, not the latter.

Apologies - now I re-read your question I see you were asking about the socket, not the sensor, on pairing. IIRC I had no real problems pairing it. I have just restarted zigbee2mqtt and put the logs below. Would you like me to try and repair the socket?

````
Aug 16 19:43:14 hassbian systemd[1]: Stopping zigbee2mqtt...
Aug 16 19:43:14 hassbian systemd[1]: Stopped zigbee2mqtt.
Aug 16 19:43:14 hassbian systemd[1]: Started zigbee2mqtt.
Aug 16 19:43:16 hassbian npm[26989]: > zigbee2mqtt@0.1.2 start /opt/zigbee2mqtt
Aug 16 19:43:16 hassbian npm[26989]: > node index.js
Aug 16 19:43:18 hassbian npm[26989]: 2018-8-16 19:43:18 INFO Starting zigbee2mqtt version 0.1.2 (commit #a3d89c0)
Aug 16 19:43:18 hassbian npm[26989]: 2018-8-16 19:43:18 INFO Starting zigbee-shepherd
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO zigbee-shepherd started
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO Currently 6 devices are joined:
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO occupancy1 (0x00158d00023db779): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO water1 (0x00158d0002333132): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO cube1 (0x00158d0002565081): MFKZQ01LM - Xiaomi Mi smart home cube (EndDevice)
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO ST_socket (0x000d6f00057b44a4): unkown - undefined unknown (Router)
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO dimmer1 (0x000b57fffe8d1442): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO dimmer2 (0x000b57fffe96397f): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 WARN `permit_join` set to  `true` in configuration.yaml.
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 WARN Allowing new devices to join.
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 WARN Set `permit_join` to `false` once you joined all devices.
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO Zigbee: allowing new devices to join.
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO Connecting to MQTT server at mqtt://127.0.0.1
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO zigbee-shepherd ready
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO Connected to MQTT server
Aug 16 19:43:22 hassbian npm[26989]: 2018-8-16 19:43:22 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
Aug 16 19:43:22 hassbian npm[26989]: spinlock: false []
Aug 16 19:43:23 hassbian npm[26989]: spinlock: true []
````

PianSom, 2018-08-16:
Continuation of the log from above (additional lines that just arrived):
````
Aug 16 19:43:23 hassbian npm[26989]: spinlock: true []
Aug 16 19:43:59 hassbian npm[26989]: 2018-8-16 19:43:59 ERROR Failed to configure 0x000b57fffe96397f
Aug 16 19:44:05 hassbian npm[26989]: 2018-8-16 19:44:05 ERROR Failed to configure 0x000b57fffe8d1442
Aug 16 19:44:20 hassbian npm[26989]: 2018-8-16 19:44:20 INFO Device incoming...
Aug 16 19:44:20 hassbian npm[26989]: 2018-8-16 19:44:20 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
Aug 16 19:44:20 hassbian npm[26989]: 2018-8-16 19:44:20 WARN Device with modelID '3200-Sgb' is not supported.
Aug 16 19:44:20 hassbian npm[26989]: 2018-8-16 19:44:20 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
Aug 16 19:44:20 hassbian npm[26989]: 2018-8-16 19:44:20 WARN Device with modelID '3200-Sgb' is not supported.
Aug 16 19:44:20 hassbian npm[26989]: 2018-8-16 19:44:20 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
Aug 16 19:44:20 hassbian npm[26989]: device already in network
````
I then toggled the socket - no log entries occurred.

PianSom, 2018-08-16:
PS I don't know why the two TRADFRIs failed to configure. I only bought them yesterday, and they seemed to work ok. I'm not worried about it.

Koenkk, 2018-08-17:
Configure can fail because the device is in sleep mode (this is no problem as this only has to be successful once). 

For the 3200-Sgb, can you try following: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices ?

PianSom, 2018-08-17:
Hi @Koenkk 

As I mentioned, I don't really care about getting the 3200-Sgb configured, as I only want to use it as a router. As I said, it's a "nice to have" but not worth spending time on.

And I have tried everything I can think of to keep the STSS-MULT-001 awake but have had no success. My suspicion is that this is not actually supported, notwithstanding the documentation.

Koenkk, 2018-08-17:
- Your 3200-Sgb is already working as a router because it's in the network, so no further action is required.
- STSS-MULT-001: lets see if more users report this, in that case I will remove it from the supported devices.

matchett808, 2018-08-24:
I need to get a new battery for the smartthings multi sensor, I did get it to pair before (IIRC, hold down the button and let go of it _before_ the solid red light goes off - but that's just what I remeber) but removed it as I wasn't getting all of the reports I wanted from it.

If I have some time this weekend an can get my hands on some more batteries, I'll have a look at it.

matchett808, 2018-08-25:
The smartthings multi sensor I have is different - it's model is 'multiv4' - sorry don't think I'll be able to help with this for you

my tip about paring still stands - it took a few attempts for me, but that's how I got it to work

PianSom, 2018-08-25:
@matchett808 - thanks. 

Yes, it's a different model, I think. The supported one has a blue and green light.

I have tried *many* times to pair, but still no luck :(

matchett808, 2018-08-25:
Could try what's in this video: https://youtu.be/7kAlHsWNntg

PianSom, 2018-08-25:
Yes, I have tried that - and every combination I can think of with holding the button down for varying amounts of time etc - many, many, many times :(

matchett808, 2018-08-26:
hm....think I've got one of each actually (noticed it's a different board and slightly different plastics - will check it out an let you knw

closed 2018-11-09 by stale[bot]

davidedmundson, 2019-01-02:
FWIW, your device sounds like mine. It requires the new zapp-CIE support to register.

Version 1.0 with https://github.com/Koenkk/zigbee-shepherd-converters/compare/master...davidedmundson:master works