Koenkk/zigbee2mqtt #264: Include and provide issue-and-pull-request-templates

tb-killa, 2018-08-08:
Hello @Koenkk please follow up this guide and include this ones (**especially Issue Template**):
https://blog.github.com/2016-02-17-issue-and-pull-request-templates/

With this we could bring in some default informations like:
**``Please follow up first the wiki article for device integration`` 
``what type of zigbee2mqtt version is used``**

and many more ..

i think this could help to stop default questions about: 
**`` what i have to do ... ??``
`` will this be implemented .. ?? ``**

Koenkk, 2018-08-08:
Just added the first version of the issue template :smile:

tb-killa, 2018-08-08:
Nice .. i like it ..
We should link to .. how to do it .. like where i could check my version eg ..
We shouldn´t not just stare at a service like hastebin .. lets link also to pastebin or own simple logfile memory location.

tb-killa, 2018-08-10:
I see you change it again.
Nice work .. i would say we can now maintain this.
Should we close this issue ?

Koenkk, 2018-08-10:
Yes, thanks!

closed 2018-08-10 by Koenkk