Koenkk/zigbee2mqtt #265: ERROR Failed to start

matusbr, 2018-08-08:
Hello,
yesterday arrives my CC2531 USB sniffer and i make a new installation like the Guide on my raspberrypi 3 with "Raspbian GNU/Linux 9 (stretch)"
https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started
But it doesn't start.
I just read all Issues and make all steps in the Issues to the problem, but nothing happend
My steps without any results:
- refresh installation
- reflash the stick with CC2531ZNP-Prod.hex
- reflash the stick with CC2531ZNP-Pro-Secure_LinkKeyJoin.hex
- run normal
- run as daemon
- all steps as pi and as root

node --version `v8.11.3`
npm --version `5.6.0`

Here my Sticks
```
root@raspberry:/dev# ls -al /dev/serial/by-id
insgesamt 0
drwxr-xr-x 2 root root 120 Aug  8 11:45 .
drwxr-xr-x 4 root root  80 Aug  8 11:45 ..
lrwxrwxrwx 1 root root  13 Aug  8 11:45 usb-0658_0200-if00 -> ../../ttyACM1
lrwxrwxrwx 1 root root  13 Aug  8 11:45 usb-FTDI_FT232R_USB_UART_AI03D9PV-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root  13 Aug  8 11:45 usb-SHA_CUL433-if00-port0 -> ../../ttyUSB1
lrwxrwxrwx 1 root root  13 Aug  8 11:45 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0012022440-if00 -> ../../ttyACM0
```

Here the Debug Message

```
pi@raspberry:/opt/zigbee2mqtt/data $ DEBUG=zigbee-shepherd* npm start
> zigbee2mqtt@0.1.2 start /opt/zigbee2mqtt
> node index.js
2018-8-8 13:50:04 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":true}}'
2018-8-8 13:50:04 INFO Starting zigbee2mqtt version 0.1.2 (commit #a3d89c0)
2018-8-8 13:50:04 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +27ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +10ms
2018-8-8 13:50:07 INFO Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-8-8 13:51:07 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +1m
  zigbee-shepherd:request REQ --> SYS:osalNvRead +9ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +25ms
2018-8-8 13:51:10 ERROR Error while starting zigbee-shepherd!
2018-8-8 13:51:10 ERROR Press the reset button on the stick (the one closest to the USB) and start again
2018-8-8 13:51:10 ERROR Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```
My Filesystem
```
root@raspberry:/opt/zigbee2mqtt# ls -al
insgesamt 536
drwxrwxr-x  10 pi   pi     4096 Aug  8 13:18 .
drwxr-xr-x   5 root root   4096 Aug  7 16:40 ..
-rwxrwxr-x   1 pi   pi   105403 Aug  7 16:40 architecture.png
drwxrwxr-x   2 pi   pi     4096 Aug  8 13:50 data
drwxrwxr-x   2 pi   pi     4096 Aug  7 16:40 docker
-rwxrwxr-x   1 pi   pi       71 Aug  7 16:40 .dockerignore
-rwxrwxr-x   1 pi   pi       25 Aug  7 16:40 .eslintignore
-rwxrwxr-x   1 pi   pi      372 Aug  7 16:40 .eslintrc.json
drwxrwxr-x   8 pi   pi     4096 Aug  8 12:43 .git
drwxrwxr-x   2 pi   pi     4096 Aug  7 16:40 .github
-rwxrwxr-x   1 pi   pi      938 Aug  7 16:40 .gitignore
-rwxrwxr-x   1 pi   pi   171440 Aug  7 16:40 header.png
-rwxrwxr-x   1 pi   pi      211 Aug  7 16:40 index.js
drwxrwxr-x   3 pi   pi     4096 Aug  7 16:40 lib
-rwxrwxr-x   1 pi   pi    35141 Aug  7 16:40 LICENSE
drwxrwxr-x 235 pi   pi    12288 Aug  7 17:21 node_modules
-rwxrwxr-x   1 pi   pi       23 Aug  7 16:40 .npmrc
-rw-r--r--   1 pi   pi   144215 Aug  8 12:43 npm-shrinkwrap.json
-rwxrwxr-x   1 pi   pi     1268 Aug  7 16:40 package.json
-rwxrwxr-x   1 pi   pi     3064 Aug  7 16:40 README.md
drwxrwxr-x   2 pi   pi     4096 Aug  7 16:40 support
drwxrwxr-x   2 pi   pi     4096 Aug  7 16:40 .travis
-rwxrwxr-x   1 pi   pi     2409 Aug  7 16:40 .travis.yml
```

I hope anyone can help me. Two days reading the Issues and doing everything else, but nothing happend.
thx

Koenkk, 2018-08-08:
How did you flash your CC2531?

matusbr, 2018-08-09:
I flash my CC2531 like the getting start description. On a windows pc with the driver from Texas and the tool V1 from Texas. To do this, i must register an account.

closed 2018-08-09 by matusbr

matusbr, 2018-08-09:
Yesterday in the afternoon my Zwave doesn’t work, after I stop the zigbee2mqtt service, zwave runs again.
Are there a Probleme to run zwave and zigbee2mqtt on one rspi?

ryanbeaton, 2018-08-09:
@matusbr maybe the USB ports remaped on restart. That has happened to me.

matusbr, 2018-08-09:
Yesterday in the afternoon my Zwave doesn’t work, after I stop the mqqt service, zwave runs again.
Are there a Probleme to run zwave and zigbee2mqtt?