Koenkk/zigbee2mqtt #268: Disconnected Sniffer

Odianosen25, 2018-08-09:
Hello,

I use docker for running my system, and it has all been good thanks. Sometimes my sniffer gets disconnected abruptly (long story how and why), and it seems there is no way of knowing when this happens.

Can you confirm what happens when the sniffer is disconnected? Is there a message sent to the broker? It will be nice if like a status message is sent, so over MQTT, one can know that it has been disconnected, or any of the coordinators or something

Thanks and regards

Koenkk, 2018-08-10:
It is detected by the underlying libraries: 

```
  serialport:main binding.close finished +1ms
  cc-znp The serialport /dev/tty.usbmodem1421 is closed. +1s
```

Problem is that this even is not propagated yet to zigbee-shepherd, so zigbee2mqtt cannot act on it yet.

Odianosen25, 2018-08-10:
Ok many thanks for this @Koenkk, should I close the issue? Or you can close it if you feel no need leaving it open.

Regards

Koenkk, 2018-08-10:
We can leave it open for now :)

closed 2018-12-22 by stale[bot]