Koenkk/zigbee2mqtt #270: Motion sensors state behavior

tatunts, 2018-08-09:
Hello everyone.

I was wondering if this behavior is normal or there is something that can be done.

I've noticed that after migrating to this platform from the Xiaomi gateways, the motion sensor cannot maintain the ON state if there is still movement on the room. It only shows the 60 seconds trigger then stays off for a few seconds that does not detects anything and then sends another ON state to HA when motion is detected.

Here is a comparison of the same motion sensors, one with this solution and the other still with connected by the gateway:

[image]

the one on the bottom is the GW connected sensor, both are the RTCGQ11LM model.

Is this the default behavior? This impacts on my logs, history, recorder (db size) and also CPU, since I have automations that triggers by the ON state of these sensors.

Thanks a lot!

Koenkk, 2018-08-10:
I've seen some RTCGQ11LM which do not report any occupancy anymore for +- 1/2 min after occupancy has been detected. Could you post your log so we could verify this? It is important to constantly trigger the motion sensor during that period.

tatunts, 2018-08-10:
Will do @Koenkk! Thanks for you reply. I am pretty sure that the sensor goes "numb" for 30 seconds after triggering. Nothing is sent to HA.

mark078, 2018-08-14:
Hello, I have the same problems with the Xiaomi Motion with Lux Sensor.

If I paire the sensor it only works for 1-2 minutes, then only sporadically and at some point not at all. I have created a log of this: [log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/2287418/log2.txt)

I was in front of the sensor all the time, making hand movements.

The other motion sensor without Lux works perfectly, for several weeks.

I tested this on a clean fresh zigbee2mqtt environment, with only one sensor. 
As coordinator I have a CC2530 with external antenna. The range to the sensor is about 3 meters with an open door in between.

I have this problem at two different locations, with different hardware.

Koenkk, 2018-08-14:
@mark078 this sound like battery issues, have you already tried replacing?

mark078, 2018-08-14:
@Koenkk no, I haven't changed the battery yet, I'll do that later.
But measured with a multimeter, it has 3.01 volts and is from Panasonic.

I have two of these sensors that cause these problems, by the way.
I ordered one from Gearbest in April and the other from Joybuy in June.

Koenkk, 2018-08-14:
It seems that your sensors doens't report the occupancy that often (I've seen that more).

Can you try setting the [`occupancy_timeout`](https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration) to `300`?

mark078, 2018-08-14:
I entered it with occupancy_timeout: 300, it looked good at first, but not any more below. There was no reaction for over 11 minutes, although there was movement.

I'll connect it tomorrow with the original Xiaomi bridge. Maybe it'll give him a firmware update.

```
2018-08-14T19:55:22.273Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":119}}
2018-08-14T19:55:22.303Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":119}'
2018-08-14T19:55:22.331Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":119}}
2018-08-14T19:59:13.591Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"modelId":"lumi.sensor_motion.aq2"}}
2018-08-14T19:59:21.213Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":61}}
2018-08-14T19:59:21.238Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":61}'
2018-08-14T19:59:21.263Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":61}}

battery removed

2018-08-14T19:59:41.619Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-08-14T19:59:41.641Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":61,"occupancy":true}'
2018-08-14T19:59:41.665Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-08-14T19:59:53.279Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":111}}
2018-08-14T19:59:53.302Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":111,"occupancy":true}'
2018-08-14T19:59:53.345Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":111}}
2018-08-14T20:00:11.100Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":118}}
2018-08-14T20:00:11.117Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":118,"occupancy":true}'
2018-08-14T20:00:11.136Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":118}}
2018-08-14T20:00:13.140Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":105}}
2018-08-14T20:00:13.168Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":105,"occupancy":true}'
2018-08-14T20:00:13.197Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":105}}
2018-08-14T20:00:26.059Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":151}}
2018-08-14T20:00:26.076Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":151,"occupancy":true}'
2018-08-14T20:00:26.096Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":151}}
2018-08-14T20:00:28.088Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":124}}
2018-08-14T20:00:28.106Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":124,"occupancy":true}'
2018-08-14T20:00:28.125Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":124}}
2018-08-14T20:00:28.447Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":155}}
2018-08-14T20:00:28.463Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":155,"occupancy":true}'
2018-08-14T20:00:28.482Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":155}}
2018-08-14T20:00:28.803Z - debug: Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-08-14T20:00:28.848Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":155,"occupancy":true}'
2018-08-14T20:00:46.731Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":109}}
2018-08-14T20:00:46.748Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":109,"occupancy":true}'
2018-08-14T20:00:46.766Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":109}}
2018-08-14T20:00:48.758Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":122}}
2018-08-14T20:00:48.773Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":true}'
2018-08-14T20:00:48.790Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":122}}
2018-08-14T20:00:50.845Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":154}}
2018-08-14T20:00:50.873Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":154,"occupancy":true}'
2018-08-14T20:00:50.898Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":154}}
[... cut ...] Id":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"CBpQEdXZluddV4S1"}

```

Koenkk, 2018-08-15:
@mark078 I've never thought of this, but these difference could indeed be explained due to different firmware versions! I'm curious if a firmware update solves this problem.

mark078, 2018-08-15:
@Koenkk 
I'm curious too. Because the two sensors with Lux (RTCGQ11LM) I had never paired with the original bridge before. 
I only paired the Motion without Lux Sensor (RTCGQ01LM) and it works without any problems.

I'll report tomorrow... 

Here's something weird from tonight in the log ([0,8],null,null,null,null,0,0,null). Since then, there's been no more motion detection.

```
2018-08-14T21:01:11.961Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":122}}
2018-08-14T21:01:11.979Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":true}'
2018-08-14T21:01:12.008Z - debug: Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":122}}
2018-08-14T21:03:56.633Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false}'
2018-08-14T22:01:21.332Z - warn: Soft resetted zigbee
2018-08-14T22:01:21.373Z - info: zigbee-shepherd ready
2018-08-14T23:01:29.876Z - warn: Soft resetted zigbee
2018-08-14T23:01:30.127Z - info: zigbee-shepherd ready
2018-08-14T23:31:31.597Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":31,"4":5032,"5":33,"6":[0,8],"10":0,"11":0,"100":0}}}
2018-08-14T23:31:31.622Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-14T23:31:31.648Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,3005,null,31,5032,33,[0,8],null,null,null,0,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}
2018-08-15T00:24:15.705Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":31,"4":5032,"5":33,"6":[0,5],"10":0,"11":0,"100":0}}}
2018-08-15T00:24:15.719Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-15T00:24:15.740Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,null,null,null,[null,5]]}}
2018-08-15T01:17:51.900Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":31,"4":5032,"5":33,"6":[0,2],"10":0,"11":0,"100":0}}}
2018-08-15T01:17:51.917Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-15T01:17:51.941Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,3005,null,31,5032,33,[0,2],null,null,null,0,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}
2018-08-15T02:12:19.123Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":32,"4":5032,"5":33,"6":[0,3],"10":0,"11":0,"100":0}}}
2018-08-15T02:12:19.137Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-15T02:12:19.158Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,32,null,null,[null,3]]}}
2018-08-15T03:06:45.573Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":31,"4":5032,"5":33,"6":[0,3],"10":0,"11":0,"100":0}}}
2018-08-15T03:06:45.588Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-15T03:06:45.612Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,3005,null,31,5032,33,[0],null,null,null,0,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}
2018-08-15T04:01:16.422Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":30,"4":5032,"5":33,"6":[0,2],"10":0,"11":3,"100":0}}}
2018-08-15T04:01:16.446Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-15T04:01:16.468Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,30,null,null,[null,2],null,null,null,null,3]}}
[... cut ...]
2018-08-15T09:24:46.032Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":32,"4":5032,"5":33,"6":[0,41],"10":0,"11":34,"100":0}}}
2018-08-15T09:24:46.047Z - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":122,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-08-15T09:24:46.070Z - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,32,null,null,[null,41],null,null,null,null,34]}}
```

Koenkk, 2018-08-15:
@mark078 this is some random data from Xiaomi sensors (pretty normal).

mark078, 2018-08-15:
@Koenkk  Okay and the soft restart is normal, too? 
Because the sensor stopped responding to movement at all afterwards. :(

Small update.. I paired both motion with Lux sensors to the Xiaomi Bridge and both have been running without problems for about 3 hours. Motion detection works every 5 seconds without any failures.
I leave them connected overnight and take them back with me tomorrow and connect them to the CC2530.

[image]
[image]

Koenkk, 2018-08-15:
I didn't notice the soft restart, but this could also cause problems (like in #274). 

Could you try changing: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L14 to `const softResetTimeout = 9999999 * 1000; // seconds * 1000.`.

This should disable the soft restart.

mark078, 2018-08-15:
@Koenkk I'll try tomorrow. 
If you want me to test anything to fix the error, let me know. :)
I would rather use zigbee2mqtt than the xiaomi cloud solution. 

The rest like xiaomi switches, motion without lux sensor and cube works very well!

mark078, 2018-08-16:
Hey @Koenkk , not good news. I didn't see an update in the Mi Home app, there was only one for the Yeelight bulb. I had paired the motion sensor at Xiaomi bridge for about 16 hours.

I paired the motion sensor to zigbee2mqtt again. Unfortunately, the behaviour has not changed. :-(

It detects motion for a few minutes, then at some point it only transmits light values and then it no longer reacts at all for a long time. 
I also paired the cube and it continues to respond. There's only problem with the motion sensor with Lux.

I adjusted the controller.js (with a 9 less) and tried to set the channel to 14, but that didn't help.

My firmware for CC2530 is from ptvo: http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
Because the FW from here (CC2530ZNP-Pro-Secure_LinkKeyJoin.hex) doesn't start zigbee2mqtt: zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
P20 is connected to GND.

Koenkk, 2018-08-16:
@mark078 could you try this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2530/bin

mark078, 2018-08-16:
@Koenkk this firmware doesn't work, I tried it on two different CC2530s: https://de.aliexpress.com/item/CC2530-Zigbee-Module-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-2-4GHz/32767470382.html

I have flashed this several times...

Pics: https://www.dropbox.com/sh/ud0mx1igu9df0r5/AACqJfgwPYGdGam42xi5iCqSa?dl=0

Zigbee2mqtt start with CC2530ZNP-Prod.hex FW:
```
> zigbee2mqtt@0.1.2 start /opt/zigbee2mqtt
> node index.js

2018-8-16 19:50:53 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[14]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":false}}'
2018-8-16 19:50:54 INFO Starting zigbee2mqtt version 0.1.2 (commit #7dd1c75)
2018-8-16 19:50:54 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +198ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +38ms
2018-8-16 19:50:58 INFO Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-8-16 19:51:58 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +1m
  zigbee-shepherd:request REQ --> SYS:osalNvRead +34ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +11ms
2018-8-16 19:52:01 ERROR Error while starting zigbee-shepherd!
2018-8-16 19:52:01 ERROR Press the reset button on the stick (the one closest to the USB) and start again
2018-8-16 19:52:01 ERROR Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```

After that I flashed the one from ptvo again and it works.

```
> zigbee2mqtt@0.1.2 start /opt/zigbee2mqtt
> node index.js

2018-8-16 19:47:44 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[14]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":false}}'
2018-8-16 19:47:45 INFO Starting zigbee2mqtt version 0.1.2 (commit #7dd1c75)
2018-8-16 19:47:45 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +192ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 2 +387ms
  zigbee-shepherd:init Non-Volatile memory is changed. +46ms
  zigbee-shepherd Starting a hardware reset... +21ms
  zigbee-shepherd:request REQ --> SYS:resetReq +16ms
  zigbee-shepherd:msgHdlr IND <-- SYS:resetInd +899ms
  zigbee-shepherd:request RSP <-- SYS:resetReq +6ms
...
```

Koenkk, 2018-08-16:
@kirovilya do you know what could cause the CC2530 firmware not to work?

mark078, 2018-08-16:
@Koenkk I don't really have any idea why the FW doesn't work for me. I just installed it on a third platform (orange pi zero) and did the same thing there. ptvo firmware works, the others fw don't. 
But I still have a cc2531 usb-stick with my girlfriend, so I could try it at the weekend, if the xiaomi motion lux sensors work with it.

@kirovilya thank you, maybe it will work! I'll try and report it. :)

mark078, 2018-08-16:
@kirovilya sadly that didn't work. I flashed your FW first, then tried it one by one:
P04 + P20 connected to GND > did not work
P05 + P20 connected to GND > did not work
P04 + P05 + P20 connected to GND > did not work

The same again with the FW from Koenkk but there also receives the error "request timeout" from the coordinator. :-(

Then the firmware of ptvo flashed again and it runs, crazy. 
Only there is the possible problem with the motion sensors...

Do you have the same CC2530 as me? Double header with 6 pins from ChdTech?
And which Zigbee2mqtt version? I have the latest: Starting zigbee2mqtt version 0.1.2 (commit #7dd1c75)

Then one more question, do you have the Xiaomi Motion Lux sensors and do they work on you?

kirovilya, 2018-08-17:
@mark078 I think so yes. It is my chip https://www.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html
My RTCGQ11LM work fine.

In last chance try set `rtscts: false` in config.

I tried all the devices from here https://github.com/Koenkk/zigbee2mqtt/issues/52#issuecomment-390380532

mark078, 2018-08-17:
Mhhh that's strange, that the firmware works for you and not for me.

I had "rtscts: false" mainly in the config. But in the meantime I took it out, without success.

I will test the cc2531 with the xiaomi motion sensor on the weekend.
and flashing the cc2530 with linux, instead of windows.

Next week I can try the arduino flashing method. I had already flashed my first cc2530 board with arduino and the custom firmware and it was running. 
I can test this with the current FW and this:
https://github.com/Koenkk/zigbee2mqtt/issues/190#issuecomment-404192164

Koenkk, 2018-08-17:
@mark078 ok great that it works with the CC2531, at least we know the sensor is not the problem!

@ptvoinfo could you provide some help here?

mark078, 2018-08-17:
@Koenkk Yeah, I'm a little happy now, after a few days of fumbling around. :)

The motion sensor still works, actually as good as with the Xiaomi Bridge.

I will flash a newest firmware on the CC2531 stick tomorrow. 
If it doesn't work anymore, could be the debugger responsible for the problems.

Because the stick has flashed to me someone else where I had no debugger yet.

mark078, 2018-08-18:
I have reflashed the CC2531 Stick with this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin
The Xiaomi motion sensors also worked with it.

So the debugger must be okay.

After that I flashed the CC2530 with Linux and CC2530 Firmware, but the problem is still that zigbee2mqtt does not start.

```
2018-8-18 13:05:01 DEBUG Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudrate":115200,"rtscts":false}}'
2018-8-18 13:05:01 INFO Starting zigbee2mqtt version 0.1.2 (commit #7dd1c75)
2018-8-18 13:05:01 INFO Starting zigbee-shepherd
2018-8-18 13:05:04 INFO Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-8-18 13:06:04 INFO Starting zigbee-shepherd
2018-8-18 13:06:07 ERROR Error while starting zigbee-shepherd!
2018-8-18 13:06:07 ERROR Press the reset button on the stick (the one closest to the USB) and start again
2018-8-18 13:06:07 ERROR Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}
```
		
I will flash the CC2530 next week with an Arduino.

mark078, 2018-08-22:
I noticed something at firmware von @dzungpv

Freshly after pairing, the Xiaomi Motion with Lux Sensor has reported every 6 seconds when it has detected a motion. So in a row...

After several hours, it only transmits a motion detection every minute.

If I briefly press the pairing button on the motion sensor, it responds again recognizes motion every 6 seconds. 
That's not normal?

< push paring button >
```
2018-8-22 13:08:59 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3005,"3":33,"4":12712,"5":65,"6":[0,18],"10":0,"11":16,"100":0},"modelId":"lumi.sensor_motion.aq2"}}
2018-8-22 13:08:59 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":10,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:08:59 DEBUG Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,3005,null,null,12712,null,[null,18],null,null,null,null,16],"modelId":"lumi.sensor_motion.aq2"}}
2018-8-22 13:09:09 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":11}}
2018-8-22 13:09:09 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":11,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:09 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":11}}
2018-8-22 13:09:09 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:09 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":11,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:23 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":10}}
2018-8-22 13:09:23 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":10,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:23 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":10}}
2018-8-22 13:09:23 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:23 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":10,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:28 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":12}}
2018-8-22 13:09:28 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":12,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:28 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":12}}
2018-8-22 13:09:28 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:28 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":12,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:33 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:33 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":12,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:38 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":11}}
2018-8-22 13:09:38 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":11,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:38 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":11}}
2018-8-22 13:09:38 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:38 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":11,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:43 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":10}}
2018-8-22 13:09:43 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":10,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:43 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":10}}
2018-8-22 13:09:44 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:44 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":10,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:55 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":11}}
2018-8-22 13:09:55 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":11,"occupancy":true,"battery":"100.00","voltage":3005}'
2018-8-22 13:09:55 DEBUG Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":11}}
2018-8-22 13:09:56 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 13:09:56 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":11,"occupancy":true,"battery":"100.00","voltage":3005}'
```

There is still a problem, despite my waving in front of the sensor it sends an occupancy false after one minute and an occupancy true immediately afterwards.
[... cut ...]
2018-8-22 12:50:06 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":18,"occupancy":false,"battery":"100.00","voltage":3005}'
2018-8-22 12:50:06 DEBUG Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
2018-8-22 12:50:06 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002006aa6', payload: '{"illuminance":18,"occupancy":true,"battery":"100.00","voltage":3005}'
```

I think this is the same problem as here: https://github.com/Koenkk/zigbee2mqtt/issues/70

dzungpv, 2018-08-22:
@mark078 You should connect it to a Mi Home gateway and check for firmware update, i have two of them and they work fine, they say it will always sleep 1 minute and  work and then sleep again, when it sleep mean it not respond to any motion, you can use Automation in Hass to solve the problem

mark078, 2018-08-22:
@dzungpv ahhh ok. So after an hour, your motion sensors only respond for a minute at a time?
I guess that's to save energy.

My second problem.. I try to solve then times to set the occupancy_timeout higher than 60 seconds. 
because it is actually not correct when a motion takes place, which it then transmits occupancy false and then immediately true. maybe occupancy_timeout solves the problem.

Because of the firmware update to the original bridge, i paired the two sensors a few days ago. But there was no firmware update. Only for the bridge and xiaomi yeelight bulb.

But the sensors run much better with your firmware now. With the other firmware from ptvo for cc2530, the sensors didn't react at all at some point.

dzungpv, 2018-08-22:
@mark078 Yes, it sleep and wake up after 1 minute. have both RTCGQ11LM(2) and RTCGQ01LM(2).
It all the same behavior. You can read more FAQ in Mi home app, open you sensor and fin it in the thee dot icon.

You can try to place near the gateway and test like this.
1 connect to Xiaomi gateway  and 1 connect to cc2530/cc2530.
Both gateway connect to Hass and you can see if it has any difference.
When signal low, it could miss the data, you can try to replace the CR2032 battery.

That why i try to make cc2530 work, it have very far better signal quality compare to cc2531 usb

fast0490f, 2018-09-02:
Hi, is there a way to change the parameters of sleep 1 minute to 5 sec?

alinelena, 2018-09-09:
I am having similar issue.
xiaomi sensor without illumination works, with illumination it does not. this is with yesterday's code base.
in debug i get
```
Sep 09 08:35:29 housebrain.dl.ac.uk npm[5482]:   zigbee2mqtt:debug 9/9/2018, 8:35:29 AM Recieved zigbee message with data {"cid":"msIlluminanceMeasurement","data":{"measuredValue":22}}
Sep 09 08:35:29 housebrain.dl.ac.uk npm[5482]:   zigbee2mqtt:debug 9/9/2018, 8:35:29 AM Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}

```

but i do not seem to get any topic post.

Koenkk, 2018-09-10:
@alinelena can you post your complete log?

tatunts, 2018-09-20:
Hello everyone! Not sure how but it seems that this issue is fixed with the latest update:

[image]

No more 1 minute on\off states on history.

I will monitor it for the next few days.

Thanks!

closed 2018-09-20 by tatunts

Koenkk, 2018-09-20:
@tatuns Version 0.1.4 changed the default occupancy to 90 seconds, that has probably solved it.

jgyoldi, 2018-10-14:
> Hi, is there a way to change the parameters of sleep 1 minute to 5 sec?
> it is possible to simulate push pairing button?
> 
> Any ideas how to fix this.

Hello. Have you had any answer about this? I think this is the question. If we can simulate the reset push via software o send a "keep alive" command to the sensor, the problem will be solved.

alinelena, 2018-10-28:
@Koenkk this went under my radar... got fixed after some updates.

jgyoldi, 2018-10-29:
So,is it fixed in earlier updates? Does zigbee2mqtt in his latest update keep xiaomi motion sensor alive so they send al state changes even if they are within the first minute?

Regards,

Jose

Koenkk, 2018-10-29:
@jgyoldi after the cooldown period of 1 minute the sensor only sends it state once a minute. This is the behavior of the sensor.

jgyoldi, 2018-10-29:
You are right. The thing is, is there a way via software to reproduce the behaviour of the physical reset? Some kind of soft-reset. It could be sent if there is a detention after the cool down period. I there is no detection  after an hour the sensor van enter in cool down period of detection from every minute and if there is again a detection send again the soft-reset.

Koenkk, 2018-10-30:
If it's not possible using the Xiaomi gateway it probably isn't.

Ric199021, 2022-09-23:
Hi, still got this issue with occupancy_timeout 90 seconds. How bad it is if I put it on 180s? I have a big hallway and I'm using 2 sensor but sometimes one of them doesn't see movent. Can I solve my problem with the 180s or should I consider purchasing another sensor? Thanks!

danielbrunt57, 2023-07-31:
I have an RTCGQ01LM which was reporting "motion" after every 55 minutes followed by "no motion detected" 90s later, with no one home! Setting the occupancy_timeout specifically to 90 (which was 'blank' and should default to 90s) solved the problem.  It's been "no motion detected" for 3 hours now!