Koenkk/zigbee2mqtt #272: implementing RGB to xy Color conversion

tb-killa, 2018-08-10:
For easier work with Color on lights we should provide some sort of RGB to xy Color conversion as extension for zigbee-converter.

Koenkk, 2018-08-10:
Good idea!

tb-killa, 2018-08-10:
@Koenkk Lets build some sort of converter ;)
```
// From: https://github.com/usolved/cie-rgb-converter/blob/master/cie_rgb_converter.js
 * Converts RGB color space to CIE color space
 * @param {Number} red
 * @param {Number} green
 * @param {Number} blue
 * @return {Array} Array that contains the CIE color values for x and y
 */
function rgb_to_cie(red, green, blue)
{
	//Apply a gamma correction to the RGB values, which makes the color more vivid and more the like the color displayed on the screen of your device
	var red 	= (red > 0.04045) ? Math.pow((red + 0.055) / (1.0 + 0.055), 2.4) : (red / 12.92);
	var green 	= (green > 0.04045) ? Math.pow((green + 0.055) / (1.0 + 0.055), 2.4) : (green / 12.92);
	var blue 	= (blue > 0.04045) ? Math.pow((blue + 0.055) / (1.0 + 0.055), 2.4) : (blue / 12.92); 

	//RGB values to XYZ using the Wide RGB D65 conversion formula
	var X 		= red * 0.664511 + green * 0.154324 + blue * 0.162028;
	var Y 		= red * 0.283881 + green * 0.668433 + blue * 0.047685;
	var Z 		= red * 0.000088 + green * 0.072310 + blue * 0.986039;

	//Calculate the xy values from the XYZ values
	var x 		= (X / (X + Y + Z)).toFixed(4);
	var y 		= (Y / (X + Y + Z)).toFixed(4);

	if (isNaN(x))
		x = 0;
	if (isNaN(y))
		y = 0;	 
	return [x, y];
}
```

Koenkk, 2018-08-13:
RGB values are now supported!

E.G.
```zigbee2mqtt/0x90fd9ffffe6494fc/set```
```{"state":"ON","color":{"r":255,"g":215, "b":0}}```

closed 2018-08-13 by Koenkk

sjorge, 2019-08-10:
Is there also a way to go the other way around? Say if you have the x and y value from zigbee2mqtt but need rgb?

pgScorpio, 2020-05-04:
I've got the same question as sjorge.

Now we can set the color in RGB, but we still get the current state in xy, so I need to convert that back to RGB as well.

AND-8, 2020-07-01:
> RGB values are now supported!
> 
> E.G.
> `zigbee2mqtt/0x90fd9ffffe6494fc/set`
> `{"state":"ON","color":{"r":255,"g":215, "b":0}}`

HI  @Koenkk ,
first of all THANK YOU. Zigbee2MQTT and your support is awesome.

But is think there is still something wrong in the implementation.
I am using the Osram LED Stripe RGBW https://www.zigbee2mqtt.io/devices/4052899926110.html

Within RGB is should be a brightness information but Zigbee2Mqtt seems to ignore it.

My Setup:
Loxone Mini Server + MQTT on Loxberry + Zigbee2MQTT on a another Pi

Sending color Code via HEX or RGB is working for Color but without brightness values.
HSV Color is working via MQTT Explorer also with brightness values.

Do you have an idea how to get the Brightness Information out of the RGB values within the conversion in Zigbee2MQTT?
Or is this a specific problem with the OSRAM Led Stripe?

pgScorpio, 2020-07-01:
> Sending color Code via HEX or RGB is working for Color but without brightness values.

This is correct, th RGB color value should be the color at maximum brightnrss, and so at least one of the r,g,b values should be 255.

> Do you have an idea how to get the Brightness Information out of the RGB values within the conversion in Zigbee2MQTT?

Get max(r,g,b)/255, this is the brightness (0.0 - 1.0).
and all values of RGB should be divided by this brightness value to make the largest value(s) 255.

AND-8, 2020-07-07:
> > Sending color Code via HEX or RGB is working for Color but without brightness values.
> 
> This is correct, the RGB color value should be the color at maximum brightness, and so at least one of the r,g,b values should be 255.
> 
> > Do you have an idea how to get the Brightness Information out of the RGB values within the conversion in Zigbee2MQTT?
> 
> Get max(r,g,b)/255, this is the brightness (0.0 - 1.0).
> and all values of RGB should be divided by this brightness value to make the largest value(s) 255.

Is it possible to add this directly into Zigbee2Mqtt?