Koenkk/zigbee2mqtt #274: WARN Soft resetted zigbee

Sjeff, 2018-08-10:
I Installed Zigbee2MQTT a couple of days ago but im running in to some weird warnings.
After the warning no mqtt messages are published anymore. And im unable to find any more errors/warnings in the log files. 

I can see that the time in the docker container is 2 hours off (21:11 in log is 23:11 my time zone).
```
2018-8-10 20:36:16 WARN Soft resetted zigbee
2018-8-10 20:36:16 INFO zigbee-shepherd ready
```
I reflashed the CC2531 sniffer and redownloaded the image to recreate the docker container.
Running for a few hours and the same warning pops up.

I'm running zigbee2mqtt version 0.1.2 (commit #a3d89c0) on ubuntu 18.04 docker.
Docker version 18.06.0-ce, build 0ffa825

MD5 hash of CC2531ZNP-Prod.hex:
3e90df9326d777ab54226413ffba04b6
SHA256 hash of CC2531ZNP-Prod.hex:
08e9ee6f1af54a87f38c059a7ecb54051c20a6afb10c0a6748c362e46552939c

Current configuration.yaml :
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://example:1883'
  user: username
  password: password
  client_id: CC2531
serial:
  port: /dev/zigbee2mqtt
devices:
  '0x00158d000245f2bc':
    friendly_name: Master_Bedroom
    retain: true
    qos: 1
advanced:
  log_level: debug
```

debug log after restart no new mqtt messages are published : https://pastebin.com/Ae05d86Z

Sjeff, 2018-08-11:
I created a new docker container. (for raspberry pi this time)
I used :  koenkk/zigbee2mqtt:arm32v6
after a couple of hours running the same warning occurred .

Full start to warning debug log: https://pastebin.com/T8MwxG1M

tb-killa, 2018-08-12:
``Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)`` could be failure when the tty port is being used by something else.
``Soft resetted zigbee`` appears (I think) when a timer runs inside zigbee-shepherd or is not reset.
Maybe @Koenk can tell you more about this.?

Sjeff, 2018-08-13:
I switched back to the Ubuntu version and changed the /def/zigbee2mqtt back to the original /dev/ttyACM1. 
Its running now for almost a full day but then the same warning appears.
Debug log doesn't provide any more information then i already put in the last comment.

Koenkk, 2018-08-13:
This soft reset mechanism is triggered once no messages are received for 1 hour. I still found this a clumsy way to do this. Can you try if the error doesn't appear anymore when changing https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L14 to 

```js
const softResetTimeout = 9999999 * 1000; // seconds * 1000.
```

Sjeff, 2018-08-14:
I changed the line to 
```
const softResetTimeout = 9999999 * 1000; // seconds * 1000.
```
and can confirm the warning message doesn't show anymore.

Also i think the sensor is to far away from the USB controller.
Moved the sensor and now its working a lot better.

Time to order a second CC2531 and configure it as router.

tb-killa, 2018-08-18:
@Sjeff what type of Devices you joined to your network?
Normaly every xiaomi device send some sort of heartbeat, maybe we have to increase the timeout value only.

Sjeff, 2018-08-18:
@tb-killa I only use the Xiaomi Aqara Temp sensors at the moment.

fcasarsa, 2018-08-20:
Hi,  i was able to recover from a stuck error resetting the USB port this way:

`echo '1-1' |sudo tee /sys/bus/usb/drivers/usb/unbind ; sleep 1 ;  echo '1-1' | sudo tee /sys/bus/usb/drivers/usb/bind`

it drops ethernet so run it in a nohup or screen session if you do it remotely.

Next time i will try this tool :

https://github.com/mvp/uhubctl

and reinforce systemctl launch script.

caldersm, 2018-08-24:
I ran into this same issue last night.  Just got my test hardware in yesterday, and was able to flash my CC2531(Thanks!).   It took me over an hour to get the door sensor to connect....a lot of trial and error with the reset pin.  Anyway....it was late and I put the sensor in a box on the floor, and the reception was really poor...and you can see that the issue happened when the link quality was below 50%, and it stopped as soon as I took it out this morning.  Just FYI.....I am a noob here...but thought I could contribute.  Notice that when I took it out of the box, the LQ went up to 63%.

One question....is the battery number next to link quality....a percentage out of 100...?

[image]

Koenkk, 2018-08-24:
The battery is indeed in percentages (0-100).

caldersm, 2018-08-27:
Koen,

The fix for the timeout....
` softResetTimeout = 9999999 * 1000; // seconds * 1000.`

Is that going to be incorporated into a new firmware release...?  If I get below 50 on link quality, it starts to throw these soft resets, and then then the unit will publish the payload every hour.   If I open the door, it responds fine....albiet....it might be one second more, if it is further away, but it still works.  If it works when I open the door, do I still need to put the repeaters all over the house, or is it ok to run with less than 50 on link quality...?

chenchen119, 2018-09-17:
I have a Sengled light bulb and a IKEA TRADFRI wireless dimmer. I noticed the same thing after a soft reset, no more mqtt msg would come in for the dimmer.

```
zigbee2mqtt:debug 2018-9-17 12:34:41 Recieved zigbee message with data {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":78,"cmdId":"moveToLevelWithOnOff","payload":{"level":0,"transtime":1}}
  zigbee2mqtt:info 2018-9-17 12:34:42 MQTT publish, topic: 'zigbee2mqtt/0x000b57fffe11fb3b', payload: '{"brightness":0}'
  zigbee2mqtt:debug 2018-9-17 12:36:35 Saving state to file /share/zigbee2mqtt/state.json
...
 zigbee2mqtt:warn 2018-9-17 13:34:47 Soft resetted zigbee
  zigbee2mqtt:info 2018-9-17 13:34:47 zigbee-shepherd ready
  zigbee2mqtt:debug 2018-9-17 13:35:28 Recieved zigbee message with data "0x000b57fffe11fb3b"
...(repeat 10 times)
  zigbee2mqtt:info 2018-9-17 13:36:31 Device incoming...
  zigbee2mqtt:info 2018-9-17 13:36:31 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2018-9-17 13:36:31 Recieved zigbee message with data "online"
  zigbee2mqtt:debug 2018-9-17 13:36:33 Recieved zigbee message with data "0xb0ce1814030d0dae"
...(repeat a few times)
```

The light bulb "0xb0ce1814030d0dae" still works after reset, but no mqtt is received for the dimmer until after a restart of zigbee2mqtt.

rcarmo, 2018-09-27:
I am getting bitten by this too. The soft reset keeps being triggered.

Koenkk, 2018-10-02:
Soft reset is disabled by default, can be enabled by settings `advanced`.`soft_reset_timeout` in seconds. (was 3600 seconds previously)

closed 2018-10-02 by Koenkk

rcarmo, 2018-10-03:
Thanks. I see it in dev - when can we expect a new release?

Koenkk, 2018-10-04:
@rcarmo first need to fix #373