Koenkk/zigbee2mqtt #276: Support for Xiaomi MiJia Honeywell Gas Leak detector (JTQJ-BF-01LM/BW)

IAmTheDude, 2018-08-11:
Hi all,

I have been given a [Xiaomi Gas Sensor](https://files.xiaomi-mi.com/files/MiJia_Honeywell/MiJia_Honeywell_Gas_Leak_Detector_EN.pdf) to take a look at.

I don't have a dev environment to play around in (running only HASSIO) but I have made a start on adding the device if someone is able to pick this up?

I have based it on the smoke detector (smoke on/off) using 'natgas' [as stated by the HASSIO docs](https://www.home-assistant.io/components/binary_sensor.xiaomi_aqara/):

To Pair: 

Plug the device in and wait for around 5mins while it performs its self-tests.
It will beep a couple of times after and the green light will be solid and is ready to be paired.

Press the button three times quite quickly in succession:

```
2018-08-11T18:03:16.148Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d000125c8fa"}'
2018-08-11T18:03:16.150Z - warn: Device with modelID 'lumi.sensor_natgas' is not supported.
```

As I cannot add the changes to the files in HASSIO (If I can I may have missed it in the docs?)

I had a crack at what they might look like:

zigbee-shepherd-converters/devices.js:

    {
        zigbeeModel: ['lumi.sensor_natgas'],
        model: 'JTQJ-BF-01LM/BW',
        description: 'MiJia Honeywell gas leak detector',
        supports: 'natgas',
        vendor: 'Xiaomi',
        fromZigbee: [fz.xiaomi_battery_3v, fz.JTQJBF01LMBW_natgas, fz.ignore_basic_change],
        toZigbee: [],
    },

converters/fromZigbee.js:

    JTQJBF01LMBW_natgas: {
        cid: '**unknown**',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            return {natgas: **unknown** === 1};
        },
    },

I hope this is enough for a start but I am happy to help where I can with this?

tb-killa, 2018-08-12:
Please provide database.db entry(s) for this Enddevice.
Which debug output you get (best is to provide the content e.g. on pastebin) if you add the device and omit the ``fz.JTQJBF01LMBW_natgas`` in the converter ??

IAmTheDude, 2018-08-12:
Hi tb-killa,

Just tried to open the db but its passworded?

I can't add the converter for debug as I am running Hassio on a Pi and don't believe I have access to the files to make changes?

I can set a spare Pi up with a Linux environment and attempt to get something setup if its an help?

tb-killa, 2018-08-13:
Normaly the db isn´t passworded .. @danielwelch could maybe help you here ..
As i don´t use hassio (i use node-red with own logic) we have to ask for another one to help , maybe @Koenkk could .. ?

Koenkk, 2018-08-13:
A spare pi with just running by https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge till step 4 is the easiest way to develop. I would highly recommend doing that.

IAmTheDude, 2018-08-16:
That's brill, I've just received and flashed another stick so I can setup a separate Dev Pi to avoid messing with my current setup.

Leave it with me and I will have a proper play at the weekend.

tb-killa, 2018-08-16:
Great, we need after successful pairing your database.db content.
Actually we are really busy But we will bring in new content if time accept :)

IAmTheDude, 2018-08-20:
So had a look last night and think I'm getting somewhere.

Its a little rough around the edges but the result so far is:
```
zigbee2mqtt:info 2018-8-20 09:04:09 Currently 1 devices are joined:
zigbee2mqtt:info 2018-8-20 09:04:09 0x00158d000125c8fa (0x00158d000125c8fa): JTQJ-BF-01LM/BW - Xiaomi MiJia Honeywell gas leak detector (Router)
 
zigbee2mqtt:info 2018-8-20 09:04:09 MQTT publish, topic: 'zigbee2mqtt/0x00158d000125c8fa', payload: '{"natgas":false,"linkquality":47}'
```

[database.db copied straight after the pairing](https://www.dropbox.com/s/if2didfqpzz2n8l/databasegas.db?dl=0) (still passworded so have uploaded it)

Converter warnings & the output to console, the .js entries at the bottom: https://pastebin.com/8epafkXT

I based it on the smoke sensor but got the result above using 
```
cid: 'ssIasZone',
type: 'statusChange',
```
rather than 
```
cid: 'genBasic',
type: 'attReport',
```
which smoke uses:

I've left the battery function in as I noticed the 65281 in the data but the sensor is mains powered, I don't know if that makes a difference or if its sending any voltage through?

And I have no idea what or where the linkquality is coming from :|

I've got the gateway somewhere so will hook it up to that to see what it sends when using it as intended.

(Also, apologies, I'm not a dev\coder so am stumbling my way through this :D)

tb-killa, 2018-08-20:
Do you test the return value?
Does the device return true value if you use some sort of Gas?
If device could be powered by battery we should Transport the Informations.

IAmTheDude, 2018-08-20:
I've picked up a lighter this morning to manually try triggering it tonight (I didn't have an extension lead long enough to get to the hob last night :))

Its definitely only mains powered (the smoke is battery).

IAmTheDude, 2018-08-21:
It was harder than expected to get the sensor to beep using just a lighter so I sprayed it with some lighter refill (the aerosol one) and got:

```
zigbee2mqtt:info 2018-8-21 08:15:53 MQTT publish, topic: 'zigbee2mqtt/0x00158d000125c8fa', payload: '{"natgas":true,"linkquality":123}'

zigbee2mqtt:info 2018-8-21 08:15:56 MQTT publish, topic: 'zigbee2mqtt/0x00158d000125c8fa', payload: '{"natgas":false,"linkquality":120}'
```
Which set the alarm off.

The 'false' directly after is me pressing the button to turn it off.

tb-killa, 2018-08-21:
Nice Look like you got the functionality implementation ;)
We now need somebody who could verify this :)

rzhukov, 2018-08-24:
Hi, guys!
I also have such sensor and it works with similar config already for two month. I can confirm that if it senses a gas, the first parameter in payload is true.
Here is my config:
fromZigbee
```
JTQJBF01LMBW_gas: {
    cid: 'ssIasZone',
    type: 'statusChange',
    convert: (model, msg, publish, options) => {
        return {gas: msg.data.zoneStatus === 1};
    },
}
```
devices:
```
{ 
    zigbeeModel: ['lumi.sensor_natgas'],
    model: 'JTQJ-BF-01LM/BW',
    vendor: 'Xiaomi',
    description: 'MiJia Gas Leek Sensor ',
    supports: 'alarm_co',
    fromZigbee: [fz.xiaomi_battery_3v, fz.JTQJBF01LMBW_gas, fz.ignore_basic_change],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
}
```

tb-killa, 2018-08-24:
@rzhukov would you please provide a PR for the converter?

tb-killa, 2018-08-27:
@IAmTheDude Thanks to @rzhukov there is now a PR merged with working converter.
Would you please test this and report if work ?
If everything seems to be okay we could close this.

IAmTheDude, 2018-08-28:
That's awesome thank to you both!

Ill test that later tonight on my main HassOS box.

Would it be merged in with the master or shall I jump over to the edge version for testing?

tb-killa, 2018-08-28:
Should be on ``master`` .. please check :)

IAmTheDude, 2018-08-28:
Just checked bot master and dev (fresh installs of both) but not there yet.

No rush for it though :)

tb-killa, 2018-08-30:
@IAmTheDude Normaly @Koenkk push it arround 2+ weeks to new final version.
but you could check via manually copy files or modify and do test ;)

rzhukov, 2018-09-16:
Hi! I'm trying to support more features of the device. I've sniffed packets between original hub and the sensor. Now I need some help.
Here is zcl packet for requesting sensitivity levels:
[image]

The question is how can I generate zcl packet with manufacturer code and custom frame control?

Koenkk, 2018-09-16:
@dzungpv do you know how to do this?

rzhukov, 2018-09-17:
I've managed to get it work.
First, I found a way to construct frame with manufacturer code
```
device.foundation(
    0x0500,
    0x00,
    [{attrId: 0xFFF0}],
    {manufSpec: 1, disDefaultRsp: 1},
    (err, data) => { console.log(err, data); }
);
```
But faced an error:
```
/app/node_modules/zigbee-shepherd/lib/components/af.js:341
        manufCode = dstEp.getManufCode();
                          ^
TypeError: dstEp.getManufCode is not a function
    at Object.af.zclFunctional (/app/node_modules/zigbee-shepherd/lib/components/af.js:341:27)
    at ZShepherd._functional (/app/node_modules/zigbee-shepherd/lib/shepherd.js:671:20)
    at Endpoint.ep.functional (/app/node_modules/zigbee-shepherd/lib/shepherd.js:539:25)
    at Timeout.setTimeout [as _onTimeout] (/app/node_modules/zigbee-shepherd-converters/devices.js:301:24)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
```
After fix the issue in zigbee-shepherd, I recieved response from device:
```
{ cmdId: 0, statusCode: 132 }
```
That's because the device has manufacturer id `4151` in database, but command requires manufacturer code `4447`. So, I have to implement support for custom manufacturer code in foundation command. 
After that I've received
```
zigbee2mqtt:warn 2018-9-17 19:26:26 No converter available for 'JTQJ-BF-01LM/BW' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"65520":"0x0201000038010000"}}'
```
I'm still can't figure out, how to decode that data.

But I will make a PR to zigbee-shepherd with fixes.

dzungpv, 2018-09-17:
@rzhukov You don't need manufacture code with foudation command. Do you save the file when capture the package from Ubiqua? Upload it here so i can check. You can try with original hub and all app feature, capture the packet and note the id with the function you use and send the file.

dzungpv, 2018-09-18:
@rzhukov You can try new foundation command in toZigbee[] with zigbee-sepherd-converters and dev branch:
```
    read_sensor_natgas: {
        key: 'read',
        attr: ['sensitivity'],
        convert: (value, message) => {
            return {
                cid: '0x0500',
                cmd: '0x00',
                type: 'foundation',
                zclData: {
                    attrId: 0xFFF0, // presentValue
                    dataType: 0x39, // dataType
                },
            };
        },
    },
```

rzhukov, 2018-09-18:
@dzungpv Is this command equal to what you suggested?
```
device.foundation(
  0x500,
  0x00,
  [{
  attrId: 0xFFF0, // presentValue
  dataType: 0x39, // dataType
  }],
  (err, data) => { console.log(err, data); }
);
```
If so, I got following
```
null, [ { attrId: 65520, status: 134 } ]
```
So, looks like I need manufacturer code.

Here is sniffed packets:
[gas_sensitivity_levels_request.zip](https://github.com/Koenkk/zigbee2mqtt/files/2391458/gas_sensitivity_levels_request.zip)

dzungpv, 2018-09-18:
@rzhukov I can help you decode the data, but please take more steps:
1, capture all the packet
2, Set sensitivity level to difference value (3 or more) and then note the packet with that level , they often use 'write Analog Output' to the device
And finally provide the packet from 1 and 2, connect device to cc2531 and send raw data you receive from zigbee2mqtt and decode value when you run read command.
And i don't know what your software you use? Old version Ubiqua?

Dynamo90, 2018-09-20:
Hi,

I have just updated to 0.1.5 to test the Gas sensor but i have received this error on the converter.

`No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":33,"5":29,"8":8456,"9":770,"100":0,"150":0}}}'`

i do however get the following device info before the above error.
`Gas Sensor (0x00158d000125c8fa): JTQJ-BF-01LM/BW - Xiaomi MiJia gas leak detector  (Router)`

`MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d000125c8fa/gas/config', payload: '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.gas }}","device_class":"gas","json_attributes":[],"state_topic":"zigbee2mqtt/Gas Sensor","availability_topic":"zigbee2mqtt/bridge/state","name":"Gas Sensor"}'`

I have also tested the edge version just in case but still the same error.
Many Thanks,

WebSpider, 2018-11-01:
I'm also regularly seeing the same message as @Dynamo90:
```2018-11-1 23:57:17 - warn: No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":38,"5":10,"8":8456,"9":512,"100":0,"150":0}}}'```

This seems to be some kind of regular update, as it is transmitted once every ~5m

Koenkk, 2018-11-02:
Is't definitely not an error.

Does this device work properly? Then we can ignore this message.

Koenkk, 2018-11-02:
Yes this indeed is some vendor specific information.

If we know what it means we can parse it accordingly.

WebSpider, 2019-01-01:
Unstale

matlab22, 2019-01-20:
Pressing the button on the gas detector will trow the following:
```
zigbee2mqtt:warn 1/20/2019, 5:32:16 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:warn 1/20/2019, 5:32:16 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_natgas"}}'
```
It would be great if this would be parsed as a click

This is a test with butane lighter gas (I pressed the button to mute the alarm):
```
Jan 20 17:32:01 loxberry npm[1740]:   zigbee2mqtt:info 1/20/2019, 5:32:01 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000249fd8b', payload '{"gas":true,"linkquality":126}'
Jan 20 17:32:16 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 5:32:16 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":133,"150":1124073472}}}'
Jan 20 17:32:16 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 5:32:16 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 17:32:16 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 5:32:16 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_natgas"}}'
Jan 20 17:32:16 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 5:32:16 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 17:32:16 loxberry npm[1740]:   zigbee2mqtt:info 1/20/2019, 5:32:16 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000249fd8b', payload '{"gas":true,"linkquality":131}'
Jan 20 17:32:49 loxberry npm[1740]:   zigbee2mqtt:info 1/20/2019, 5:32:49 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000249fd8b', payload '{"gas":false,"linkquality":110}'
```
Every ~5min it will send somthing like:
```
Jan 20 16:12:12 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:12:12 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:12:12 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:12:12 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:17:12 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:17:12 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:17:12 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:17:12 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:22:19 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:22:19 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:22:19 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:22:19 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:27:35 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:27:35 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:27:35 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:27:35 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:32:47 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:32:47 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:32:47 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:32:47 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:33:22 loxberry npm[1740]:   zigbee2mqtt:info 1/20/2019, 4:33:22 PM MQTT publish: topic 'zigbee2mqtt/Aquara Cube', payload '{"battery":55,"voltage":2995,"linkquality":155}'
Jan 20 16:38:05 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:38:05 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:38:05 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:38:05 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:43:27 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:43:27 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:43:27 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:43:27 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
Jan 20 16:48:57 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:48:57 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 16:48:57 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 4:48:57 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
[... cut ...]
Jan 20 18:02:07 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 6:02:07 PM No converter available for 'JTQJ-BF-01LM/BW' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"3":40,"5":24,"8":8456,"9":256,"100":0,"150":0}}}'
Jan 20 18:02:07 loxberry npm[1740]:   zigbee2mqtt:warn 1/20/2019, 6:02:07 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
```
Would be great if you could parse it so we can use it at least as a is alive signal. even better if someone can figure out what it actually is trying to say us.

Koenkk, 2019-01-22:
The problem with `{"cid":"genBasic","data":{"modelId":"lumi.sensor_natgas"}}` is that it is also send when no button is clicked.

WebSpider, 2019-03-23:
Unstale

WebSpider, 2019-05-22:
Unstale

WebSpider, 2019-07-21:
Unstale

WebSpider, 2019-09-19:
*cough*

Koenkk, 2019-11-03:
@WebSpider what do you expect to be added?

WebSpider, 2019-11-04:
The last comment of @matlab22 pretty much sums it up for now: 

```Would be great if you could parse it so we can use it at least as a is alive signal. even better if someone can figure out what it actually is trying to say us.```

Koenkk, 2019-11-04:
These messages should provide, among other things, the battery level on 65281.1, but in the logs above I don't see the attribute (starts a `3`). 

> This struct seems to be documented here: https://github.com/Frans-Willem/AqaraHub/tree/master/documentation/devices

Where can I find the documentation?

closed 2020-01-10 by stale[bot]