Koenkk/zigbee2mqtt #277: CC2531 Router On by default

talondnb, 2018-08-12:
Is there a way to have a CC2531 boot straight into being On?

tb-killa, 2018-08-12:
What exactly do you mean by that question?

talondnb, 2018-08-12:
So when a CC2531 router is plugged in somewhere, it’s not enabled by default. You have to press one of the buttons on the device to enable it to function.

After a power failure, it will go back to the non enabled state. I would like it to boot into the enabled state by default without having to press any buttons.

tb-killa, 2018-08-13:
Maybe @ptvoinfo could help you .. i actually doesnt use this type of router ..

ptvoinfo, 2018-08-13:
@talondnb The router reconnects automatically after first successfull pairing. You need to re-pair the router after reflashing of a router or coordinator.

talondnb, 2018-08-13:
I have two routers paired, they function fine. But the default state on bootup is not ‘online’.

Every time I have a power failure or the router loses power for a moment, I have to go to each router and press SW1 to bring each one back online. It will still be paired but I have to bring them back online manually because the default state is a standby mode.

http://ptvo.info/cc2531-based-router-firmware-136/

“Buttons:

SW1 – Changes a virtual state of the router (the router sends it in a diagnostic report) and a state of the green light.“

This is what I mean. I have press SW1 so the green light is on, meaning router is online. I would like the device to boot straight into having its green light on.

I hope that makes sense.

mercenaryballs, 2018-08-14:
I do not think this is how this works. I have a CC2531 as a router and do not have to press any buttons after powering it up. I could never figure out what that green light really means, but it still routes without the green light on.

I had a power outage yesterday and I am seeing the router in the logs:
`2018-8-14 16:48:52 [34mDEBUG[39m Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
2018-8-14 16:48:52 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/Router1', payload: '{"state":false}'`

talondnb, 2018-08-14:
Hmm, I could be mistaken, but I was taking the payload: false (as you have) as not being online. If you press SW1 so the green light is on, that payload will become true. Is this not how it works?

talondnb, 2018-08-14:
Also, if you go to the dev tools and look for your device, does it have a state of ‘off’ or ‘on’? I have one with a state of ‘off’ currently, payload is ‘false’, the other is ‘on’ with a payload of ‘true’.

SargonofAssyria, 2018-08-16:
My router is also 'Off' after a powerloss, and has to be set to 'On' manually.
It would be nice to have the state set to 'On' after a powerloss.

talondnb, 2018-08-16:
@ptvoinfo, would this be possible?

mercenaryballs, 2018-08-16:
If the router is off then how are your receiving messages from it? Are we sure that that the status is not just reflecting the button status?

SargonofAssyria, 2018-08-17:
I am NOT recieving messages (from the router and device after the router) after a powerloss, that's why I manual have to set the router to ON again.

ptvoinfo, 2018-09-19:
Sorry for my late response. This virtual router state "On/Off" does not matter at all. The router operates and retransmits packets even if the green light is on/off. By default the green light is off because it is very bright. But you may press the button and turn on/off this light if you want. I may try to add a possibility to save the green light state and restore it after power loss. Vote.

lolorc, 2018-09-19:
what does it mean "virtual router" ?
IOW what's does it change if it's on or off ?

anyway, at the moment, I've been unable to pair a cc2531 router to cc2531 coordinator, looking forward to see all the firmwares upgraded to Z-Stack 3.0

basmeerman, 2018-09-30:
Might be interesting to know the exact firmware version used for flashing the cc2531. I ended up using "CC2531-router-diag-usb.hex" and it 'remembers' the state of the "switch" after powerloss (pulling the plug from the socket) and re-inserting it somewhere else (light remains green, status in zigbee2mqtt is on).

closed 2018-12-22 by stale[bot]