Koenkk/zigbee2mqtt #279: Add support for HEIMAN devices

sapkra, 2018-08-12:
Unsupported devices
https://www.aliexpress.com/wholesale?SearchText=HEIMAN+zigbee

I found some cheap zigbee devices on AliExpress but those are not supported by zigbee2mqtt. I would appreciate it when some one can add the support for those devices.

tb-killa, 2018-08-12:
You got a device like that by any chance?
If yes, please include the log files and the contents of the database.db file according to the [WIKI article](https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices).
I assume that this is an IASZONE device, so we have to take a few things into account here in general!
But it might also be relatively easy, we definitely need the support of @Koenkk

floris-b, 2018-10-06:
It seems I am able to pair the device but I never get a the message WARN No converter available in the log when I trigger the smoke alarm.
zigbee2mqtt:info 2018-10-6 09:22:10 0x005043c903389643 (0x005043c903389643): HS1SA - HEIMAN HEIMAN smoke detector (EndDevice)

I get the following message in the logs, nothing happens when I enable pairing:
zigbee2mqtt:info 2018-10-6 09:27:50 Device incoming...
zigbee2mqtt:info 2018-10-6 09:27:50 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'

I do not know where to find the database.db file, I am using hasssio

floris-b, 2018-10-07:
with log_level: debug 

```
  zigbee2mqtt:debug 2018-10-7 12:01:02 Recieved zigbee message of type 'devIncoming' with data '"0x005043c903377b0d"' of device 'SMOK_V16' (0x005043c903377b0d)
  zigbee2mqtt:info 2018-10-7 12:01:02 Device incoming...
  zigbee2mqtt:info 2018-10-7 12:01:02 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2018-10-7 12:01:02 Recieved zigbee message of type 'devStatus' with data '"online"' of device 'SMOK_V16' (0x005043c903377b0d)
  zigbee2mqtt:debug
```

diogopms, 2018-10-08:
Thanks for open this issue, I'm still interested in adding this device to zigbee2mqtt ([Heiman Siren](http://www.heimantech.com/product/153.html)).

termdude, 2018-11-22:
Here's the database.db entry:

`{"id":15,"type":"EndDevice","ieeeAddr":"0x005043c9a3306ba6","nwkAddr":5818,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_V16","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_V16","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"ssIasWd":{"dir":{"value":1},"attrs":{}}}}},"_id":"hOqFS3m5v77cRPef"}`

Koenkk, 2018-11-23:
@termdude can you try with adding to `devices.js`

```js
{
        zigbeeModel: ['SMOK_V16'],
        model: 'SMOK_V16',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

termdude, 2018-11-24:
2018-11-24 09:59:15 - warn: No converter available for 'SMOK_V16' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0018ed1702"}}'
2018-11-24 09:59:15 - warn: Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-11-24 10:00:00 - error: Failed to configure 0x0017880102c15019 0x0017880102c15019

Koenkk, 2018-11-24:
Can you provide the output when running with `DEBUG=* npm start`?

termdude, 2018-11-26:
I'm running under hassio, so I'm not quite sure how to catch the log output! I added DEBUG=* to run.sh, and managed to catch this:
`2018-11-26T07:40:35.632Z serialport:poller received "readable"
2018-11-26T07:40:35.632Z serialport:bindings read
2018-11-26T07:40:35.632Z serialport:unixRead Starting read
2018-11-26T07:40:35.633Z serialport:unixRead Finished read 8 bytes
2018-11-26T07:40:35.633Z serialport:main binding.read finished
Mon, 26 Nov 2018 07:40:35 GMT cc-znp { sof: 254,
  len: 3,
  type: 'AREQ',
  subsys: 'AF',
  cmd: 'dataConfirm',
  payload: { status: 240, endpoint: 1, transid: 1 },
  fcs: 55,
  csum: 55 }
2018-11-26T07:40:35.643Z serialport:main _read reading
2018-11-26T07:40:35.643Z serialport:bindings read
2018-11-26T07:40:35.644Z serialport:unixRead Starting read
Mon, 26 Nov 2018 07:40:35 GMT cc-znp:AREQ <-- AF:dataConfirm, { status: 240, endpoint: 1, transid: 1 }
Mon, 26 Nov 2018 07:40:35 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Mon, 26 Nov 2018 07:40:35 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1
2018-11-26T07:40:35.665Z zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: AF data request fails, status code: 240. MAC transaction expired.'
  zigbee2mqtt:error 2018-11-26 08:40:35 Failed to configure 0x005043c9a3316267 0x005043c9a3316267
2018-11-26T07:40:35.669Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-11-26T07:40:35.670Z serialport:poller Polling for "readable"
2018-11-26T07:40:36.096Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
2018-11-26T07:40:38.104Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
Mon, 26 Nov 2018 07:40:38 GMT zigbee-shepherd:request REQ --> ZDO:bindReq
Mon, 26 Nov 2018 07:40:38 GMT cc-znp:SREQ --> ZDO:bindReq, { dstaddr: 12229, srcaddr: '0x0017880102c15019', srcendpoint: 1, clusterid: 6, dstaddrmode: 3, addr_short_long: '0x00124b0018ed1702', dstendpoint: 1 }
2018-11-26T07:40:38.527Z serialport:main _write 28 bytes of data
2018-11-26T07:40:38.527Z serialport:bindings write 28 bytes
2018-11-26T07:40:38.528Z serialport:unixWrite Starting write 28 bytes offset 0 bytesToWrite 28
2018-11-26T07:40:38.529Z serialport:unixWrite write returned null 28
2018-11-26T07:40:38.530Z serialport:unixWrite wrote 28 bytes
2018-11-26T07:40:38.530Z serialport:unixWrite Finished writing 28 bytes
2018-11-26T07:40:38.530Z serialport:main binding.write write finished
2018-11-26T07:40:38.542Z serialport:poller received "readable"
2018-11-26T07:40:38.542Z serialport:bindings read
2018-11-26T07:40:38.543Z serialport:unixRead Starting read
2018-11-26T07:40:38.544Z serialport:unixRead Finished read 6 bytes
2018-11-26T07:40:38.545Z serialport:main binding.read finished
Mon, 26 Nov 2018 07:40:38 GMT cc-znp { sof: 254,
  len: 1,
  type: 'SRSP',
  subsys: 'ZDO',
  cmd: 'bindReq',
  payload: { status: 0 },
  fcs: 69,
  csum: 69 }
2018-11-26T07:40:38.555Z serialport:main _read reading
2018-11-26T07:40:38.555Z serialport:bindings read
2018-11-26T07:40:38.555Z serialport:unixRead Starting read
Mon, 26 Nov 2018 07:40:38 GMT cc-znp:SRSP <-- ZDO:bindReq, { status: 0 }
2018-11-26T07:40:38.559Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-11-26T07:40:38.560Z serialport:poller Polling for "readable"
2018-11-26T07:40:40.110Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
2018-11-26T07:40:42.119Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
2018-11-26T07:40:44.130Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
2018-11-26T07:40:46.139Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
2018-11-26T07:40:48.148Z pm2:client Calling daemon method pm2:getMonitorData on rpc socket:/root/.pm2/rpc.sock
Mon, 26 Nov 2018 07:40:48 GMT zigbee-shepherd:request REQ --> ZDO:bindReq
Mon, 26 Nov 2018 07:40:48 GMT cc-znp:SREQ --> ZDO:bindReq, { dstaddr: 12229, srcaddr: '0x0017880102c15019', srcendpoint: 1, clusterid: 6, dstaddrmode: 3, addr_short_long: '0x00124b0018ed1702', dstendpoint: 1 }
2018-11-26T07:40:48.529Z serialport:main _write 28 bytes of data
2018-11-26T07:40:48.529Z serialport:bindings write 28 bytes
2018-11-26T07:40:48.530Z serialport:unixWrite Starting write 28 bytes offset 0 bytesToWrite 28
2018-11-26T07:40:48.531Z serialport:unixWrite write returned null 28
2018-11-26T07:40:48.532Z serialport:unixWrite wrote 28 bytes
2018-11-26T07:40:48.532Z serialport:unixWrite Finished writing 28 bytes
2018-11-26T07:40:48.532Z serialport:main binding.write write finished
2018-11-26T07:40:48.543Z serialport:poller received "readable"
2018-11-26T07:40:48.543Z serialport:bindings read
2018-11-26T07:40:48.543Z serialport:unixRead Starting read
2018-11-26T07:40:48.544Z serialport:unixRead Finished read 6 bytes
2018-11-26T07:40:48.544Z serialport:main binding.read finished
Mon, 26 Nov 2018 07:40:48 GMT cc-znp { sof: 254,
  len: 1,
  type: 'SRSP',
  subsys: 'ZDO',
  cmd: 'bindReq',
  payload: { status: 0 },
  fcs: 69,
  csum: 69 }
2018-11-26T07:40:48.548Z serialport:main _read reading
2018-11-26T07:40:48.549Z serialport:bindings read
2018-11-26T07:40:48.549Z serialport:unixRead Starting read
Mon, 26 Nov 2018 07:40:48 GMT cc-znp:SRSP <-- ZDO:bindReq, { status: 0 }
2018-11-26T07:40:48.550Z serialport:unixRead waiting for readable because of code: EAGAIN
2018-11-26T07:40:48.550Z serialport:poller Polling for "readable"
`

termdude, 2018-11-26:
```
cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1282, srcaddr: 17129, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 94, securityuse: 0, timestamp: 5775028, transseqnumber: 0, len: 9, data: <Buffer 18 15 01 00 00 00 21 f0 00> } +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +3ms
  serialport:poller Polling for "readable" +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 2018-11-26 13:49:59 Received zigbee message of type 'devInterview' with data '"0x005043c9a3316267"'
  zigbee2mqtt:info 2018-11-26 13:49:59 Connecting with device...
  zigbee2mqtt:info 2018-11-26 13:49:59 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd Device: 0x005043c9a3316267 join the network. +11ms
  zigbee2mqtt:debug 2018-11-26 13:49:59 Received zigbee message of type 'devIncoming' with data '"0x005043c9a3316267"' of device 'SMOK_V16' (0x005043c9a3316267)
  zigbee2mqtt:info 2018-11-26 13:49:59 Device incoming...
  zigbee2mqtt:info 2018-11-26 13:49:59 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-11-26 13:49:59 New device with address 0x005043c9a3316267 connected!
  zigbee2mqtt:info 2018-11-26 13:49:59 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x005043c9a3316267"}'
  zigbee2mqtt:debug 2018-11-26 13:49:59 Received zigbee message of type 'devStatus' with data '"online"' of device 'SMOK_V16' (0x005043c9a3316267)
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 22 +316ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 17129, destendpoint: 1, srcendpoint: 1, clusterid: 1280, transid: 22, options: 48, radius: 30, len: 14, data: <Buffer 00 16 02 10 00 f0 02 17 ed 18 00 4b 12 00> } +0ms

```
```
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1280, srcaddr: 17129, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 94, securityuse: 0, timestamp: 5779147, transseqnumber: 0, len: 4, data: <Buffer 18 16 04 00> } +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +3ms
  serialport:poller Polling for "readable" +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 23 +3ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 17129, destendpoint: 1, srcendpoint: 1, clusterid: 1280, transid: 23, options: 48, radius: 30, len: 6, data: <Buffer 00 17 0c 00 00 f0> } +0ms
  serialport:main _write 21 bytes of data +7ms
  serialport:bindings write 21 bytes +7ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'OK' +0ms
```
```
  cc-znp { sof: 254,
  cc-znp   len: 33,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload: 
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 1280,
  cc-znp      srcaddr: 17129,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 94,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 5799452,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 13,
  cc-znp      data: <Buffer 18 1a 01 12 00 00 20 02 13 00 00 20 00> },
  cc-znp   fcs: 149,
  cc-znp   csum: 149 } +14ms
  serialport:main _read reading +2ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1280, srcaddr: 17129, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 94, securityuse: 0, timestamp: 5799452, transseqnumber: 0, len: 13, data: <Buffer 18 1a 01 12 00 00 20 02 13 00 00 20 00> } +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2ms
  serialport:unixRead waiting for readable because of code: EAGAIN +5ms
  serialport:poller Polling for "readable" +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee2mqtt:debug 2018-11-26 13:50:07 Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0018ed1702"}}' of device 'SMOK_V16' (0x005043c9a3316267)
  zigbee2mqtt:warn 2018-11-26 13:50:07 No converter available for 'SMOK_V16' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0018ed1702"}}'
```

Koenkk, 2018-11-27:
Good!

Can you do the same while having: `console.log(JSON.stringify(msg));` @ https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L670

termdude, 2018-11-28:
I pasted the log here: https://pastebin.com/raw/0BpTTNVd

JLFN, 2018-11-28:
Mine Heiman Smokedetector looks like this 
https://paste.ubuntu.com/p/d96JzrhWQB/

JLFN, 2018-11-28:
Hey should there be something like this on fromZigbee.js ?

```.js
    },
    HS1SA_smoke: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            return {smoke: msg.data.zoneStatus === 1};
        },
    },
    HS1SA_sensitivity: {
        cid: 'ssIasZone',
        type: 'devChange',
        convert: (model, msg, publish, options) => {
            const data = msg.data.data;
            const lookup = {
                '1': 'low',
                '2': 'medium',
                '3': 'high',
            };

            if (data && data.hasOwnProperty('65520')) {
                const value = data['65520'];
                if (value && value.startsWith('0x020')) {
                    return {
                        sensitivity: lookup[value.charAt(5)],
                    };
                }
            }
        },
````

JLFN, 2018-11-28:
Here is some info what the fuctions for this smokedetector https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-smoke-sensor.src/zigbee-smoke-sensor.groovy

JLFN, 2018-11-28:
@Koenkk  Do you want us to remove it then repair it or just make that change and restart zigbee2mqtt or you want also to re-pair it then trigger it? 
Dataentry after the edited device.js

```.js {"id":35,"type":"EndDevice","ieeeAddr":"0x005043c95333194d","nwkAddr":65040,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"b5db59bfd81e4f1f95dc57fdbba17931","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"b5db59bfd81e4f1f95dc57fdbba17931","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsFrequency":0,"mainsAlarmMask":0,"mainsVoltMinThres":0,"mainsVoltMaxThres":65535,"mainsVoltageDwellTripPoint":0,"batteryVoltage":30,"batteryPercentageRemaining":194,"batteryManufacturer":"","batterySize":255,"batteryAHrRating":0,"batteryQuantity":0,"batteryRatedVoltage":30,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryVoltThres1":0,"batteryVoltThres2":0,"batteryVoltThres3":0,"batteryPercentMinThres":40,"batteryPercentThres1":0,"batteryPercentThres2":0,"batteryPercentThres3":0,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"l8pZrsiNGSFvj1zi"} ```

Koenkk, 2018-11-28:
Yes can you try repairing and check if it works?

JLFN, 2018-11-28:
Nov 28 18:32:56 hassbian npm[5022]:   zigbee2mqtt:warn 2018-11-28 18:32:56 No converter available for 'HS1SA' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x00124b0014d9de2e","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}}'
Nov 28 18:32:56 hassbian npm[5022]:   zigbee2mqtt:warn 2018-11-28 18:32:56 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
Nov 28 18:32:58 hassbian npm[5022]:   zigbee2mqtt:warn 2018-11-28 18:32:58 No converter available for 'HS1SA' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x00124b0014d9de2e","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}

termdude, 2018-11-28:
Here we go again:
https://pastebin.com/raw/iQxM7nM5

Koenkk, 2018-11-28:
Can you try triggering the alarm and check if it causes any no converter available message?

JLFN, 2018-11-28:
@Koenkk  Nothing happens https://paste.ubuntu.com/p/zwHn3TwKsQ/

JLFN, 2018-11-28:
Here is the manual http://www.heimantech.com/uploads/2018/01/030813172350.pdf

termdude, 2018-11-29:
@Koenkk 
Ooops, I pasted the wrong debug log..

https://pastebin.com/raw/SPRBC9Nx is a new one!

Koenkk, 2018-11-29:
Can you try with 
```js
{
        zigbeeModel: ['SMOK_V16', 'b5db59bfd81e4f1f95dc57fdbba17931'],
        model: 'SMOK_V16',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 23}, cb),
                (cb) => device.read('ssIasZone', 'zoneState', cb),
            ];

            execute(device, actions, callback, 1000);
        },
    },
```

Koenkk, 2018-11-29:
I don't see the `failed to configure` or `successfully configured` messages which I saw in your previous logs. Can you try repairing?

Koenkk, 2018-11-30:
Turns out I used the wrong `enrollrspcode` (1) where 0 should be used: http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf (8.2.2.3.1.1 Payload Format) 

- 0: Success
- 1: This specific Zone type is not known to the CIE and is not supported.

Can you try with:

```js
{
        zigbeeModel: ['SMOK_V16', 'b5db59bfd81e4f1f95dc57fdbba17931'],
        model: 'SMOK_V16',
        vendor: 'HEIMAN',
        description: 'Smoke detector',
        supports: 'smoke',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.read('ssIasZone', 'zoneState', cb),
                (cb) => device.read('ssIasZone', 'zoneType', cb),
                (cb) => device.read('ssIasZone', 'zoneStatus', cb),
                (cb) => device.read('ssIasZone', 'zoneId', cb),
            ];

            execute(device, actions, callback, 1000);
        },
    },
```

termdude, 2018-11-30:
@Koenkk 
https://pastebin.com/raw/sdPhw4Ay

JLFN, 2018-11-30:
```@Koenkk
Nov 30 17:27:33 hassbian npm[26956]:   zigbee2mqtt:debug 2018-11-30 17:27:33 Received zigbee message of type 'devIncoming' with data '"0x005043c95333194d"' of device 'b5db59bfd81e4f1f95dc57fdbba17931' (0x005043c95333194d)
Nov 30 17:27:33 hassbian npm[26956]:   zigbee2mqtt:info 2018-11-30 17:27:33 Device incoming...
Nov 30 17:27:33 hassbian npm[26956]:   zigbee2mqtt:info 2018-11-30 17:27:33 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
Nov 30 17:27:33 hassbian npm[26956]:   zigbee2mqtt:info 2018-11-30 17:27:33 New device with address 0x005043c95333194d connected!
Nov 30 17:27:33 hassbian npm[26956]:   zigbee2mqtt:info 2018-11-30 17:27:33 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x005043c95333194d"}'
Nov 30 17:27:33 hassbian npm[26956]:   zigbee2mqtt:debug 2018-11-30 17:27:33 Received zigbee message of type 'devStatus' with data '"online"' of device 'b5db59bfd81e4f1f95dc57fdbba17931' (0x005043c95333194d)
```

Koenkk, 2018-11-30:
Great, looks like both of you managed to enroll the device.

The `No converter available for 'SMOK_V16' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":32}'` indicates the state of the zone.

What is the `zoneStatus` without and without the alarm?

termdude, 2018-11-30:
Looking good! zoneStatus 33 seems to be with alarm, and 32 without.

JLFN, 2018-11-30:
When it joins it goes to Zone 29
When you trigger the alarm it goes to Zone 33 when it beeps and when beep stops its go back to 32

Koenkk, 2018-11-30:
Can you check if `smoke: true` and `smoke: false` are published correctly; in the zigbee2mqtt directory, execute:

```
cd node_modules
rm -rf zigbee-shepherd-converters
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git -b SMOK_V16
cd ..
npm start
```

termdude, 2018-11-30:
> zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
> node index.js

                Zigbee2mqtt requires node version >=8.11 10, you are running v8.13.0!

  zigbee2mqtt:info 2018-11-30 19:32:55 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-11-30.19-32-54'
module.js:550
    throw err;
    ^

Error: Cannot find module 'debounce'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js:3:18)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-11-30T18_32_55_519Z-debug.log

termdude, 2018-11-30:
Yep, seems to publish correctly!

```
zigbee2mqtt:info 2018-11-30 19:37:53 MQTT publish, topic: 'zigbee2mqtt/0x005043c9a3316267', payload: '{"smoke":true,"linkquality":89}'
{"groupid":0,"clusterid":1280,"srcaddr":13130,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":14796770,"transseqnumber":0,"len":9,"data":{"0":25,"1":0,"2":0,"3":33,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":0,"cmdId":"statusChangeNotification","payload":{"zonestatus":33,"extendedstatus":0}}}
  zigbee2mqtt:info 2018-11-30 19:37:55 MQTT publish, topic: 'zigbee2mqtt/0x005043c9a3316267', payload: '{"smoke":false,"linkquality":84}'
{"groupid":0,"clusterid":1280,"srcaddr":13130,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":84,"securityuse":0,"timestamp":14801722,"transseqnumber":0,"len":9,"data":{"0":25,"1":0,"2":0,"3":32,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":0,"cmdId":"statusChangeNotification","payload":{"zonestatus":32,"extendedstatus":0}}}

```

JLFN, 2018-11-30:
It works for me , but strange it say state offline.
Niice now both mine smoke detector works.
See messages
https://pastebin.com/jsutmFVm

Koenkk, 2018-11-30:
offline/online is not always accurate. 

I will merge the support for this device, thanks!

termdude, 2018-11-30:
@Koenkk Great work! Thanks!

JLFN, 2018-11-30:
@Koenkk thank you! now i can use mine 9 pieces

sapkra, 2018-12-17:
4 HEIMAN devices are supported now - when someone needs more he/she can create a new issue.
Many thanks for the support.

closed 2018-12-17 by sapkra

Notrial, 2019-03-12:
Anyone bought the siren from the same manufacturer?

diogopms, 2019-03-12:
I would like to know if the siren from Heiman works using zigbee2mqtt. Thanks

JLFN, 2019-03-12:
> I would like to know if the siren from Heiman works using zigbee2mqtt. Thanks

Buy it and then try to add, someone needs to be first :)

Notrial, 2019-03-12:
These are all also indoor, can't find a viable outdoor solution either.

JLFN, 2019-03-12:
> These are all also indoor, can't find a viable outdoor solution either.
> Only thing that would work for now are those RF 433mhz versions connected to Sonoff RF Bridge...but I'm offtopic.

There are outdoor versions. http://www.climax.com.tw/z-sirens.php

Notrial, 2019-03-12:
This is all great, but what is the price, and where can I get it (outside Australia) :)

msadeghz, 2019-08-08:
@Koenkk 
Hi, I have Heiman PIR. It wants to the coordinator have an application with output cluster "`ssIasZone`". so I create an app on `0x0B` endpoint and the "`Match Descriptor Request`" is responded successfully. but the "enroll request" is sent to endpoint `0x0B` (not `0x01`). so your configuration in `device.js` doesn't worked. is there anyway to resolve this issue?

Koenkk, 2019-08-09:
@msadeghz the issue is that we actually don't respond to the enroll requests, but we just send an enroll reponse on zigbee2mqtt startup and when the device pairs (this is when the `configure` section will be executed) (and hope that the device did an enroll request).

I will implement correct answering of enroll requests in zigbee-herdsman (during my current refactor). That will likely fix the issue. I will also add the ias to endpoint 1 by default.

msadeghz, 2019-08-10:
> @msadeghz the issue is that we actually don't respond to the enroll requests, but we just send an enroll reponse on zigbee2mqtt startup and when the device pairs (this is when the `configure` [...]

Got it, so if the enroll response is sent in insufficient time, the end-device can't get it.
can I ask where do you want to edit? actually, which file and which method? 
I found that I can get it in serial port:
[image]

Koenkk, 2019-08-10:
Code will end up here: https://github.com/Koenkk/zigbee-herdsman/blob/controller/src/controller/controller.ts however this part is not functional/complete yet.

paulgking, 2020-01-04:
I just got a similar device as the hs1sa marked as orvibo. It identifies as zigbee model 98293058552c49f38ad0748541ee96ba 

Putting this in current herdsman as an additional id and it all seems to work ok, so this model should probably also be added.

arnonel, 2020-01-07:
I have that same Orvibo device (98293058552c49f38ad0748541ee96ba)

Please add to devices. Tx

Koenkk, 2020-01-08:
@paulgking @arnonel added, will be available in latest dev branch in a few hours.

arnonel, 2020-01-08:
Legend