Koenkk/zigbee2mqtt #280: Docker Stop and Bright Offline Message

Odianosen25, 2018-08-12:
When the docker image is stopped using a command like `docker stop 746f7fc1f845`. The brdige doesn't send an `offline` message. Is this normal and can this be added please?

If I am missing something, do kindly let me know.

Thanks

tb-killa, 2018-08-12:
Normaly acutally on ``dev`` branch @Koenkk implemented [Last will message](https://github.com/Koenkk/zigbee2mqtt/issues/267).
Please check this.
Maybe this can answer your question?

Odianosen25, 2018-08-12:
Hello @tb-killa,

Thanks for the response, but I don't understand where I am to edit or something.

I will just wait until there is an upgrade and it is moved from dev.

Regards

tb-killa, 2018-08-12:
if I am not mistaken there are also docker images from the dev branch.
maybe just set up a new instance for the test, copy the content from the ``data`` directory over and stop the original container.
Afterwards everything should continue as in the original.

closed 2018-11-09 by stale[bot]